jeremiah-c-leary/vhdl-style-guide #461: Report proper gitlab SHA

staerz, 2020-09-23:
**Environment**
Fedora32, VSG 2.0.0

**Describe the bug**
`vsg --version` outputs the following when calling from a directory which is not a non-git repository:
```bash
[staerz@staerz-dellt5820 datadrive]$ vsg --version
VHDL Style Guide (VSG) version 2.0.0
<blank line>
```
When being called from a directory that **is a git repository**, a git commit SHA is indicated:

```bash
[staerz@staerz-dellt5820 LASP (212-implement-vsg-2-0-0-coding-rules)]$ vsg --version
VHDL Style Guide (VSG) version 2.0.0
Git commit SHA: 113cda6
```
The problem though is that this "Git commit SHA" is **not** the commit SHA of the VSG installation, but the current SHA of the local directory. Simply check out a different commit, run `vsg --version` again and the problem becomes evident.

**To Reproduce**
Steps to reproduce the behavior:
1. Navigate into any directory that is (part of) a git repository (not being VSG itself)
2. Run `vsg --version`
3. Check out any other commit of that repository
4. Run `vsg --version` again and be amazed

**Expected behavior**
I expect to be indicated the git commit SHA of VSG, not the one of the local repository.

**Additional context**
I was trying to debug an installation issue of VSG on one of my machines and stumbled over that issue, trying to find the reported commit in the VSG commit history...

jeremiah-c-leary, 2020-10-08:
hey @staerz ,
I think I need to rethink how I get the sha.  It is still the chicken and egg thing I can't figure out.

staerz, 2020-10-08:
Hi @jeremiah-c-leary,

the chicken egg thing is solved: evolution tells us that egg was first.

But for your problem, here what I would do (under the assumption that my understanding of an installation process is right - which I'm absolutely not sure it is):
- have script (as easy as `git reflog HEAD -n1 --pretty=%h`) [or you do sth. that retrieves the sha from your remote repo) in the suite of your install prep scripts
- the install scripts simply call that script and write the output to a local config file
- vsg uses that config file to indicate the sha

That would be problem solved. Now I leave it to you if that way of thinking actually works for your processes as again, I have no clue how that works :wink:

staerz, 2021-01-05:
I would close this issue as it a duplicate of #486 and in the discussion there, a solution is already proposed.

closed 2021-01-05 by staerz