jeremiah-c-leary/vhdl-style-guide #463: Wrong exit code when running into configuration error

staerz, 2020-10-07:
**Environment**
Fedora 32, VSG 2.0.0

**Describe the bug**
VSG exits with exit code 0 when a faulty configuration is found, although the following is printed:
```bash
ERROR: Invalid configuration file: .vsg_local/configuration.json
while parsing a flow sequence
  in ".vsg_local/configuration.json", line 2, column 15
expected ',' or ']', but got '}'
  in ".vsg_local/configuration.json", line 12, column 5
```

Further note that the error report is output to `stdout`, but should be output to `stderr`.

**To Reproduce**
Steps to reproduce the behavior:
1. Have some code checked by VSG set up, using VSG configuration files
2. In a VSG configuration file, insert a typo, i.e. another `}`
3. Run VSG using this configuration on your code base
4. See error

**Expected behavior**
- Upon error, output should be printed to `stderr` (and not to `stdout`).
- Upon error, VSG must be terminated with an exit code different from `1`.

**Additional context**
Exit codes are essential when using any tool in further processing chains - in my case a CI setup.

jeremiah-c-leary, 2020-10-07:
Hey @staerz,

I can switch that to exit code and print to stderr.

You mentioned it should terminate with an exit code different from 1.  Did you mean exit with an exit code different than 0?

staerz, 2020-10-08:
> Hey @staerz,
> 
> I can switch that to exit code and print to stderr.
> 
> You mentioned it should terminate with an exit code different from 1. Did you mean exit with an exit code different than 0?

Hi @jeremiah-c-leary,

yes, sorry, that was indeed a typo which I just fixed. Of course I meant "different from zero" as zero is exit without error.
You may chose to make it exit with 2 or so (so actually also different from one) - so a user could even distinguish "unable to run vsg due to misconfiguration/wrong parameters" from an "exit after successfully running but finding violations".

jeremiah-c-leary, 2020-10-08:
I seem to remember why this occurred.  Someone I worked with wanted to know whether there was an error in a configuration file, so I created a JUnit output to catch it.  It didn't seem right at the time, and now that you point this out it makes perfect sense to just error out. 

I am working on this now.  I am getting tired of refactoring my tests to the new parser.

jeremiah-c-leary, 2020-10-08:
I fixed this issue. it will be in the 2.2.0 release.

staerz, 2020-10-08:
Hi @jeremiah-c-leary,

thanks for acting on this one so quickly.

I see you went for the exit 1 only option. That's fine although I would have wished for an exit code 2 if anything with the configuration goes wrong. That would have given an additional handle. But alright, your solution at least solves my problem :+1:

jeremiah-c-leary, 2020-10-08:
I will change it to 2 before the 2.2.0 release.

closed 2020-10-08 by jeremiah-c-leary