jeremiah-c-leary/vhdl-style-guide #468: Unexpected crash of VSG 2.0.0 when checking rules

staerz, 2020-10-13:
**Environment**
Ubuntu 18.04, VSG 2.0.0

**Describe the bug**
VSG fails with some python exception when running over a particular file.

**To Reproduce**
Steps to reproduce the behavior:
1. Run VSG on "slice.vhd" (see attachment below)
4. See error (see below)

**Expected behavior**
no failure

**Log output**
```bash
================================================================================
File:  src/deep_testing/components/slice.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 348
Total Violations:    3
----------------------------+------------+--------------------------------------
  Rule                      |  line(s)   | Solution
----------------------------+------------+--------------------------------------
  constant_004              |        121 | Change constant identifiers name to uppercase.
  constant_013              |        121 | Inconsistent capitalization of words: hf_load_generics, lf_load_generics
  constant_013              |        121 | Inconsistent capitalization of words: hf_load_generics, lf_load_generics
----------------------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
Traceback (most recent call last):
  File "/home/gitlab-runner/.local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==2.0.0', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.6/dist-packages/vsg-2.0.0-py3.6.egg/vsg/__main__.py", line 375, in main
    oJunitTestsuite.add_testcase(oRules.extract_junit_testcase(sFileName))
  File "/usr/local/lib/python3.6/dist-packages/vsg-2.0.0-py3.6.egg/vsg/rule_list.py", line 254, in extract_junit_testcase
    sLine += oRule._get_solution(dViolation)
  File "/usr/local/lib/python3.6/dist-packages/vsg-2.0.0-py3.6.egg/vsg/rules/constant/rule_013.py", line 55, in _get_solution
    sSolution = self.solution + ': ' + lTemp[0]
IndexError: list index out of range
```

**Additional context**
[slice.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/5373478/slice.txt)
^ rename to slice.vhd

Notes:
- just ran into it. Please let me know if I should also provide the vsg config files for that run.
- might be fixed when reworking the rules anyway
- no big prio for now (or any blocker for us), just reporting

jeremiah-c-leary, 2020-10-13:
Hey @staerz ,
I ran this on version 2.2.0 and it did not crash.  I tried with the --fix and without.

Could you provide the config file?

staerz, 2020-10-13:
Sure,
and really, it wasn't meant you to act on it right now. I really just wanted to keep track of the issue that we found in our setup, well knowing that it might not be an issue anymore with your reworked VSG.

We run VSG with the following parameters: `-c .vsg/rules/config.json .vsg/rules/disabled.json .vsg/rules/alignment.json .vsg/rules/casing.json .vsg_local/configuration.json -lr .vsg/rules`

Note that `.vsg/rules` doesn't actually contain any further local rule.
[disabled.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/5374698/disabled.txt)
[config.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/5374699/config.txt)
[casing.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/5374700/casing.txt)
[alignment.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/5374701/alignment.txt)
[configuration.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/5374720/configuration.txt)

And it may also well be that this configuration is actually not consistent with VSG 2.0.0 - as we're just migrating from VSG 1.11.1...

jeremiah-c-leary, 2020-12-04:
Hey @staerz ,
I would like to revisit this issue since my re-write it essentially done.

I ran the example file using master and it did not have an issue.

Would it be possible to run with master in your environment and see if it still crashes.

Tnanks,

--Jeremy

staerz, 2020-12-09:
I confirm it doesn't happen with the latest master (2.0.0-457-g30bfcad7) anymore.

closed 2020-12-09 by staerz