jeremiah-c-leary/vhdl-style-guide #471: output_configuration not working

Brandon-Valley, 2020-10-15:
**vsg -oc configuration.json not working**

When I run `vsg -oc configuration.json` I get the following error:

```
C:\projects\drc>vsg -oc configuration.json
Traceback (most recent call last):
  File "C:\Users\mt204e\AppData\Local\Programs\Python\Python37-32\Scripts\vsg-script.py", line 11, in <module>
    load_entry_point('vsg==1.11.1', 'console_scripts', 'vsg')()
  File "c:\users\mt204e\appdata\local\programs\python\python37-32\lib\site-packages\vsg\__main__.py", line 194, in main
    for sFileName in commandLineArguments.filename:
TypeError: 'NoneType' object is not iterable
```

I am trying to follow the example from:  https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#

Does anyone know what I am doing wrong?  Thanks in advance!

jeremiah-c-leary, 2020-10-15:
Greetings @Brandon-Valley ,

For the version you are running with, you need to also include the `-f` option.

I removed this limitation in version 2.0.0.

Regards,

--Jeremy

staerz, 2020-10-15:
Hi @jeremiah-c-leary,

I'm having a related problem with VSG 2.0.0.

Not that VSG would exit like @Brandon-Valley reports, but when I then look a the produced configuration file it doesn't represent the particular configuration that I put in: As I gave in #468, we use a set of configuration files (simply to have different kinds of rules gathered together instead of having one big config file).

And in particular, I'd expect to see a subset of rules to be disabled for a certain set of files (since I know the syntax is correct, but VSG is mis-firing warnings).

In fact, in #468, I gave a simplified version of our configuration file, so here is the full one:
[configuration.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/5387955/configuration.txt)

As you can see, for `modules/fpga/src/private/s10_devkit_top_arch.vhd` I need to disable some of the indentation rules, but also allow a particular architecture name.

Now the export of the full configuration doesn't distinguish rules between this and any other file, i.e. there is only one entry for `archtecture_025` - and this entry is the default entry from `config.json`.

When VSG runs, it seems though to apply the rule properly, but I'm still curious to know if I understand things properly, and especially the "global rules" vs. "rules for a particular file (set)".

BTW: I think [your configuration example](https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#rule-configuration-priorities) contains a bug: there is "rule" nested in "rule" - that doesn't seem right and it appears as if one line containing "rule: {" would be too much here.

staerz, 2020-10-15:
And maybe to add that a colleague has a problem that doesn't appear on the machine that I'm using:

Using VSG 2.0.0 here.

With python 3(.5)
```bash
Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/vsg/__main__.py", line 332, in main
    configuration = read_configuration_files(dStyle, commandLineArguments)
  File "/usr/local/lib/python3.5/dist-packages/vsg/__main__.py", line 132, in read_configuration_files
    validate_file_exists(sFilename, sConfigFilename)
  File "/usr/local/lib/python3.5/dist-packages/vsg/__main__.py", line 113, in validate_file_exists
    print('ERROR: Could not find file ' + sFilename + ' in configuration file ' + sConfigName)
TypeError: Can't convert 'dict' object to str implicitly
```

Similar with python 2(.7)
```bash
Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/vsg/__main__.py", line 332, in main
    configuration = read_configuration_files(dStyle, commandLineArguments)
  File "/usr/local/lib/python2.7/dist-packages/vsg/__main__.py", line 132, in read_configuration_files
    validate_file_exists(sFilename, sConfigFilename)
  File "/usr/local/lib/python2.7/dist-packages/vsg/__main__.py", line 113, in validate_file_exists
    print('ERROR: Could not find file ' + sFilename + ' in configuration file ' + sConfigName)
TypeError: cannot concatenate 'str' and 'dict' objects
```

That happens when trying to read a configuration with particular rules for some files.

So now I'm wondering if I fully misunderstood how the configuration files work...

jeremiah-c-leary, 2020-10-15:
Hey @staerz,

hmm...looking at the code I should use sExpandedFilename instead of sFilename.

This should be an easy change.

jeremiah-c-leary, 2020-10-15:
I have a fix already.  It now reports the file being called out that could not be found.

staerz, 2020-10-15:
OK, great, here is another similar one by another colleague:

```bash
Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 8, in <module>
    sys.exit(main())
  File "/hepuser/srdejong/.local/lib/python2.7/site-packages/vsg/__main__.py", line 332, in main
    configuration = read_configuration_files(dStyle, commandLineArguments)
  File "/hepuser/srdejong/.local/lib/python2.7/site-packages/vsg/__main__.py", line 134, in read_configuration_files
    for sGlobbedFilename in glob.glob(expand_filename(sFilename)):
  File "/hepuser/srdejong/.local/lib/python2.7/site-packages/vsg/__main__.py", line 209, in expand_filename
    return os.path.expanduser(os.path.expandvars(sFileName))
  File "/usr/lib/python2.7/posixpath.py", line 254, in expanduser
    if not path.startswith('~'):
AttributeError: 'dict' object has no attribute 'startswith'
srdejong@atlas-lab-002:lolli>python --version
Python 2.7.12
```

Brandon-Valley, 2020-10-16:
> Greetings @Brandon-Valley ,
> 
> For the version you are running with, you need to also include the `-f` option.
> 
> I removed this limitation in version 2.0.0.
> 
> Regards,
> 
> --Jeremy

This solved my issue,
- I had a version 1 variant installed from `pip install vsg`
  - On that version, running `vsg -oc configuration.json -f configuration.json` created **configuration.json**
- I have now upgraded to version 2.2.0
  - Now `vsg -oc configuration.json` works

Thank you so much for your quick response.  This is a fantastic tool with fantastic documentation!

closed 2020-10-16 by Brandon-Valley