jeremiah-c-leary/vhdl-style-guide #472: How to force VSG to use a particular python version

staerz, 2020-10-15:
**What is your question?**
How to force VSG to use a particular python version?

Reason of the question: Apparently there is different behaviour for VSG depending the underlying python installation.
VSG itself doesn't seem to require a particular python version (which I think it should) but since python3 is not backwards compatible to python2, it would be of uttermost importance to get the dependencies sorted out right.

Related to #471, #468.

jeremiah-c-leary, 2020-12-05:
I found this in the packaging documentation:

[image]

A little history...

VSG version 1.* was developed using python version 2.7
VSG version 2.* was intended to transition to 3.* because version 2.7 was not going to be supported anymore
The hope is that is would be backwards compatible with 2.7, but my intent was not to support 2.7.

It seems it may be time with version 3.0.0 to force a minimum version of python. 
I am currently running 3.8.2.  I could make that the minimum someone needs to run VSG.

Thoughts?

--Jeremy

staerz, 2020-12-16:
Absolutely, force a minimum python version - and make that python3!

Now, if you want to force a subversion of python3 is a good question.
Personally, I'm running Python 3.8.6, but on another machine it's 3.5.3 while on yet another one it's 3.6.9 (that's our CI machine for running VSG 2.0.0).

So from that I'd say it's safe to require python 3.5 or higher (but not yet 4 maybe...)

jeremiah-c-leary, 2020-12-17:
> So from that I'd say it's safe to require python 3.5 or higher (but not yet 4 maybe...)

I will set it to 3.5 or higher.

jeremiah-c-leary, 2020-12-17:
Hey @staerz ,

I added the minimum requirement of 3.5 to the setup.py file.  PIP will enforce this requirement.

I will close this issue as it can only be resolved after the next release.

closed 2020-12-17 by jeremiah-c-leary

staerz, 2020-12-17:
Absolutely fine with me. I'm happy that you found a solution (which seems to be the official one).