jeremiah-c-leary/vhdl-style-guide #474: function_010 documentation

staerz, 2020-10-27:
**Environment**
VSG 2.0.0, Fedora 32

**Describe the bug**
In the documentation of function_010 the second "Violation" should actually be "Fix" I presume.

**To Reproduce**

1. Go see [the documentation](https://vhdl-style-guide.readthedocs.io/en/latest/function_rules.html#function-010)

**Additional context**
I'm actually wondering if this rule would make any assumption on the casing of function names themselves, just like process_017 and process_19 - or if such rule simply doesn't exist yet. If so, adding rules on the function name capitalisation would be appreciated, although there might be multiple options allowed (UpperCamelCase, lowerCamelCase, ONLYUPPERCASE, onlylowercase, separation_by_underscore_all_lower, SEPARATION_BY_UNDERSCORE_ALL_UPPER, anything_MIXED - these are those that come to my mind :wink: [this could be a nice extension of the casing options anyway...]

jeremiah-c-leary, 2020-12-04:
@staerz ,

> In the documentation of function_010 the second "Violation" should actually be "Fix" I presume.

You are correct.

> I'm actually wondering if this rule would make any assumption on the casing of function names themselves, just like process_017 and process_19 - or if such rule simply doesn't exist yet.

It looks like I was missing that rule.  I added function_017, which defaults to lower case.

> although there might be multiple options allowed

At the moment enforcing styles other than upper and lower are out of my scope.

Could you grab the latest master and let me know if function_017 will work for you.

--Jeremy

staerz, 2020-12-09:
Alright, I see you fixed the typo.

Yes, function_017 also works as expected, thanks for adding it!

And agreed, it's enough to support (all) upper or (all) lower case as a first feature.

If you're getting bored, then you could allow other casing conventions in a later version :wink: but that's something I wouldn't need.

closed 2020-12-09 by staerz