jeremiah-c-leary/vhdl-style-guide #475: constant_007 false alert for dereferred constant definition

staerz, 2020-10-27:
**Environment**
VSG 2.0.0, Fedora 32

**Describe the bug**
constant_007 complains when assignment is not in same line as constant declaration.

This behaviour doesn't consider that the constant declaration can be spread over the package definition and package body where the action value is given (that may depend other functions that are only defined in that package body.

**To Reproduce**
Steps to reproduce the behavior:
1. Run on the example code below

***Example Code***
```vhdl
package mypackage is

  type t_mytype is record
    entry1 : std_logic_vector;
    entry2 : std_logic;
  end record;

  function to_mytype (in : std_logic_vector) return t_mytype;

  constant C_MYTYPE : t_mytype;

end package mypackage;

package body mypackage is

  function to_mytype (in : std_logic_vector) return t_mytype is
  begin
    return (entry1 <= in; entry2 <= '1');
  end function to_mytype;

  constant C_MYTYPE : t_mytype := to_mytype("010010");

end mypackage;
```

jeremiah-c-leary, 2020-12-04:
hmmm, this still seems to be an issue on master.  I am investigating.

jeremiah-c-leary, 2020-12-04:
@staerz ,
I figured out the issue.  The rule was being too greedy and was assuming the assignment operator := in the second constant declaration belonged to the first constant declaration.

The issue has been resolved.  Just waiting for confirmation.

It feels nice to get back to working these issues after spending so much time updating the parser.

--Jeremy

staerz, 2020-12-09:
I confirm that with VSG 2.0.0-457-g30bfcad7 this doesn't get reported anymore.

closed 2020-12-09 by staerz