jeremiah-c-leary/vhdl-style-guide #476: with ... select crashes VSG

staerz, 2020-11-06:
**Environment**
Fedora 32, python3, VSG 2.0.0

**Describe the bug**
A vhdl statement `with ... select` crashes VSG badly.

**To Reproduce**
Steps to reproduce the behavior:
1. Run VSG on the example code (see below)
2. See error

**Expected behavior**
No crashing of VSG

**Example Code**
```vhdl
library IEEE;
  use IEEE.std_logic_1164.all;

entity make_vsg_crash is
end make_vsg_crash;

architecture tb of make_vsg_crash is
  signal counter        : integer := 0;
  signal mnl_rst        : std_logic;
begin
    with counter select mnl_rst <=
      '1' when 9 to 12,
      '0' when others;
end;
```

**Logs**
Running on that code results in
```bash
Phase 1 of 7... Reporting
Total Rules Checked: 60
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==2.0.0', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-2.0.0-py3.8.egg/vsg/__main__.py", line 377, in main
    oRules.report_violations(commandLineArguments.output_format)
  File "/usr/local/lib/python3.8/site-packages/vsg-2.0.0-py3.8.egg/vsg/rule_list.py", line 252, in report_violations
    report.vsg_stdout.print_output(dRunInfo)
  File "/usr/local/lib/python3.8/site-packages/vsg-2.0.0-py3.8.egg/vsg/report/vsg_stdout.py", line 14, in print_output
    print_violations(dRunInfo)
  File "/usr/local/lib/python3.8/site-packages/vsg-2.0.0-py3.8.egg/vsg/report/vsg_stdout.py", line 97, in print_violations
    sOutputString += dViolation['solution']
TypeError: can only concatenate str (not "NoneType") to str
```

jeremiah-c-leary, 2020-12-04:
Hey @staerz ,
I ran your example on master and it didn't crash.  Looks like my parser rewrite is working:
```mono
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide$ vsg -f debug.vhd
================================================================================
File:  debug.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 81
Total Violations:    5
  Error   :     5
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  entity_015                | Error      |          5 | Add entity keyword
  signal_007                | Error      |          8 | Remove default assignment.
  with_001                  | Error      |         11 | Rewrite with as a process
  architecture_010          | Error      |         14 | Add *architecture* keyword.
  architecture_024          | Error      |         14 | Add architecture simple name
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide$ vsg -f debug.vhd --fix
================================================================================
File:  debug.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 81
Total Violations:    2
  Error   :     2
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  signal_007                | Error      |          9 | Remove default assignment.
  with_001                  | Error      |         14 | Rewrite with as a process
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
I am going to move this issue to the 3.0.0 release.

Let me know if you want to close it or if I should.

--Jeremy

staerz, 2020-12-04:
Hi @jeremiah-c-leary,

thanks for looking into this.

Yes, I was hoping that your rewrite catches it, but I had no time at all so far to test your new dev version, sorry.

I would try to reserve some time next week to run the validation on my side on our code base if that's fine for you - and we can close then.

staerz, 2020-12-09:
Alright,
I first had to update a few rules since you introduced generic map and port map (good move!) and then let it run over part of my code base.

I found a case where VSG still crashes with a not so helping error message and could reduce it to the following snippet:

```vhdl
...
architecture ...
[  signal ...]
begin

  make_tx_interface : block
  begin

  end block;

  -- Receive part for the UDP interfaces
  --
  -- default: always indicate being ready to receive data to allow
  -- modules to start transmission
  -- block the interface for intermediate states (trailer...)
  with tx_state select udp_rx_ready_o <=
    ip_tx_ready_i when IDLE | UDP,
    '0' when others;

end behavioral;
```

It seems the parser doesn't like to find a `with` after the end of a block.

Unfortunately, the error log is simply this:
```bash
Error: Unknown token in concurrent_selected_signal_assignment(146:3)
       Expecting with, found end
```
where the `146` is the line of the `end block;`

staerz, 2020-12-09:
And on a related rule that you have introduced: with_001:

How do you dare to tag this as an error?
This is totally valid VHDL code ...

Is there any motivation behind this (which would mean I've been doing it wrong for years...)!?

jeremiah-c-leary, 2020-12-09:
> It seems the parser doesn't like to find a with after the end of a block.

When I comment out the block it processes the rest.  The error:

```mono
Error: Unknown token in concurrent_selected_signal_assignment(146:3)
       Expecting with, found end
```
is saying the parser did not like something at line 146 at column three while trying to classify a `concurrent_selected_signal_assignment`.

In my test with your snippet of code it chokes on the `end`.  Although there is no concurrent assignment at that point.  My classifier must be looking too far ahead and thinking the with statement is part of the block.

I will investigate.

>How do you dare to tag this as an error?
>This is totally valid VHDL code ...
>
>Is there any motivation behind this (which would mean I've been doing it wrong for years...)!?

These rules were originally based off of what my company follows.  We prefer to use processes instead of `with` statements.  Hence the reason there is only a single rule.  Now that VSG is more widely used, it would seem prudent to not restrict what code structures could be used.

I can add an issue for updating the `with` rules.

staerz, 2020-12-09:
Alright.

So I'm glad to have found an exception :smiling_imp:

And for the rule `with_001`, I think it's a really good idea to have it, but I'd simply disable it by default, that's all.

jeremiah-c-leary, 2020-12-10:
Hey @staerz ,

I just pushed a fix for the error.

Essentially, while in the `block_statement_part` I was searching for concurrent statements.  Part of the check is looking for a `concurrent_selected_signal_assignment` by search for the keyword `with` within a certain number of tokens.  It turned out the structure you had the `with` was within that search range.  So it was failing on the block end keyword because it thought it was in the `concurrent_selected_signal_assignment`.

staerz, 2020-12-17:
Alright, with VSG in f1ba134c that problem is gone.

I checked in all files where I had disabled (`--vsg_off`) due to the used `with` statement in one of our components. That was 12 files...

I think you can close this one.

jeremiah-c-leary, 2020-12-17:
Awesome.  I will close it.

closed 2020-12-17 by jeremiah-c-leary