jeremiah-c-leary/vhdl-style-guide #484: End of package declaration is spliced into the package body

james-tait, 2020-11-25:
**Environment**
VSG 2.2.0

**Describe the bug**
When VSG tries to fix simultaneous package_007 and package_014 violations, it splices the `end package` inside the package body.

**To Reproduce**
Steps to reproduce the behaviour:

Command: `vsg --fix -f file.vhd`
Example code: 

```
package pkg is
  
  procedure proceedure_name;  

end;

package body pkg is

  procedure proceedure_name (
    constant a : out integer ) is
  begin
    while (condition) loop
	
    end loop;
  end;
end;
```

**Expected behaviour**

* The `end;` on line 6 becomes `end package pkg;` (pacakge_007, package_014)
* All the indents remain the same

**Observed behaviour**
* The `end;` on line 15 becomes `end package;`
* The indent of line 15 is set to zero
* No errors or warnings are reported afterwards

```
package pkg is

  procedure proceedure_name;

end;

package body pkg is

  procedure proceedure_name (
    constant a : out integer) is
  begin
    while (condition) loop

end package loop;
  end;
end;
```

**Additional context**
Again, thank you for all your work

jeremiah-c-leary, 2020-11-25:
Hey @james-tait ,

I was able to reproduce the issue with version 2.2.0.  It looks like VSG is getting confused with which **end** is associated with the package declaration.

I am currently working towards a 3.0.0 release in January.  I ran the example code on master, which will become the 3.0.0 release, and it produced:

```vhdl
  1
  2 package pkg is
  3
  4   procedure proceedure_name;
  5
  6 end package pkg;
  7
  8 package body pkg is
  9
 10   procedure proceedure_name (
 11     constant a : out integer) is
 12   begin
 13     while (condition) loop
 14
 15     end loop;
 16   end;
 17
 18 end;
```

I re-wrote the parser and underlying methodology for version 3.0.0 to address these types of issues I was running into.

Could I request you run with what I currently have on master?

james-tait, 2020-12-03:
Firstly thank you for getting back to me so quickly. Secondly, I'm sorry for how long it has taken for me to be able to test it.

I have now tested the file with the master version (commit ccc38036131f89fe3d0f60d10edf7d80ddf9005), and it works as expected. Thanks for your help!

closed 2020-12-03 by james-tait