jeremiah-c-leary/vhdl-style-guide #486: Version report is out of date

james-tait, 2020-12-03:
**Environment**
VSG, pulled from master at commit ccc38036131f89fe3d0f60d10edf7d80ddf9005a

**Describe the bug**
When VSG reports it's version, it reports an old version number/commit

**To Reproduce**
Steps to reproduce the behavior:
1. `vsg --version`

**Expected behavior**
A more recent version number.

**Screenshots**
```
VHDL Style Guide (VSG) version 2.0.0
Git commit SHA: 95ceabb
```

**Additional context**
I realise this isn't a release so what's expected is debatable! In any case it caused some confusion,  so we thought it worth mentioning.

jeremiah-c-leary, 2020-12-04:
@james-tait ,
For some reason I have had trouble handling the version correctly.  I need to rethink the strategy because there are other issues people have ran into.

jeremiah-c-leary, 2020-12-05:
Hey @james-tait ,

I have  a fix in for this issue if you care to check it out.

Otherwise I will close this issue when I release 3.0.0

staerz, 2020-12-09:
Hi @jeremiah-c-leary,

I'm finally coming to test your new VSG version.

And yes, you remember we had this discussion.

What I get now is the following:
```bash
Finished processing dependencies for vsg===master
[staerz@steffenswkdell VSG (master)]$ vsg --version
VHDL Style Guide (VSG) version: master
Git commit SHA: NA
[staerz@steffenswkdell VSG (master)]$ cd ..
[staerz@steffenswkdell datadrive]$ vsg --version
VHDL Style Guide (VSG) version: master
Git commit SHA: NA
[staerz@steffenswkdell datadrive]$ cd -
/mnt/datadrive/VSG
[staerz@steffenswkdell VSG (master)]$ git describe
2.0.0-457-g30bfcad7
```

Version "master" is not really meaningful as that's a branch is not a static reference :wink:

As I already said, I don't know what the usual procedure is when you ship it via pip, but for the local install, what I would suggest is:
- upon running `python setup.py install` you evaluate `git describe` and dump the output to a file, e.g. `vsg.version` in the installation directory.
- a call of `vsg --version` should then simply go grab (`cat`) the content of that file.

I would hope that shipping via pip would also allow you to do something similar, i.e. to provide a version file that would either be derived when you release to pip or you prepare it in your deployment process.

jeremiah-c-leary, 2020-12-09:
> upon running python setup.py install you evaluate git describe and dump the output to a file, e.g. vsg.version in the installation directory.
> a call of vsg --version should then simply go grab (cat) the content of that file.

I think that solves all the issues I was having with the SHA.

> I would hope that shipping via pip would also allow you to do something similar, i.e. to provide a version file that would either be derived when you release to pip or you prepare it in your deployment process.

I don't know if PIP issues a `setup.py install` when it downloads the package.  That could be an issue since it would not have access to the Git repo to build the file.  If it is an issue, maybe I can have it check to see if the file exists and then act accordingly.  If it already exists when installing, then do not update the file.  If the file does not exist, then build the file.  Although in that case the first install would build the file, but subsequent ones would not.

I will investigate.

Thanks,

--Jeremy

jeremiah-c-leary, 2021-02-08:
Hey @staerz and @james-tait ,

I think I finally have this all figured out and implemented.  There are essentially three possible use cases for generating a version:

  1)  pip
  2)  python setup.py install
  3)  vsg --version when not installed via pip

The solution was to have a file `version_info.py`, which is never committed into the repo.  The file contains two variable definitions:  `sVersion` and `sShaNum`.  To support method 1, I will create the `version_info.py` after I tag for release and before publishing to pypi.  The `setup.py` file searches for `version_info.py` and if it finds it, it will use the variables values during install via pip.
```mono
$ vsg --version
VHDL Style Guide (VSG) version: 3.0.0
Git commit SHA: 7a4fd205e3df036ec6a069df034c1517881595d1
```
To support method 2, the `setup.py` file searches for `version_info.py` but can not find it.  It then calls two functions in `version.py` to pull the git SHA and the current branch.  When requesting the information, it temporarily switches the directory to the local git repo so it can get the correct SHA and branch name.  This fixes the issue where `--version` would pull the git SHA from a repo other than VSG's.  The version pip will report for VSG will be of this form:  `0.0.0-<branch_name>-<git_sha>`
```mono
urllib3                1.25.8
vsg                    0.0.0-master-7a4fd205e3df036ec6a069df034c1517881595d1
wadllib                1.3.3
```
To support method 3, where you are working in a repo and VSG was not installed (either through pip or setup.py), the `version_info.py` file does not exist.  Therefore, VSG will report the version as `None` and the SHA of the head will be reported:
```mono
$ vsg --version
VHDL Style Guide (VSG) version: None
Git commit SHA: 7a4fd205e3df036ec6a069df034c1517881595d1
```

I believe I have every corner case covered for properly reporting the version and SHA.  When you have some free time, if you could double check it is working on your end I would appreciate it.

Thanks,

--Jeremy

staerz, 2021-02-08:
Hi @jeremiah-c-leary,

I have pulled the changes and am happy to report that it works - at least for the local installation from the repo.

I would still suggest a few minor modifications in what you report/retract as git information.

Don't make your life complicated, and directly use `git describe --tags` for the version field.
You can call this from any branch or commit HEAD and it will give the last tag with offset of commits and a short commit SHA.

Anybody using git (= anybody who has installed VSG locally from the repo) would know how to interpret the output of `vsg --version` then.
It would also shorten the name of your installation directory (and will still always be unique).

When running `vsg --version` when not being installed via pip, couldn't you also simply run the `git describe --tags`? (I.e. I'm not quite sure to understand that case: That would be having vsg in a repo, calling the local file, right?)

I would find a version "0.0.0" or "None" always a little disturbing.

Giving the full SHA in the second line is fine in any case.

staerz, 2021-02-08:
Hi @jeremiah-c-leary,

triggered by the other thread:

By running `sudo python setup.py install 2> install_err.txt` (and then checking that file:

```
/usr/lib/python3.8/site-packages/setuptools/dist.py:480: UserWarning: The version specified ('0.0.0-master-77af5e639a467f99f42247c1ebd8090055323fd9') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
...
```

Please check what `setuptools` requires. I would hope that it would be compatible with the output of `git describe --tags`.
If not, then we have to see what better solution there is.

staerz, 2021-02-08:
... OK, looking into PEP 440, I see that it's not compatible with `git describe --tags`.

But you could go with `<version>[rc<N>.dev<M>]` where `N` could be the issue number and `M` the "distance" from the last tag, i.e. the central part of the output of `git describe --tags`.

You get `<version>` by `git describe --abbrev=0`.

Only in development, the suffix `rc` (Release Candidate) `.dev` would be there.

imd1, 2021-02-08:
Hi

I think there is an assumption that the files are part of a git repository but I tend to download a zip file so I'm getting the following error when I run `python setup.py install`

```
fatal: Not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "setup.py", line 15, in <module>
    sBranch = version.get_git_branch()
  File "/vhdl-style-guide-master/vsg/version.py", line 42, in get_git_branch
    lActual = subprocess.check_output(['git', 'branch', '--show-current'])
  File "/Python-3.8.1/Lib/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/Python-3.8.1/Lib/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', 'branch', '--show-current']' returned non-zero exit status 128.

```

jeremiah-c-leary, 2021-02-09:
Hey @imd1 ,

Looks like you found the one case I did not consider.  Let me investigate a little to see how to handle a zip file.

--Jeremy

jeremiah-c-leary, 2021-02-09:
Hey @staerz ,

> ... OK, looking into PEP 440, I see that it's not compatible with git describe --tags.
>
> But you could go with <version>[rc<N>.dev<M>] where N could be the issue number and M the "distance" from the [...]

So I read most of PEP 440 and it appears your suggestion could work.  If we would through the combinations:

pypi releases
--------------

  The vsg/version_info.py file will be filled out before publishing

python setup.py install
--------------------------

  The `vsg/version_info.py` does not exist, so I will issue `git describe --abbrev=0`.  On my system this returns:

```mono
$ git desribe --tags
2.0.0-570-g6b134f56
```
I split the returned value on dashes and will get this list:
```python
   lDescribe = ['2.0.0', '570', 'g6b134f56']
```
I create the file `vsg/version_info.py` file as follows:
```python
sVersion = '2.0.0.dev570'
sShaNum = 'g6b134f56'
```
This file will be installed along with VSG into the site-packages location.

vsg --version without install
--------------------------------

This would follow the previous method except the `vsg/version_info.py` file is not created.

installation via zip file
------------------------

The zip file has the branch name in it though:  `vhdl-style-guide-master`.  However, without git, I have no means of extracting any information to create a proper version number.

I could use `0.0.0.dev0` for the version.

jeremiah-c-leary, 2021-02-09:
Hey @staerz and @imd1 ,

I pushed a possible solution for the version that will hopefully address the issues.  Still not sure about the the zip file solution though, but at least it should not crash.

When you get a chance, please try it out and let me know if it works.

Thanks,

--Jeremy

imd1, 2021-02-09:
> Hey @staerz and @imd1 ,
> 
> I pushed a possible solution for the version that will hopefully address the issues. Still not sure about the the zip file solution though, but at least it should not crash.
> 
> When you get a chance, please try it out and let me know if it works.
> 
> Thanks,
> 
> --Jeremy

I would have thought many people will pick a release (3.0.0) and download a zip file, and expect the version number to be correct. In this case, the version must be baked into some file somehow?

imd1, 2021-02-09:
> Hey @staerz and @imd1 ,
> 
> I pushed a possible solution for the version that will hopefully address the issues. Still not sure about the the zip file solution though, but at least it should not crash.
> 
> When you get a chance, please try it out and let me know if it works.
> 
> Thanks,
> 
> --Jeremy

Tried latest master and still have a problem...

```
fatal: Not a git repository (or any of the parent directories): .git
0.0.0.dev0
unknown
Traceback (most recent call last):
  File "setup.py", line 24, in <module>
    with open(sVersionInfoFileName, 'w') as oFile:
FileNotFoundError: [Errno 2] No such file or directory: 'vsg/version_info.py'

```

jeremiah-c-leary, 2021-02-09:
@imd1 ,

Where are you relative to the `setup.py` file when you issue `python setup.py install`?

jeremiah-c-leary, 2021-02-09:
@imd1 ,
Until I get this figured out, create the file `vsg/version_info.py` with the following contents:
```python
sVersion = '0.0.0'
sShaNum = 'unknown'
```

The odd thing is I can see the correct values were returned for version (0.0.0.dev0) and the SHA (unknown).

I did have an issue similar to yours, but I thought I pushed a fix for it.  Essentially the current working directory was incorrect when I tried to reference `vsg/version_info.py`.  Could you double check line 67 exists in your `vsg/version.py` file:
```python
 54 def get_version_info(sVersion, sShaNum):
 55     if sVersion is not None and sShaNum is not None:
 56         return sVersion, sShaNum
 57
 58     try:
 59         sReturnPath = os.getcwd()
 60         sPath = os.path.dirname(__file__)
 61         os.chdir(sPath)
 62         lActual = subprocess.check_output(['git', 'describe', '--tags'])
 63         lActual = str(lActual.decode('utf-8')).split('\n')
 64         lVersion = lActual[0].split('-')
 65         sVersion = str(lVersion[0]) + '.dev' + str(lVersion[1])
 66         sShaNum = str(lVersion[-1])
 67         os.chdir(sReturnPath)
 68         return sVersion, sShaNum
 69     except subprocess.CalledProcessError:
 70         return '0.0.0.dev0', 'unknown'
```

oh...maybe I just figured it out.  I think I need to duplicate line 67 under the except.

jeremiah-c-leary, 2021-02-09:
Hey @imd1 ,

I just pushed an update.  If you could check it out I would appreciate it.

Thanks,

--Jeremy

staerz, 2021-02-09:
Hi @jeremiah-c-leary,

I think for the case of the zip file, we should also find a way.

How is that zip file created (and distributed)? Is it automatically done via the git repo process or are you manually providing it?

I could imagine a CI to provide it, then it would be relatively easy.

jeremiah-c-leary, 2021-02-09:
Hey @staerz and @imd1 ,
I used this option on github:
[image]
which I suspect @imd1 used also.

I check the tags page:
[image]

When downloaded the file is named:
[image]

It would have been nice if github included more meta data in the zip file name.  It would make sense for them **not** to include extra files in the zip, but they could do something with the name of the zip.

I compared a file listing between the zip file and the local git sandbox and the zip file is missing the `.git` directory.  So we can not interrogate anything in that directory.

imd1, 2021-02-09:
> Hey @imd1 ,
> 
> I just pushed an update. If you could check it out I would appreciate it.
> 
> Thanks,
> 
> --Jeremy

Installation completes with your placeholder version displayed
```
$ vsg --version
VHDL Style Guide (VSG) version: 0.0.0.dev0
Git commit SHA: unknown
```

imd1, 2021-02-09:
> Hey @staerz and @imd1 ,
> I used this option on github:
> [image]
> which I suspect @imd1 used also.
>
> I check the tags page:
> [image]
>
> When downloaded the file is named:
> [image]
>
> It would have been nice if [...]

I hit the download zip file button

staerz, 2021-02-09:
Hm, I just checked, and actually on the GitLab side it's the same: You just get the bare content of the repo when downloading the zip file.
That makes sense to the extend that you get the repo.

Well, under these conditions I'd say @jeremiah-c-leary, you should add a version file to the repo itself.

I suggest you simply have a simplified version, i.e. "2.0.0" (or "2.2.0", or then later "3.0.0") that you update when ever you know something new is being prepared. I know, it's the chicken-egg-problem, as tagging usually happens outside of the repo itself, but under these conditions, I'd say that rather having something "good enough" than having nothing ("0.0.0") is the better way to go.

And you could make sure that "vsg --version" only relies on that simplified version file if the other options don't work - i.e. only for the case when you downloaded the repo as a zip.

imd1, 2021-02-09:
> Hm, I just checked, and actually on the GitLab side it's the same: You just get the bare content of the repo when downloading the zip file.
> That makes sense to the extend that you get the repo. [...]

Could you not get your CI system to generate a version file from a tag and include it in the release ZIP file?

staerz, 2021-02-09:
> Could you not get your CI system to generate a version file from a tag and include it in the release ZIP file?

Well, this is exactly what I thought in the first place, unfortunately this is not linked to the zip download button, but would only be linked to a release (where you could download artifacts from the release page - there you can do everything).

jeremiah-c-leary, 2021-02-10:
> Well, under these conditions I'd say @jeremiah-c-leary, you should add a version file to the repo itself.
>
> I suggest you simply have a simplified version, i.e. "2.0.0" (or "2.2.0", or then later [...]

I can keep the version in the version.py file and keep the SHA in another file.  Then it is a matter of checking for a `.git` directory and whether the SHA file exists:

```mono
.git      SHA file
====      ========
False       X        Zip file detected.  Use version in version file and add use local version identifier:  3.1.0+zip_file
True       True      Pip install.  Use version in version file: 3.1.0
True       False     Local install.  Use version in version file and git desribe --tags for sha:  3.1.0.dev576
```

jeremiah-c-leary, 2021-02-10:
Hey @staerz and @imd1,
I just pushed an update that I hope covers all the use cases.  Please let me know if I missed something.

Thanks,

--Jeremy

imd1, 2021-02-10:
> Hey @staerz and @imd1,
> I just pushed an update that I hope covers all the use cases. Please let me know if I missed something.
> 
> Thanks,
> 
> --Jeremy

Installs OK and displays version as: 
```
$ vsg --version
VHDL Style Guide (VSG) version: 2.0.0+zip.file
Git commit SHA: Unknown.  Installed via zip file.
```

staerz, 2021-02-10:
Hi @jeremiah-c-leary,

@imd1 should best be suited to reply this, but I think your version now for the zip file should have triggered the version (PEP 440) warning during install.

I think you should try to avoid that as well.

As said, when installing via zip file, you have no chance to get any reliable version other than via the prepared (checked in) file, so I'd suggest to remove the post-version suffix. If you want a suffix, then use the "postN" suffix, where N would be any number that you could simply fix to "1" (as I cannot imagine a reliable and reproducible way of getting a "version" out of this zip file alone).

Given that you have the line on the git commit SHA stating that it's installed via a zip file, I think that's just fair enough.

For the rest (install via pip or via python) I agree.

So basically you always get the version from the file and have or don't have a suffix dependent on the way you installed.

jeremiah-c-leary, 2021-02-11:
> As said, when installing via zip file, you have no chance to get any reliable version other than via the prepared (checked in) file, so I'd suggest to remove the post-version suffix. If you want a [...]

I thought about using the post release, but the description did not seem to fit:

[image]

It is clearly not a Pre or Final release.  It might be considered a Development release:

[image]

but we really do not know anything about the relationship between the zip file and the git repo.

So I thought a local version would work:
[image]
...but that does not really fit either.  However, it does allow me to indicate it came from a zip file.

I ran an experiment to set the version to `2.0.0+zip.file` and performed an install.  I verified no warning was issued.

staerz, 2021-02-11:
Alright, if that didn't throw a warning, who am I to complain :wink:

jeremiah-c-leary, 2021-02-11:
> Alright, if that didn't throw a warning, who am I to complain 😉

If "they" had just given us more information in the zip file name.

jeremiah-c-leary, 2021-02-11:
Hey @imd1 ,

I just pushed an update to check for the `.git` directory and my initial testing shows it it working.  Could you check if it reports correctly in your Docker examples?

Thanks,

--Jeremy

jeremiah-c-leary, 2021-02-11:
Hey @staerz ,

Just wanted to thank you for pointing out `setup.py` can report warnings.  I never would have thought to look.

staerz, 2021-02-11:
Hi @jeremiah-c-leary,

I have just tried the new update on the version and am very happy with the solution.
For me it was now the install via the install script. What I get now is:

```
VHDL Style Guide (VSG) version: 2.0.0.dev579
Git commit SHA: g1b2b3ba1
```
and I tried this from the repository as well as any other location, the result is the same, which is as it should be.

Thanks for implementing this!

From my side, you can close the issue.

imd1, 2021-02-11:
I confirm that I get the version number + zip file comments and no SHA when I install as root from a zip file

I still get strange behavior when I install as non root and with `python setup.py install --user` where I never get a zip file comment and get the SHA of my VHDL project GIT repo if at its root, and blank if not. However, I do not plan to do this now, so I'm happy to close this issue now

imd1, 2021-02-11:
Of course I want to see `3.0.0` when this version is released :-)

jeremiah-c-leary, 2021-02-11:
> Of course I want to see 3.0.0 when this version is released :-)

Getting closer with every issue we close!!

I can't believe I just have a handful of issues left to complete before 3.0.0.

closed 2021-02-11 by jeremiah-c-leary