jeremiah-c-leary/vhdl-style-guide #487: New bugs related to candidate for 3.0.0

staerz, 2020-12-09:
**Environment**
Latest version 2.0.0-457-g30bfcad7.

**Describe the bug**
This is a collection of bugs and issues found with the new version for new features.

**Expected behavior**
A clear and concise description of what you expected to happen.

### Broken link to Configuring Type of Instantiations
The doc for `instantiation_034` points to 'Configuring Type of Instantiations' but the link is broken, although the page actually exists (dashes vs. underscores)

... I'll keep adding things along the way of testing ...

jeremiah-c-leary, 2020-12-10:
@staerz ,
I believe I fixed the link.

staerz, 2020-12-17:
Yup, the link now works.

I have added a few other issues though. I'm done checking for today and will likely be able to pursue the next couple of days, so let me know if you need any feedback.

jeremiah-c-leary, 2020-12-17:
> Configuration of blank lines

Just to clarify the issue...you would like the comments to be considered part of the rule check.  Such that a blank line would be added above the comments?

staerz, 2020-12-17:
> > Configuration of blank lines
> 
> Just to clarify the issue...you would like the comments to be considered part of the rule check. Such that a blank line would be added above the comments?

Yes, blank line above the comment (required). If there is no comment, also no problem, as that falls back to the current situation.

jeremiah-c-leary, 2020-12-19:
Hey @staerz ,
I believe the following issues have been addressed:
* Unclear error message upon unexpected error
* instantiation_003 circular error message
* Typo in doc for instantiation_010
* Documentation for architecture-002
* Update copyright statement

staerz, 2020-12-19:
Alright, I updated the issue description. In fact you also already fixed the other typo doc (block_202)

jeremiah-c-leary, 2020-12-19:
Hey @staerz ,

I put in a rule to handle this issue:
* Add a rule for architecture similar to block_205

When you get a chance could you validate it.

Thanks,

--Jeremy

staerz, 2020-12-21:
Hi @jeremiah-c-leary,

I have reverted to the status d7403e62 ("Fixing 3.0.0 release date") and got the following bug, when `comment_001` is disabled:

```
Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg===master', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/__main__.py", line 418, in main
    oRules.check_rules(bAllPhases=commandLineArguments.all_phases, lSkipPhase=commandLineArguments.skip_phase)
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/rule_list.py", line 215, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/rules/align_tokens_in_region_between_tokens_when_between_tokens_unless_between_tokens.py", line 63, in analyze
    lToi = oFile.get_tokens_bounded_by_token_when_between_tokens(self.left_token, self.right_token, self.lBetween[0], self.lBetween[1])
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/vhdlFile/vhdlFile.py", line 164, in get_tokens_bounded_by_token_when_between_tokens
    return extract.get_tokens_bounded_by_token_when_between_tokens(oLeft, oRight, oStart, oEnd, self.lAllObjects, self.oTokenMap, include_trailing_whitespace)
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/vhdlFile/extract/get_tokens_bounded_by_token_when_between_tokens.py", line 14, in get_tokens_bounded_by_token_when_between_tokens
    if iStart < iLeft and iRight < iEnd:
TypeError: '<' not supported between instances of 'NoneType' and 'int'
```

I traced it down to the commit 712513d2 (related to #501) which seems to introduce this bug: I removed that commit from the history and ran again, then it's fine.

I also hereby confirm that the commits related to introducing `architecture_200` work fine, I'll update the issue description.

jeremiah-c-leary, 2021-02-06:
Hey @staerz ,

I was reviewing this issue and it looks like there are three remaining items:

1) Configuration of blank lines
2) Installation warning
3)  comment_001

I believe the first one is covered by issue #504

I tried the second one and installed without an error.

For the third one, comment_001 no longer exists.  Looking at the log file it appears to have occured back in March.  Could you check if this is still an issue?

Thanks,

--Jeremy

staerz, 2021-02-08:
Hi @jeremiah-c-leary,

I updated the issue description, yes, 1) is covered by #504.

2) Is still occurring. This seems to be an issue with your `MANIFEST.in`: It contains the line `include README.md`, but there is no `README.md` in the repository.

3) `comment_001`: I think I never mentioned it, so sure, that's done.

jeremiah-c-leary, 2021-02-09:
Hey @staerz ,

I changed README.md to README.rst in the MANIFEST.in file.  I tested it and I no longer see the warning.

staerz, 2021-02-09:
Happy you found the bug.

Since that now addresses the last open point, please feel free to close this issue :+1:

jeremiah-c-leary, 2021-02-09:
Woot!!

closed 2021-02-09 by jeremiah-c-leary