jeremiah-c-leary/vhdl-style-guide #489: Documentation bugs

james-tait, 2020-12-14:
**Environment**
Hosted documentation at time of posting.

**Describe the bug**
A selection of documentation bugs and odd word choices spotted while reading the hosted docs, with suggested fixes.

- [ ] [port_map_006](https://vhdl-style-guide.readthedocs.io/en/latest/port_map_rules.html?highlight=port%20map#port-map-006) doesn't exist but has a title and a link in the documentation
- [ ] [signal_012](https://vhdl-style-guide.readthedocs.io/en/latest/signal_rules.html?highlight=signal_012#signal-012) "assignments" should read "declarations"
- [ ] [generate_016](https://vhdl-style-guide.readthedocs.io/en/latest/generate_rules.html?highlight=generate_016#generate-016) "generic case statement" should read "generate case statement"
- [ ] [generate_17](https://vhdl-style-guide.readthedocs.io/en/latest/generate_rules.html?highlight=generate_016#generate-017) "on generic statement labels" should read "on generate statement labels"
- [ ] [concurrent_008](https://vhdl-style-guide.readthedocs.io/en/latest/concurrent_rules.html#concurrent-008) "sequential concurrent statements" should read "consecutive concurrent statements"
- [ ] [port_014](https://vhdl-style-guide.readthedocs.io/en/latest/port_rules.html#port-014) "closing parenthesis of the port map are on" should read "closing parenthesis of the port map is on"

jeremiah-c-leary, 2020-12-14:
Hey @james-tait ,

Thanks for finding those documentation issues.

--Jeremy

james-tait, 2020-12-14:
No problem, happy to help. I've checked them and they look fine.

closed 2020-12-14 by james-tait