jeremiah-c-leary/vhdl-style-guide #490: Add rules for function/procedure calls

james-tait, 2020-12-15:
**Is your feature request related to a problem? Please describe.**
Currently, no rules exist for the indents, case, etc. of function/procedure calls. For example:

```vhdl
architecture tb of a_tb is

begin

  main : process is
  begin

    a_sequential <= assignment;

  FUNCT_1(arg);

  end process main;

    funct_2(arg);
  a_concurrent <= assignment;

end architecture tb;
```

**Describe the solution you'd like**
Implementation of rules for indentation similar to those for sequential and concurrent statements. Likewise case rules similar to function_017 but in the call rather than the declaration.

So the above example would become:

```vhdl
architecture tb of a_tb is

begin

  main : process is
  begin

    a_sequential <= assignment;

    funct_1(arg);

  end process main;

  funct_2(arg);
  a_concurrent <= assignment;

end architecture tb;
```

jeremiah-c-leary, 2020-12-15:
Hey @james-tait ,

I just pushed some rules to address procedure calls.

Let me know if the latest master fixes the issue for you.

Thanks,

--Jeremy

james-tait, 2020-12-15:
Terrific, I'm afraid it won't be right away but I will try to do that.

james-tait, 2020-12-24:
Thanks for your patience, I've now tested the files on master commit 0e4f4a4dca4a561c8df2ac49a001b1f4b8abbe17, I'm happy with the result.

closed 2020-12-24 by james-tait