jeremiah-c-leary/vhdl-style-guide #491: Add option to `library_003` to allow library "blocks"

staerz, 2020-12-17:
**Is your feature request related to a problem? Please describe.**
Feature request related to personal style preference.

**Describe the solution you'd like**
[`library_003`](https://vhdl-style-guide.readthedocs.io/en/latest/library_rules.html#library-003) checks for blank lines above the `library` keyword.

That is totally fine in the case where a single library is used or multiple libraries are used with a "use" keyword.

It may though be sufficient to simply call multiple libraries without the `use` keyword.

Such a block would look like this:
```vhdl
-- empty line first

library lib_a;
  use lib_a.something;

library lib_b;

library lib_c;

library lib_d;
```

I propose an option to that rule that the line above the `library` keyword may also be `library` (and hence doesn't have to be blank, similar to checking for a comment).
The above example would hence look like this:
```vhdl
-- empty line first

library lib_a;
  use lib_a.something;

library lib_b;
library lib_c;
library lib_d;
```

**Describe alternatives you've considered**
Disabling this rule.

jeremiah-c-leary, 2020-12-17:
Hey @staerz ,

I just pushed an update to this issue.  When  you get a chance, could you validate it?

Thanks,

--Jeremy

staerz, 2020-12-17:
Awesome! Works, thanks for implementing this so super quickly!

jeremiah-c-leary, 2020-12-17:
>  Awesome! Works, thanks for implementing this so super quickly!

Sweet...I have some free time until January and I am pretty bored.

closed 2020-12-17 by jeremiah-c-leary

staerz, 2020-12-17:
> Sweet...I have some free time until January and I am pretty bored.

Haha, so somebody tell me that COVID-19 is not good for at least something :rofl:

(Well, and also I'm now "free" for reviewing my code with the new VSG options ... still a long way, but you see the issues that I open ... :wink:)

jeremiah-c-leary, 2020-12-17:
> Haha, so somebody tell me that COVID-19 is not good for at least something

Agree.  I definitely like working from home.  My setup is so much better than work.  I don't think I want to go back.

> Well, and also I'm now "free" for reviewing my code with the new VSG options ... still a long way

Please keep them coming.  There is only so much code I can look at and it all starts to blend together.