jeremiah-c-leary/vhdl-style-guide #492: Prefix rules only work correctly if they are lowercase

joshrsmith, 2020-12-17:
**Environment**
Running from master circa https://github.com/jeremiah-c-leary/vhdl-style-guide/commit/e80c16da1518e18fd9ba67f1e884f6da486bdfc1 

Running on linux.

**Describe the bug**
I am testing out my (slightly updated) configuration of VSG with latest from master. I am seeing some regressions related to prefix handling. For example, in my configuration file I have generic_020 defined with a single prefix. But I get error messages like this: "Prefix G_MY_GENERIC with one of the following: G_" Where G_ is my configured prefix. This seems to be happening for generic_020, port_011, constant_015, and type_015 rules. 

I output the configuration file via -oc and that looks correct there. So it is reading the setting OK.

OK, seems like it only works if the prefixes are lowercase. If they are uppercase it flags an error. So if I changed my configuration setting to prefixes=['g_'] and also update the code to match, it works as expected.

This is a regression compared to 2.2.0 when this rule worked fine with uppercase prefixes.

**To Reproduce**
"generic_020": {
   "disable": false,
   "prefixes": ["G_"],
}

Use uppercase generic in a generic declaration, e.g. `G_MY_GENERIC`.

**Expected behavior**
Upper and lowercase prefixes should both be allowed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

jeremiah-c-leary, 2020-12-17:
Hey @joshrsmith ,

I will take a look at this.

jeremiah-c-leary, 2020-12-17:
It looks like I forgot to lowercase the prefixes before I did my comparison.  I update all the prefix and suffix rules to allow upper or lower case.

If you have time, could you validate the fix?

Thanks,

--Jeremy

joshrsmith, 2020-12-17:
Yes, you fixed it. Thanks.

closed 2020-12-17 by joshrsmith