jeremiah-c-leary/vhdl-style-guide #493: Concurrent_002 doesn't like "with ... select"

staerz, 2020-12-17:
**Environment**
VSG 3.0.0 release candidate 86ae2434

**Describe the bug**
VSG throws the error
```
  concurrent_002            | Error      |        212 | None
```

on the following code:
```vhdl
  with counter select avst_stream1_i.valid <=
    '1' when 50 to 55,
    '0' when others;
```
Note that the line 212 is `with ... select`.

Apparently to check for this rule in the context of `with ... select` is not useful.

I propose that this rule (check) should not be applied here.

**To Reproduce**
Steps to reproduce the behavior:
1. Run VSG on the example
2. See error

**Expected behavior**
I wouldn't expect this error be thrown here.

jeremiah-c-leary, 2020-12-17:
The rule should allow for the assignment operator to be on a different line than the `selected_waveforms`.

I think I see where the issue is.

jeremiah-c-leary, 2020-12-17:
Hey @staerz ,

I just pushed a fix for this issue.  Could you confirm it works on your side?

Thanks,

--Jeremy

staerz, 2020-12-17:
Hi @jeremiah-c-leary,

yes, it works for the `with` statement now (also if it is within a block, I did though not check multiple block levels).

The issue still remains for the other case:
```vhdl
  avst_streams_int(i).data <=
    (others => '1') when bcid = BCID_ALTER and bcr_locked = '1' else
    avst_streams_i(i).data;
```

Not sure if you meant to have fixed that as well, since I added this to the bug report only later...

staerz, 2020-12-17:
I updated the reply, should I open a dedicated issue for `concurrent_009`?

jeremiah-c-leary, 2020-12-17:
> I updated the reply, should I open a dedicated issue for concurrent_009?

Yes please.  It helps keep the issues focused on a single issue.

Thanks,

--Jeremy

jeremiah-c-leary, 2020-12-18:
Can we close this issue?

staerz, 2020-12-18:
Absolutely. I opened #499 for concurrent_009 ...

closed 2020-12-18 by staerz