jeremiah-c-leary/vhdl-style-guide #494: Add a rule instantiation_xxx for checking prefix/suffix for label

staerz, 2020-12-17:
**Is your feature request related to a problem? Please describe.**
Related to the problem of "consistency".

**Describe the solution you'd like**
Some elements allow labels and for most of the labels [rules for suffix (inclusive) or prefix](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_prefix_suffix.html#rules-enforcing-prefixes-and-suffixes) exist, but not for instantiations.

Being able to enforce a prefix or suffix also for instantiations would improve the consistency with existing rules.

**Describe alternatives you've considered**
None.

jeremiah-c-leary, 2020-12-17:
Hey @staerz ,

Checking for prefixes and suffixes on instantiation labels makes sense.  We usually use `U_` for a prefix.

I just pushed an update for this feature request.  If you could validate sometime I would appreciate it.

Thanks,

--Jeremy

staerz, 2020-12-17:
Alright, I'll have a look. I usually use `inst_` - but since the prefix/suffix can be configured, the default doesn't actually matter too much.

jeremiah-c-leary, 2020-12-17:
I think I should make the default `inst_`.  At least it would be consistent with the suffix rule.

jeremiah-c-leary, 2020-12-17:
I just pushed an update to default rule 601 to `inst_`.

staerz, 2020-12-17:
And I think then you should also update 600 to be `_inst` by default - and not `_u` ...
(I see you opted for skipping the count to the 600 ... nothing that I particularly like, but your choice ...)

jeremiah-c-leary, 2020-12-17:
> And I think then you should also update 600 to be _inst by default - and not _u ...

They are now both set to `_inst` and `inst_` by default.

>  I see you opted for skipping the count to the 600 ... nothing that I particularly like, but your choice ...

Yeah...It is a lot easier for me to create new rules if they are categorized like this.

staerz, 2020-12-17:
I think you missed to update the doc for 601.

jeremiah-c-leary, 2020-12-19:
Good catch.  I just pushed an update to the documentation.

staerz, 2020-12-19:
OK, I see it now. I think you can close the issue then :+1:

jeremiah-c-leary, 2020-12-19:
Sweet, thanks.

--Jeremy

closed 2020-12-19 by jeremiah-c-leary