jeremiah-c-leary/vhdl-style-guide #495: Crash/exit when parsing alias_declaration

joshrsmith, 2020-12-17:
**Environment**
Circa https://github.com/jeremiah-c-leary/vhdl-style-guide/commit/e4ffbfa0475ff236ec9745ba1ef9769963e4b884 
Linux

**Describe the bug**

```
Unexpected token detected while parsing alias_declaration @ (Line, Column)
   Expecting : ; 
   Found     : (
```
VSG exists immediately.

**To Reproduce**

Here is the line that is tripping it up:
`alias my_alias : std_logic other_signal(C_CONSTANT'high);`

**Expected behavior**
Parsing would continue without error.

jeremiah-c-leary, 2020-12-17:
I'm on it.

joshrsmith, 2020-12-17:
http://vhdl.renerta.com/mobile/source/vhd00003.htm

The intention of the code is to do something like this - take a slice of an array signal and give it a name

```
signal Instruction : Bit_Vector(15 downto 0);
alias OpCode : Bit_Vector(3 downto 0) is Instruction(15 downto 12);
alias Source : Bit_Vector(1 downto 0) is Instruction(11 downto 10);
alias Destin : Bit_Vector(1 downto 0) is Instruction(9 downto 8);
alias ImmDat : Bit_Vector(7 downto 0) is Instruction(7 downto 0);
```

jeremiah-c-leary, 2020-12-17:
> http://vhdl.renerta.com/mobile/source/vhd00003.htm

Nice, I have that site bookmarked. 

I copied that code snippet down and received the same error you posted.

jeremiah-c-leary, 2020-12-17:
Hey @joshrsmith,

I just pushed an update that should fix this.  If you can double check I would appreciate it.

Thanks,

--Jeremy

joshrsmith, 2020-12-17:
No longer crashing, seems fixed.

closed 2020-12-17 by joshrsmith