jeremiah-c-leary/vhdl-style-guide #496: Unexpected behaviour when using "-of summary" option

staerz, 2020-12-17:
**Environment**
VSG in 93009711

**Describe the bug**
The log is:
```
[staerz@steffenswkdell LASP (VSG3_0)]$ vsg $(cat .vsg/params.txt)
================================================================================
File:  modules/fpga/src/private/fpga_registers.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 440
Total Violations:    0
  Error   :     0
  Warning :     0
================================================================================
File:  modules/fpga/src/private/sctrl_tester.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 316
Total Violations:    4
  Error   :     0
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  instantiation_010         | Error      |        142 | Move => -1 columns
  instantiation_010         | Error      |        144 | Move => -1 columns
  instantiation_010         | Error      |        146 | Move => -1 columns
  instantiation_010         | Error      |        148 | Move => -1 columns
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
================================================================================
File:  modules/fpga/src/private/s10_devkit_top_arch.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 317
Total Violations:    1
  Error   :     0
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  block_401                 | Error      |        419 | Move : -13 columns
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
================================================================================
File:  modules/fpga/src/public/avmm_registers.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 428
Total Violations:    0
  Error   :     0
  Warning :     0
================================================================================
File:  modules/fpga/src/private/fpga_top.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 437
Total Violations:    0
  Error   :     0
  Warning :     0
================================================================================
File:  modules/fpga/simulation/private/fpga_top/fpga_top_tb.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 438
Total Violations:    0
  Error   :     0
  Warning :     0
[... 94 lines cut ...]
File: modules/fpga/simulation/private/fpga_top/fpga_top_tb.vhd OK (438 rules checked) [Error: 0] [Warning: 0]
File: modules/fpga/simulation/private/s10_devkit_top/s10_devkit_top_tb.vhd OK (440 rules checked) [Error: 0] [Warning: 0]
File: modules/fpga/simulation/public/bcid_generator/bcid_generator_tb.vhd OK (440 rules checked) [Error: 0] [Warning: 0]
File: modules/fpga/simulation/public/bcr_to_bcid/bcr_to_bcid_tb.vhd OK (440 rules checked) [Error: 0] [Warning: 0]
File: modules/fpga/context/mapping.vhd OK (437 rules checked) [Error: 0] [Warning: 0]
File: modules/fpga/context/constants.vhd OK (437 rules checked) [Error: 0] [Warning: 0]
File: modules/fpga/context/fpga_if.vhd OK (437 rules checked) [Error: 0] [Warning: 0]
File: modules/fpga/context/fpga_map.vhd OK (437 rules checked) [Error: 0] [Warning: 0]
File: modules/fpga/context/functions.vhd OK (437 rules checked) [Error: 0] [Warning: 0]
File: modules/fpga/context/interfaces.vhd OK (437 rules checked) [Error: 0] [Warning: 0]
File: modules/fpga/context/registers.vhd OK (439 rules checked) [Error: 0] [Warning: 0]
File: modules/fpga/target_hw/devkit/s10_devkit_top_ent.vhd OK (439 rules checked) [Error: 0] [Warning: 0]
```

As one can see from these logs, apparently `instantiation_010` and `block_401` are not counted as error nor warnings.

That looks pretty unexpected to me.

**To Reproduce**
1. Write code not compliant with `instantiation_010` or `block_401`
2. Run VSG on it
3. See error (or not ... depending which output format you choose)

**Expected behavior**
I clearly expect violations also of  `instantiation_010` or `block_401` be counted and reported.

jeremiah-c-leary, 2020-12-17:
That is strange.  I have a dictionary that defines all the violations and statistics of the run.  Then I pass the dictionary to the output formatter.  I would think it would be the same dictionary passed to both formatters.  I will see if I can replicate the issue on my side.

jeremiah-c-leary, 2020-12-18:
Hey @staerz ,

I think my fix for issue #500 fixes this one also.

I created a simple file that throws an `instantiation_010` error:

```vhdl
architecture RTL of FIFO is

begin

  U_FIFO : component FIFO
    generic map (
      g_width => 8,
      g_delay   => 2
    )
    port map (
      wr_en => wr_en,
      rd_en => rd_en,
      overflow => overflow
    );

end architecture RTL;
```
...and this is output I get now:

```mono
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide$ vsg -f issue_validation/issue_496/example.orig.vhd
================================================================================
File:  issue_validation/issue_496/example.orig.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 351
Total Violations:    3
  Error   :     3
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  instantiation_010         | Error      |          9 | Move => -2 columns
  instantiation_010         | Error      |         12 | Move => 3 columns
  instantiation_010         | Error      |         13 | Move => 3 columns
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide$ vsg -f issue_validation/issue_496/example.orig.vhd -of summary
File: issue_validation/issue_496/example.orig.vhd ERROR (351 rules checked) [Error: 3] [Warning: 0]
```

The base rule I use for `instantiation_010` was part of the update for issue #500.

Could you try this again with the latest from master and let me know if the issue is resolved on your end.

Thanks,

--Jeremy

staerz, 2020-12-18:
I have tested again (putting back the errors in place for trial) in these files and now they are reported properly in the summary.
I did though not test any other rule violations ... I'd hope you you cross-checked.

jeremiah-c-leary, 2020-12-18:
I believe I updated all the rules which would result erroneous reporting.

I search for `self.violations.append` and replaced them with the method `self.add_violation`.

I will close this issue and we can re-open it if you run into it again.

closed 2020-12-18 by jeremiah-c-leary