jeremiah-c-leary/vhdl-style-guide #497: Unexpcted token detected when parsing protected type

joshrsmith, 2020-12-17:
**Environment**
Linux

circa https://github.com/jeremiah-c-leary/vhdl-style-guide/commit/a18ab6414d1b076ee418494c423675cc8d4749f4

**Describe the bug**

Error: Unexpected token detected while parsing full_type_declaration
Expecting : ;
Found : protected

Example code where issue gets triggered:

```
type T_FLAG_TYPE is protected        -- protected type declaration
   procedure init (foo : real);
   impure function myfunct return boolean;
end protected T_FLAG_TYPE;
```

**To Reproduce**
Run vsg on a file containing protected type declaration.

**Expected behavior**

It doesn't crash/error

**Additional context**

About protected types: https://www.aldec.com/en/support/resources/documentation/articles/1179

jeremiah-c-leary, 2020-12-17:
@joshrsmith 
I just checked my notes and I did not implement the `protected_type_definition`.  I will work on it.

jeremiah-c-leary, 2020-12-17:
@joshrsmith ,

I just push an update that should parse `protected_type_definition`s.  Please let me know if this solves the issue for you.

Thanks,

--Jeremy

joshrsmith, 2020-12-17:
Got past that issue. Thanks!

closed 2020-12-17 by joshrsmith