jeremiah-c-leary/vhdl-style-guide #498: Add rules for attributes similar to block_401

staerz, 2020-12-17:
**Is your feature request related to a problem? Please describe.**
For consistency it would be a nice-to-have feature.

**Describe the solution you'd like**

The position of a colon for a signal declaration is well defined by `block_401` (or others if not in a block), but no respective rules for attributes seem to exist, so the following is fully OK with the current version of VSG:
```
  signal bcr : std_logic;

  attribute keep of bcr   : signal is true;
```

I would propose a rule to also tie the number of colons here to 1:
```
  signal bcr : std_logic;

  attribute keep of bcr : signal is true;
```

**Describe alternatives you've considered**
Complain to the VHDL developer directly :stuck_out_tongue_closed_eyes: 

**Additional context**
Consistency is key.

staerz, 2020-12-17:
Side note: I guess `block_401` is a `block` rule since it is checked internally in VSG as part of the block rules, but I'd still say it's a signal declaration rule, similar to `signal_006`, so as a user I'd expect this to be called `signal_xxx`. (But in the end it's just names ...)

jeremiah-c-leary, 2020-12-18:
There are currently no `attribute_specification` rules, but I could add some.

> so as a user I'd expect this to be called signal_xxx

So the current philosophy regarding those rules is:
1)  Rules addressing intra-construct will be handled by the rule of the same name.
2)  Rules addressing inter-construct will be handled by the construct utilizing it.

So all the signal rules deal with how to format a `signal_declaration` within the declaration itself.

All rules that handle how parts of a signal relate to other signals, and other constructs like constants and variables, are handled by other rules e.g. `block_400`, `block_401`, `architecture_026`, etc...

The downside is I need duplicate rules that apply to blocks, architectures, processes etc...

staerz, 2020-12-18:
Hi @jeremiah-c-leary,

Yes, adding rules for attributes accordingly would be desired.

And since you say you have "duplicate" rules, well, let's say that's a price one can pay if it works - and your parser reworking is a major improvement as I am not getting any false indentation complaints (at least not so far) now anymore.

One recommendation then maybe: If you could improve the documentation of these highly similar rules by adding crosslinks between these related rules, that would be handy I think. Something like:

Relates to:
- this rule
- that rule

Would be super nice, I think.

jeremiah-c-leary, 2020-12-18:
Hey @staerz ,

I just pushed some additional rules for attributes.

Could you check them out when you get a chance?

Thanks,

--Jeremy

staerz, 2020-12-18:
Hi @jeremiah-c-leary,

pulled and ran again - et voilà - ran into some right away. Nice.
They seem to work nicely (as I usually formatted attributes like signals) and only got a few complaints where I really forgot alignment of the colon ([entity_specificaton_101](https://vhdl-style-guide.readthedocs.io/en/latest/entity_specification_rules.html#entity-specification-101) was thrown correctly).

Just a minor remark on the doc of that rule: there you literally type ":" while usually you write "colon" - I'd prefer the latter as the current doc looks like odd punctuation...

jeremiah-c-leary, 2020-12-19:
>  Just a minor remark on the doc of that rule: there you literally type ":" while usually you write "colon"

yeah....consistency isn't my strong suit.

I changed `:` to `colon` in the documentation.

I am assuming I can close this issue.

closed 2020-12-19 by jeremiah-c-leary