jeremiah-c-leary/vhdl-style-guide #499: Reformat concurrent_009

staerz, 2020-12-17:
**Is your feature request related to a problem? Please describe.**
It's a nice-to-have feature request, related to the current rule not fitting my expectation.

**Describe the solution you'd like**
The current rule format on conditional signal assignments is
```vhdl
  avst_streams_int(i).data <= (others => '1') when bcid = BCID_ALTER and 
                                                   bcr_locked = '1' else
                              avst_streams_i(i).data;
```
I find this highly unreadable.

I would much prefer the following formatting, which, in my eyes, much better meets the existing indentation standards checked by VSG:
```vhdl
  avst_streams_int(i).data <=
    (others => '1') when bcid = BCID_ALTER and bcr_locked = '1' else
    avst_streams_i(i).data;
```
In this formatting each conditional assignment is on it's own line (containing the `else` keyword, indented once (2 spaces in the current indentation settings).

One could further argue about possible line breaks for long conditions:
```vhdl
  avst_streams_int(i).data <=
    (others => '1') when
      bcid = BCID_ALTER and
      bcr_locked = '1' else
    avst_streams_i(i).data;
```
In the example above, another level of indentation would be added. 

This might even be repeated if parenthesis would be used to express more complex logical combinations:
```vhdl
  avst_streams_int(i).data <=
    (others => '1') when
      bcid = BCID_ALTER and
      (
        bcr_locked = '1' or
        another_signal = '0'
      ) else
    avst_streams_i(i).data;
```
(The opening parenthesis could also be allowed at the end of the previous line, but if that complicates the parsing, I'd be fine with that.)

**Describe alternatives you've considered**
Disabling this rule.

staerz, 2020-12-17:
A similar proposal would hold true for `concurrent_03`.

jeremiah-c-leary, 2021-02-06:
Hey @staerz ,

I wonder if we could use #450 as a template for this.  Here are some possible options:

ignore_single_line = ignore assignments that are not multiline.
first_expression_new_line = move code after => to a new line
align_left = move code to left, with proper indent
single_line_expression_when_condition_else = keep the expression, when keyword, condition and else keyword on a single line
align_when_keywords = keeps all when keywords aligned
wrap_condition_at_when = if a condition wraps to new line, then align with when keyword.  otherwise align left with proper indent.

I will create another branch and write some configuration documentation that we can review before I start implementing.

jeremiah-c-leary, 2021-02-07:
Hey @staerz ,

I just pushed some documentation for some options to this rule under docs/configuring_conditional_expressions on the issue-499 branch.  Could you review the file and let me know what you think?

Thanks,

--Jeremy

staerz, 2021-02-17:
Hi @jeremiah-c-leary,

I found a notification on that issue which I didn't react on in my inbox.

I'd say let's conclude #450 first and then we focus on this one. Generally speaking, yes, I think it will be possible to use #450 as a template for hear as well. But let's do one thing at a time (properly).

jeremiah-c-leary, 2021-02-18:
> I'd say let's conclude #450 first and then we focus on this one. Generally speaking, yes, I think it will be possible to use #450 as a template for hear as well. But let's do one thing at a time (properly).

I completely agree.  Let's learn everything we can from #450 before tackling this one.

jeremiah-c-leary, 2021-03-04:
Hey @staerz ,

I was going to push this issue to release 3.1.0 so I can get 3.0.0 out the door.

--Jeremy

staerz, 2021-03-04:
Hi @jeremiah-c-leary,

well, that's your decision, of course, but I had hoped that we would have addressed it similarly to what we did #450.

And I think it is very similar, but if you feel pressed, well, who am I to tell you what to do :wink:

jeremiah-c-leary, 2021-03-05:
Hey @staerz ,

It looks like that is your last submitted issue.  I can switch it back to 3.0.0.

I just pushed a documentation update and will start on updating tests.

Could you review the file `vsg/docs/configuring_conditional_expressions.rst` on the `issue-499` branch?  It lists the options I am currently thinking of implementing.  I think they would cover your preferred style.

I we want to add returns after parenthesis, then I would create another rule to handle the structure.

staerz, 2021-03-06:
Hi @jeremiah-c-leary,

after having rebased the branch to master I was able to use `git diff` to see your additions.

I agree to them, looks good.

There is one point that you are missing: In my examples above, I'm always having a line break after the assignment operator (`<=`). This is, of course, again personal preference, but I think it adds readability.

I think you could simply cover this by yet another option "wrap_at_assignment" (or so).

jeremiah-c-leary, 2021-03-08:
Hey @staerz ,

This weekend was fairly productive:

  1)  Updated the base rule of `concurrent_003` to allow for aligning left and aligning based of parenthesis.  This is covered by the `Configuring Multiline Indent Rules` documentation.  I also added which rules it applies to.
  2)  Finished `concurrent_009` with all options implemented and tested.
       a)  Removed option `ignore_single_line` as there is nothing to align if everything is on a single line.

I just pushed an update for this issue with the `concurrent_003` and `concurrent_009` updates.  

> There is one point that you are missing: In my examples above, I'm always having a line break after the assignment operator (<=). This is, of course, again personal preference, but I think it adds readability.
>
> I think you could simply cover this by yet another option "wrap_at_assignment" (or so).

I will make another rule to handle the structure of multiline concurrent statements.

jeremiah-c-leary, 2021-03-08:
Hey @staerz ,

I add rule `concurrent_011` to handle structural changes to concurrent assignments.  I should handle line break after the assignment operator.

I just pushed the update.

When you get a chance, could you check out the updates.

Thanks,

--Jeremy

jeremiah-c-leary, 2021-03-08:
Hey @staerz ,

I found an indenting issue when if you use `concurrent_011` to push the first expression to the next line.  I realized none of my tests include this style.  I will add it to my tests and update `concurrent_009`.

staerz, 2021-03-08:
Hi @jeremiah-c-leary,

I pulled and tested the new feature.

We're getting there I'd say, but it's a bit tricky now to see if I get my configuration right: Is there a way I can see the documentation on readthedocs on your issue-branch?

I think when you give the example documentation for `concurrent_009`, you actually typed `constant_009`.

Also, like we have it for the other rules, where a configuration (e.g. blank line) may apply to certain rules, it would be very helpful to have the list of rules for which this configuration applies also on the page of the configuration. It seems we have also forgotten this for the previous issue on `constant_016` (et al.).

And yes, I think I see the indentation issue with `concurrent_009`.

jeremiah-c-leary, 2021-03-10:
Hey @staerz ,

> We're getting there I'd say, but it's a bit tricky now to see if I get my configuration right: Is there a way I can see the documentation on readthedocs on your issue-branch?

I did that once, and it was a bit of pain.  I would rather wait until I push this to master.  Do we feel it is close enough to push to master?

> I think when you give the example documentation for concurrent_009, you actually typed constant_009.

Yeah...that looks like a copy paste error.  That should be fixed on my latest push.

> Also, like we have it for the other rules, where a configuration (e.g. blank line) may apply to certain rules, it would be very helpful to have the list of rules for which this configuration applies [...]

I added the rules for what we are working on now.  I can update the others on  master.

I just pushed an update that fixes the indention issue.  It seems to work well on my end.  I am curious how it works on yours.

--Jeremy

staerz, 2021-03-10:
Hi @jeremiah-c-leary,

I pulled the latest update and I think we have the typical +-1 problem:
```
  concurrent_003            | Error      |        103 | Adjust indent to column 5
```
My indentation is by 2 each time, so it should be 6 here :wink:

The code is this one:
```
    mregs_ro_fpga_firmware <=
    (
      1 => to_multireg(G_FPGA_LASP_FW_VER_TAG, FPGA_REGS_FIRMWARE_VERSION_TAG_BLK_BADDR),
      2 => to_multireg(std_logic_vector(G_FPGA_LASP_FW_VER_SHA), FPGA_REGS_FIRMWARE_VERSION_SHA_BLK_BADDR),
      3 => to_multireg(fpga_uptime, FPGA_REGS_UPTIME_REGS_BADDR),
      4 => to_multireg(fpga_downtime, FPGA_REGS_DOWNTIME_REG_BADDR),
      others => MREG_EMPTY
    );
```
(and 103 is `1 => ...` and, of course, it also complains about the 4 following lines)

Given that this is a minor fix, I think you could date to push your next update to master, which would make it easier for me to check the documentation as well (you know, the links etc. ...)

jeremiah-c-leary, 2021-03-10:
Hey @staerz 

> I pulled the latest update and I think we have the typical +-1 problem:

Yeah...I do not think I used `indentSize`.  I will update `concurrent_003` and then merge it down to master.

How are the `concurrent_009` updates working for you?

staerz, 2021-03-11:
Hi @jeremiah-c-leary,

I just tested for `concurrent_009` and it looks (so far) very good.

I'd go for a more complex testing (on larger fractions of our code base once I can double-check with the documentation from master, but the basics look good so far (it's just to catch any strange edge cases) :wink:

jeremiah-c-leary, 2021-03-12:
Hey @staerz ,

I think I made `concurrent_003` better and I merged this branch down to master.

--Jeremy

staerz, 2021-03-12:
OK, now I can check the documentation.

The link to the configuration in [concurrent_009](https://vhdl-style-guide.readthedocs.io/en/latest/concurrent_rules.html#concurrent-009) and also concurrent_011 are broken.

But before you fix that, I would suggest a different title (the title is very long!): Why not simply "Configuring concurrent structure" and "Configuring concurrent alignment"?

Configuring 009:
- "wrap_at_when":  Is the option "wrap_at_when" or "wrap_condition_at_when"? The documentation uses both.
- "align_paren" appears in the table, but not in the example rules
- "ignore_single_line" appears in the example rules, but not in the table

jeremiah-c-leary, 2021-03-12:
Hey @staerz ,

I just pushed the documentation updates and I believe I addressed the above issues.

Let me know what you think.

--Jeremy

staerz, 2021-03-15:
Hi @jeremiah-c-leary,

first a few notes on documentation:

From the previous issue, but for completeness, I suggest you also add the list of rules which apply for [Configuring Multiline Structure Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#configuring-multiline-structure-rules).

[Alignment Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#configuring-concurrent-alignment-rules): I think there's a typo:
"Each option allows one two values: ‘yes’ and ‘no’." &rarr; "Each option allows one of the two values: ‘yes’ and ‘no’."

Also, I'd capitalise the rule description in the table (like is done for all others):
- "each when keyword will be aligned" &rarr; "Align each 'when' keyword."
- "each else keyword will be aligned" &rarr; "Align each 'else' keyword."
- "Indent multiline condition at when keyword." &rarr; "Indent multiline condition at 'when' keyword." (quotes)
- also watch out your punctiation: each option description should end with a period (dot, '.') as is done with other tables

The [Example: align_paren ‘yes’ and align_left ‘no’](https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#example-align-paren-yes-and-align-left-no) (fixed code) looks really odd to me. Not that I would mind (I'd use a different setting), but just to double check with you if this is intended.

Then, when I ran over a different part of our code base I ran into this:

```
  entity_015                | Error      |         59 | Add entity keyword
  architecture_010          | Error      |         85 | Add *architecture* keyword.
```
I think consistency is key, so, I know, it's nothing to do with the current issue (maybe you want to create a dedicated one, very well for 3.1.0) but I think you should make a consistency check for the keyword highlighting. And I would go with simple quotes instead of star.

For functionality, I'd say we're good. The features that I requested work as expected. I did though not try all different combinations of settings.

jeremiah-c-leary, 2021-03-16:
Hey @staerz ,

> first a few notes on documentation

I think I need a documentation similar to VSG.  😄 

I pushed a version with the documentation updates.

> Then, when I ran over a different part of our code base I ran into this.

I opened issue #538 to cover the reporting of the keyword.

> The Example: align_paren ‘yes’ and align_left ‘no’ (fixed code) looks really odd to me. Not that I would mind (I'd use a different setting), but just to double check with you if this is intended.

I double checked it, and it looks correct.  Thank you for the thorough review of the documentation.

> For functionality, I'd say we're good. The features that I requested work as expected. I did though not try all different combinations of settings.

That is awesome.

Please let me know if the documentation updates are working.

--Jeremy

staerz, 2021-03-16:
Hi @jeremiah-c-leary,

I checked it again and the documentation now looks good to me, I didn't find anything weird now.

From my side, I'd say you can close the issue now. Thanks for implementing this!

jeremiah-c-leary, 2021-03-16:
Hey @staerz ,

> From my side, I'd say you can close the issue now. Thanks for implementing this!

awesome!!!

I will start getting 3.0.0 ready for release.

closed 2021-03-16 by jeremiah-c-leary