jeremiah-c-leary/vhdl-style-guide #500: length_001 cannot be temporarily disabled

staerz, 2020-12-17:
**Environment**
VSG efda079b

**Describe the bug**
Temporarily disabling `length_001` via [code tags](https://vhdl-style-guide.readthedocs.io/en/latest/code_tags.html) seems not to work.

**To Reproduce**
Steps to reproduce the behavior:
1. Have a code line to exceed the default 120 characters
2. Add `-- vsg-disable-next-line length_001` directly above that line or enclose this line in `-- vsg_off` and `-- vsg_on`
3. Run VSG on that file
4. Still see error

**Expected behavior**
I wouldn't expect the error if it's explicitly disabled (locally).

Note: Disabling the rule per file or globally works.

**Additional context**
Additional note: The code tags `vsg-disable-next-line` and `vsg_off` and `vsg_on` use different character separators: Underscore vs. dash. I believe VSG 3.0.0 would be a good occasion to move to a consistent separator, e.g. make it `vsg_disabe_next_line` (one word as per posix) or `vsg-off` and `vsg-on` (multiple words).

jeremiah-c-leary, 2020-12-18:
Hey @staerz ,

I think I fixed this one.  Turns out not all the rules were using a method which enforced the code tags.

Please let me know if this update fixes the issue for you.

Thanks,

--Jeremy

staerz, 2020-12-18:
Yes, confirmed, it works now.
Thanks also for renaming the next line option :+1:

jeremiah-c-leary, 2020-12-18:
I really appreciate you help in running this against your code.

closed 2020-12-18 by jeremiah-c-leary