jeremiah-c-leary/vhdl-style-guide #501: Rule similar to signal_005 in generate environment

staerz, 2020-12-18:
**Is your feature request related to a problem? Please describe.**
No problem other than a not covered case (as far as I can see).

**Describe the solution you'd like**
I would expect a rule similar to `signal_005` (or more precisely `block_401`) to also exist for `generate` environment but there seems not to be any: Signals, constants etc. can also be defined locally in a generate environment and I would expect that the very same rules apply to indentation and where to put the colon as when defining them at the top level architecture or any block environment.

**Describe alternatives you've considered**
None.

**Additional context**
I like consistency a lot :wink:

jeremiah-c-leary, 2020-12-19:
Hey @staerz ,

I just pushed some rules to handle alignments in generates.

When you get a chance could you see if it works on your code.

Thanks,

--Jeremy

staerz, 2020-12-22:
Hi @jeremiah-c-leary,

so when I run with the commit for this issue, VSG crashes on a certain file with the following:

```
Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg===master', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/__main__.py", line 418, in main
    oRules.check_rules(bAllPhases=commandLineArguments.all_phases, lSkipPhase=commandLineArguments.skip_phase)
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/rule_list.py", line 215, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/rules/align_tokens_in_region_between_tokens_when_between_tokens_unless_between_tokens.py", line 63, in analyze
    lToi = oFile.get_tokens_bounded_by_token_when_between_tokens(self.left_token, self.right_token, self.lBetween[0], self.lBetween[1])
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/vhdlFile/vhdlFile.py", line 164, in get_tokens_bounded_by_token_when_between_tokens
    return extract.get_tokens_bounded_by_token_when_between_tokens(oLeft, oRight, oStart, oEnd, self.lAllObjects, self.oTokenMap, include_trailing_whitespace)
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/vhdlFile/extract/get_tokens_bounded_by_token_when_between_tokens.py", line 14, in get_tokens_bounded_by_token_when_between_tokens
    if iStart < iLeft and iRight < iEnd:
TypeError: '<' not supported between instances of 'NoneType' and 'int'
```
The file where it crashes on doesn't actually contain the `generate ... signal` section, so I'm really wondering what's going on here.

jeremiah-c-leary, 2020-12-22:
Hey @staerz ,

Does it have a generate, but without a signal?

--Jeremy

staerz, 2020-12-22:
Alright, for your debugging, let me here put the snippets of the file that have the `generate` keyword:

```vhdl
architecture abc of def is
  signal a : std_logic;
begin
  ena_ordering : if G_PACKER_ENABLE generate
    inst_ordering : entity packer.ordering_top
    port map (
      clk    => clk,
      rst    => rst,
      din_i  => din_i,
      init_i => init_i,
      sel_o  => sel_int
    );
  else generate
    sel_int <= (others => '0');
  end generate;

  set_sel_o : if OUTPUT_INVERT generate
    sel_o <= not sel_int;
  else generate
    sel_o <= sel_int;
  end generate;

  blk_bcid_gen : block
    signal b : std_logic;
  begin

    gen_empty_bits_bcid : if log2ceil(BCID_DATA_CNT) - 1 < bcid_o.data'high generate
      bcid_o.data(bcid_o.data'high downto log2ceil(BCID_DATA_CNT)) <= (others => '0');
    end generate;

  end block;

  blk_avst_alterer : block
  begin

    alter_data_gen : for i in avst_streams_int'range generate
      avst_streams_int(i).data <=
        (others => '1') when bcid = BCID_ALTER and bcr_locked = '1' else
        avst_streams_i(i).data;

      avst_streams_int(i).valid <=
        '0' when time_slot = TIME_SLOT_ALTER and bcr_locked = '1' else
        avst_streams_i(i).valid;
    end generate;

  end block;
```

jeremiah-c-leary, 2020-12-22:
That commit added 6 rules:  generate_400, generate_401, generate_402, generate_403, generate_404 and generate_405.  Could you isolate which one is failing?

jeremiah-c-leary, 2020-12-22:
Just validated it failed.  Investigating....

jeremiah-c-leary, 2020-12-22:
Hey @staerz ,

I pushed a fix for the above issue.  Your snippet does not crash anymore.

--Jeremy

staerz, 2020-12-22:
Alright, that seems to work now, I had a few complaints about `generate_402` and `generate_403` now.

Looking at the doc, I found a typo:
- [`generate_403`](https://vhdl-style-guide.readthedocs.io/en/latest/generate_rules.html#generate-403): I think the double dot should be replaced by one dot and a line break :wink:
- Same for `generate_401` and `generate_405`

jeremiah-c-leary, 2020-12-22:
Document updates pushed, may take a few minutes before they appear.

staerz, 2020-12-22:
Alright, I see it's updated, great! Another one to close :+1:

jeremiah-c-leary, 2020-12-22:
Woot!!

closed 2020-12-22 by jeremiah-c-leary