jeremiah-c-leary/vhdl-style-guide #502: Comment_010 gets indentation wrong

staerz, 2020-12-18:
**Environment**
VSG 2f3d4c09

**Describe the bug**
See the following context definition, where (on purpose) a use case is commented out.
```vhdl
--! full list of LASP interfaces
--! inherits from LASP_constants
context interfaces is

  library fpga;
    context fpga.constants;
    use fpga.fpga_if.all;
    -- use fpga.registers.all;
    use fpga.functions.all;

end context interfaces;
```

The following error is thrown:
```
  comment_010               | Error      |         17 | Indent level 1
```
When trying to do what is suggested (remove one level of indentation), the following (expected) error is thrown:
```
  library_009               | Error      |         17 | Align comment with *use* keyword.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run VSG with `commen_010` enabled on the code snippet above
2. See error

**Expected behavior**
I would not expect an error here.

**Additional context**
Looks like the indentation (new use/context) hasn't been propagated to `comment_010`?

jeremiah-c-leary, 2020-12-18:
That is an interesting one.  I will take a look.

jeremiah-c-leary, 2020-12-18:
Hey @staerz ,

I think I fixed the indenting issue.  Could you double check?

Thanks,

--Jeremy

staerz, 2020-12-18:
The fix works for me, the complaint is gone now - and if I do it wrong intentionally (and direction) I get a complaint, just like it should be - well done! Thanks.

jeremiah-c-leary, 2020-12-19:
Awesome, the solution was easier than I thought it would be.  Which made me suspicious it would not work.

closed 2020-12-19 by jeremiah-c-leary