jeremiah-c-leary/vhdl-style-guide #503: Make VSG robust against preprocessor statements

staerz, 2020-12-19:
**Is your feature request related to a problem? Please describe.**
It's a nice to have feature. It would make obsolete the use of `-- vsg_off` and `-- vsg_on` in some cases - and in general improve the usage of VSG in general.

**Describe the solution you'd like**
VHDL can be intersected with preprocessor commands, i.e. to define a given group of ports or generics only upon certain conditions (not visible to VHDL directly).

A typical use case it the following:

```vhdl
    -- vsg_off
    #ifdef G_I2C_ENABLE
    -- vsg_on

    --! @name Global I2C, controlled by G_I2C_ENABLE
    --! @{

    --! Global I2C, clock
    lt_io_scl           : inout std_logic;
    --! Global I2C, data
    lt_io_sda           : inout std_logic;
    --! @}

    -- vsg_off
    #endif
    -- vsg_on
```

In this example, the two ports `lt_io_scl` and `lt_io_sda` are only present if a constant `G_I2C_ENABLE` is set (via the preprocessor).

As can be seen, currently, the preprocessor commands must be enclosed in `-- vsg_off` and `vsg_on` in order to comply with VSG.
This is certainly already a good feature in general, but VSG could be able to self-detect these preprocessor commands and simply ignore them.

**Describe alternatives you've considered**
Enclosing in `-- vsg_off` and `vsg_on`.

**Additional context**
Doxygen support preprocessor commands. It needs to be configured, but can deal with it.

It would be nice if VSG would also at least not get confused by these preprocessor commands.

jeremiah-c-leary, 2020-12-19:
Hey @staerz ,

Where are the preprocessor commands defined?

I looked in the LRM, and the only occurrence I found was in a c header file under the VHPI section.

staerz, 2020-12-19:
Hi @jeremiah-c-leary,

well, I think it depends the preprocessor that is used - and that could be anything.

I think the simplest thing is to base yourself on the definition of what a preprocessor command is: It starts with `#` - this is clearly not part of VHDL definition, so clearly something "else".

^ and yes, that's exactly the point. We have a file that is VHDL with some preprocessor statements, this preprocessor then parses it and builds a clean VHDL file (according to the preprocess statements) and this is then injected to Quartus or what ever.

But the original source file that VSG runs on, is still this file that contains the preprocessor statements.

jeremiah-c-leary, 2020-12-19:
I could see the possibility of adding a preprocessor option into the parser to detect the start of one, but how do I know when the preprocessor ends?

Also, can the preprocessor command be anywhere in the code, or are they restricted?

Does my parser make it through the file with the pre-processor commands?  I would think it would croak on them.

jeremiah-c-leary, 2020-12-20:
Interesting...I see my parser ignores anything at the `design_unit` level that is not a `context_clause` or a `library_unit`.

In my example, I did not need to put the `-- vsg_off` or the `-- vsg_on` around the `#ifdef`.  However, once inside a `context_clause` or a `library_unit` the parser will error out.

staerz, 2020-12-21:
Hi @jeremiah-c-leary,

in my opinion, your parser should simply completely ignore the preprocessor commands.

Of course, now one could also check if they are correctly indented or what ever - but how do you define that!?

The point of the preprocessor is that this is one level (in another dimension if you want so) above VHDL, and VSG parses VHDL, not anything above that. My approach would be to really filter any line matching "^\s*#" away.

So I'm not fully sure to understand your last comment ...

jeremiah-c-leary, 2020-12-21:
>  My approach would be to really filter any line matching "^\s*#" away.

That seems easy enough, but I wonder what would happen if there are directives buried in processes and how the other rules would react.

I can run some experiments.

staerz, 2020-12-21:
Well, "filter **any** line matching "^\s*#" away" should do the trick.

Again, sure, you can worry about what the proprocess would do to the files, but that's in fact out of scope of VSG if you ask me. 
What ever you preprocess in or out, it would still have to be compliant with the current rules, right? So simply filtering them away rigorously, no matter in which scope they are found, gets rid of the problem. My opinion.

jeremiah-c-leary, 2020-12-21:
> So simply filtering them away rigorously, no matter in which scope they are found, gets rid of the problem.

My parser does ignore whitespace and comments.  I would just need to detect and classify the preprocessor rules.  The `#` is a valid VHDL token, so I would need to detect more than just `^\s*#`.

staerz, 2020-12-21:
How is `#` a valid VHDL token? I cannot think of any context where it would be but it seems I'm missing something, could you give an example?

jeremiah-c-leary, 2020-12-21:
It is defined in the basic character set, section 15.2 of the LRM:

[image]

and is used when defining the base of a number in section 15.5.3:

[image]
[image]

staerz, 2020-12-21:
I see, yes, I was thinking in that direction as well, but I never use that syntax.

The difference is though that a preprocessor command `#` must be the first non-white space character of a line.

In VHDL, you'd always find the `#` after another alphanumerical character, so I'd say my proposed "filter" is still valid.

jeremiah-c-leary, 2020-12-21:
> Now if we define a block comment as "blank; (at least 3) comments; blank", then we would also be compatible with all the other rules.

I don't believe the literals can be split across lines, so I would agree.

staerz, 2020-12-21:
I think you start mixing issues :wink:

But yes, I think splitting these literals is invalid.

jeremiah-c-leary, 2020-12-21:
Hey @staerz ,

I just pushed an update for this issue.  I have no tests for it, but it appears to be working.  Could you check it out.  Thanks,

I essentially search for a line starting with `^\s*#` and then classify the entire line as a preprocessor token.  The parser can now get through an entire file without throwing an error.  Preprocessor lines may still affect some rules.

Thanks,

--Jeremy

staerz, 2020-12-21:
Yes, I confirm that now the offending file is not throwing `instantiation_001` anymore.

What do you mean by "Preprocessor lines may still affect some rules."?

jeremiah-c-leary, 2020-12-21:
>  What do you mean by "Preprocessor lines may still affect some rules."?

There are not rules about the preprocessor token, but I wonder if having a preprocessor command buried somewhere in the code that could cause a rule to not behave the same.  Maybe an alignment rule?

I also found a slight issue I need to fix.
I just ran an example with this snippet:
```vhdl
  1
  2 #ifdef G_I2C_ENABLE
  3 entity ent is
  4 #something
  5   port (
  6 #something
  7     I_PORT : in std_logic;
  8 #something
  9     O_PORT : out
 10 #something
 11      std_logic;
 12 #something
 13     O_PORT2 : out std_logic
 14 #something
 15 );
 16 #something
 17 end entity ent;
 18
```
..and VSG ran through it fine, but ended up loosing the preprocessor commands:
```vhdl
  1
  2
  3
  4 entity ent is
  5
  6   port (
  7
  8     i_port  : in    std_logic;
  9
 10     o_port  : out
 11
 12      std_logic;
 13
 14     o_port2 : out   std_logic
 15
 16   );
 17
 18 end entity ent;
 19
```
I will push an update soon.

jeremiah-c-leary, 2020-12-21:
I just pushed an update to bring back the preprocessor commands.

staerz, 2020-12-22:
Hi @jeremiah-c-leary,

alright, I didn't actually check the fixing of VSG. Good catch.

For me if it doesn't complain anymore, that's already it :wink:

I think we can close this one, thanks!

jeremiah-c-leary, 2020-12-22:
Sweet.

closed 2020-12-22 by jeremiah-c-leary