jeremiah-c-leary/vhdl-style-guide #504: Adjust behaviour for comments (allow_comments configuration)

staerz, 2020-12-23:
**Is your feature request related to a problem? Please describe.**
It is rather an inconsistency on how rules apply, but has likely a personal bias.

**Describe the solution you'd like**
A few rules request a blank line above and have a configuration option `allow_comments`.
This configuration option, as the name suggests, allows that the line before is a comment and not a blank line.

This option effectively is like disabling the rule entirely as then a blank line above is not checked anymore.

Instead of the current behaviour for this option, the first line above the comment which is not a comment should be checked and this should be required to be a blank line.

For example, `entity_003` has this option.
```vhdl
library xyz;

-- A comment describing my_entity
-- And maybe we need some more lines to explain more details.
entity my_entity is
```
This example has an blank line (line 2). In the current behaviour of VSG this blank line can also be removed if the option `allow_comments` is set and VSG won't complain.

Instead, with the option `allow_comments` set, this line 2 should be enforced to be a blank line.

**Describe alternatives you've considered**
There is none :wink:

jeremiah-c-leary, 2021-01-13:
Hey @staerz ,

I have an initial attempt at this feature.  When you get a chance, could you check it out and let me know if it meets your needs?

Thanks,

--Jeremy

staerz, 2021-01-14:
Hi @jeremiah-c-leary,

I updated my local VSG installation to 0bec7656 and ran it over my code base again but couldn't find a difference.

I have tried to violate `architecture_003` and `entity_003`, both have `allow_comments` configured to be true, but I see no effect.

Could you be more specific what I should try?

The particular piece of code I ran over was this one:
```vhdl
--! @cond
library fpga;
  context fpga.interfaces;

--! @endcond
--! Non-locking multiplexer for incoming AVMM master interfaces.
entity avmm_master_mux is
  port (
    --! Input clock
    clk_avmm_i     : in    std_logic;
    --! Reset (sync to clock)
    rst_avmm_i     : in    std_logic;
    --! Incoming AVMM master interfaces
    avmm_masters_i : in    t_avmm_masters;
    --! Outgoing AVMM master interface
    avmm_master_o  : out   t_avmm_master
  );
end entity avmm_master_mux;

--! Clocked implementation of non-locking multiplexer for incoming AVMM master interfaces.
architecture behavioral of avmm_master_mux is
```
vs. this without the blank lines.

Both give the very same result.

jeremiah-c-leary, 2021-01-14:
I added another configuration option:  `blank_above_comment` to give users the flexibility.
```json
{
  "rule": {
    "entity_003": {
      "indentSize": 2,
      "phase": 3,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "allow_comments": false,
      "blank_above_comment": false
    }
  }
}
```
Try setting it to true.

--Jeremy

staerz, 2021-01-14:
Hi @jeremiah-c-leary,

OK, I didn't realise the new configuration parameter.

I haven't tried and would expect it to work then, but honestly, wouldn't you agree that `allow_comments` would be enough? As I tried to lay out in the issue description: Enabling `allow_comments` is like disabling the rule, so I really don't see the usefulness of that configuration parameter, if `blank_above_comment` wouldn't be implied by it.

Please let me know if I'm mistaken and am missing a point here.

jeremiah-c-leary, 2021-01-14:
> Enabling allow_comments is like disabling the rule

I am not sure that is a true statement.  Consider this example with `entity_003`:

```vhdl
--without comments
library ieee;
  use ieee.std_logic_1164.all;
entity fifo is

--with comments
library ieee;
  use ieee.std_logic_1164.all;
--Define entity
entity fifo is
```
In both cases, the check would fail if `allow_comments` is `false`.

The first case would fail if `allow_comments` was `true`.

I suppose you could say with regards to comments, that setting `allow_comments` would be disabling the rule.

I interpret `allow_comments` as:
*  comments are allowed on the line above the entity declaration.

So if they are there then ignore the check, otherwise enforce a blank line.

Further enforcing a blank line above the comment with just `allow_comments` set seems non-intuitive to me.  It seems the check is:

* enforce a blank line above a first comment in consecutive comments above an entity declaration.

Maybe that is where I saw the difference and added the `blank_above_comment` option.

So when both options are enabled, the check is:
*  Enforce blank line above entity declarations unless a comment exists
*  Enforce a blank line above consecutive comments if a comment exists above the entity declaration

staerz, 2021-01-14:
Hi @jeremiah-c-leary,

thanks for laying out your point.

A agree with you that from your above example, the first might be undesired while the second might be allowed.
But if "I stick to the book", and read out again the documentation for `enttity_003` (and the similar rules), they state "This rule checks for a blank line above the entity keyword." and I can clearly not see that in your example.

Again, I fully agree that the above example is a valid choice of formatting, but then these should be 2 separate rules: if you want to "check for a blank line above the entity keyword.", then there really should be a blank line.

Now you certainly remember that it was me to propose that comments should be considered (at least via a configuration), but the general statement of "a blank line above the [the comment above the] entity keyword" should still be true.

What you seem to check with `allow_comments` and `blank_above_comment` is to check that the entity has a comment. But if you ask me, that's a different rule (well reasoned). And I think VSG doesn't yet have such a rule, and I would be the last one being against adding it (I actually am very in favour of it as my developers should document their code!) - or am I missing something? ... but then please add it as a dedicated rule and don't mix it up with the original check for a blank line.

jeremiah-c-leary, 2021-01-15:
> What you seem to check with allow_comments and blank_above_comment is to check that the entity has a comment. But if you ask me, that's a different rule (well reasoned). And I think VSG doesn't yet [...]

Interesting....maybe I misunderstood your original intent.

What about adding a `require_comment` option that would fire in these cases:

```vhdl
-- Violation because no comment detected
library ieee;
entity fifo is

--Violation because comment exists but it is not the first line above the entity
library ieee;
-- Entity declaration

entity fifo is

-- No violation because comment is the first line before
library ieee;
-- Entity declaration
entity fifo is
```
It could then be paired with `blank_above_comment`:
```vhdl
library ieee;

-- Entity declaration
entity fifo is
```
The rule then changes from **blank line above** to more a more generic **line above** rule.  This also gives us freedom to expand the rule in the future.

staerz, 2021-01-15:
I think I would still separate it in 2 rules: require_blank_line and require_comment - or to simplify from "allow_comments" to "require_comments".

Again my point is that I really don't see the purpose of the rule with "allow_comments" enabled in the current situation - enabling this option is like disabling the rule.
So either you go and "allow_comments" and that includes that the lines above may be comments (but comments are not required) and then a blank line or you re-dedicate this to "require_comments" (with the implication that a comment is actually required); or you ignore comments at all in that rule (that should still allow commented lines between the entity and the blank line) and introduce a new set of rules checking for comment on certain key words.

Or, maybe a last option that appears reasonable to me: you'd have exclusive options "allow_comments" and "require_comments": Allow comments would be indifferent towards comments and just check if there's a blank line above entity (or any comment that may directly precede the entity); and the "require_comment" that checks for the existence of a comment **and** the blank line above it.

jeremiah-c-leary, 2021-01-15:
> I think I would still separate it in 2 rules: require_blank_line and require_comment - or to simplify from "allow_comments" to "require_comments".

With two separate rules you would end up with conflicting rules.  So then you would have to disable one.  It can become confusing.  If I have all the possible options in a single rule then I can eliminate the confusion.

> Or, maybe a last option that appears reasonable to me: you'd have exclusive options "allow_comments" and "require_comments":

That would keep the **blank_line_above** concept and include a blank line above comments if `require_comments` is used.
So if `allow_comments` and `require_comments` is not set then the following would be true:
```vhdl
--Rule Violation
library ieee;
--Comment
entity fifo is

--Rule Pass
library ieee;
--Comment

entity fifo is
```
If `allow_comments` is set:
```vhdl
--Rule Violation
library ieee;
entity fifo is

--Rule Pass
library ieee;
--Comment
entity fifo is
```
if `require_comments` is set:
```vhdl
--Rule Violation
library ieee;
entity fifo is

library ieee;

entity fifo is

library ieee;
-- Comment

entity fifo is

--Rule Pass
library ieee;

--Comment
entity fifo is
```

staerz, 2021-01-15:
OK, I can understand why having individual rules might be not such a good idea.

Yes, agreed to your last examples with the exception for the `allow_comments` case.

Again, "This rule checks for a blank line above the entity keyword." - there is no blank line in that example, so I cannot understand how this rule would be satisfied.

To be explicit, for `allow_comments` I would require:
```vhdl
--Rule Violation
library ieee;
entity fifo is

--Rule Pass (1)
library ieee;

--Comment
entity fifo is

--Rule Pass (2)
library ieee;

entity fifo is

--Rule Pass (3)
library ieee;

--Comment

entity fifo is
```

And, of course, we understand that in all these examples `library ieee` is a placeholder for any valid vhdl code as well as the comment might be spread over multiple lines.

A particular edge case would be if a block comment would be added. This should be compliant with `allow_comment` as well as `require_comment` options.

jeremiah-c-leary, 2021-01-15:
> To be explicit, for allow_comments I would require:

I agree with your expanded example for `allow_comments.

> Again, "This rule checks for a blank line above the entity keyword." - there is no blank line in that example, so I cannot understand how this rule would be satisfied.

I think this rule has evolved into something else and should be reworded.  Struggling with how to reword it, but maybe something like:

"This rule checks for allowable elements on lines above the entity keyword."

The rule could then be further expanded. For example:

> A particular edge case would be if a block comment would be added. This should be compliant with allow_comment as well as require_comment options.

we could add a `require_block_comment` option.  Maybe for every process you want a block comment instead of just a comment.  Eventually, you could also enforce a particular form of block comment.

So we can continue to evolve the rule

staerz, 2021-01-15:
Hi @jeremiah-c-leary,

OK, I have the feeling that we converge. And yes, in that sense, you would rededicate the rule not only to "blank line" but "pre-keyword" (we'll find a good wording for the rule, give it 1 or 2 more iterations until we're 100% sure to know what the rule is to do exactly :wink:)

OK, good suggestion for the alternative configuration. Simple question: With the `require_block_comment` added, I see that as an exclusive option to `require_comment` and `allow_comment`, while `allow_comment` would also allow block comments. Of course, one could also add the option `allow_block_comment` if one wanted to be super flexible and explicit in the rule configuration.

If we'd now have 4 exclusive options, maybe it would be better to rename it into 1 option that anyway needs to be set and defaults to not allowing any comments (so we are back to the roots, just a blank line above).

Then the option could take (in the end then 5) different values (where we can still discuss the best suited name, depending the name of the option): blank_line (deault), allow_comment, require_comment, allow_block_comment, require_block_comment.

I'm a bit unsure if allowing both, comments and block comments would be a useful option. If so, we could call that simply "allow_[any_]comments" (plural "s", if that seems not obvious enough, add "any").

What do you think?

jeremiah-c-leary, 2021-01-16:
> OK, I have the feeling that we converge

I agree.

> I'm a bit unsure if allowing both, comments and block comments would be a useful option. If so, we could call that simply "allow_[any_]comments" (plural "s", if that seems not obvious enough, add "any").

Although a block comment is different than a regular comment.  So an `allow_comment` and `require_comment` would pass in these cases:
```vhdl
-- Comment
entity fifo is

--+------------
--!  Comment block
--+------------
entity fifo is
```
..but a `require_block_comment` would fail in the first case.

> OK, good suggestion for the alternative configuration. Simple question: With the require_block_comment added, I see that as an exclusive option to require_comment and allow_comment, while [...]

I will create a branch and write up some documentation on configuring the possible options on this rule.  I think it will help us explore the options in greater detail.  When we are satisfied with that document I can work on implementing the rule.

jeremiah-c-leary, 2021-01-16:
Hey @staerz ,

I created pull request #523 to track the proposed changes for this issue.  When you get chance, could you review it and give me any feedback.

Thanks,

--Jeremy

staerz, 2021-01-18:
Hi @jeremiah-c-leary,

alright, I had a look at the pull request.

I think you slightly misunderstood my proposal:
"If we'd now have 4 exclusive options, maybe it would be better to rename it into 1 [configuration] that anyway needs to be set and defaults to not allowing any comments (so we are back to the roots, just a blank line above)."

So instead of having 4 configuration parameters with values of true/false, just have 1 configuration parameter with 4 (or more if needed) different values.

For the currently documented 4 configuration parameters I'd say that this is fully feasible:
- choice `require_blank_line` requires a blank line. No comment, no block comment.
- choice `no_code` requires the line above to be no code, i.e. being a blank line or a comment (that's your current understanding of 'allow_comment')
- choice `allow_comment` allows a comment in the line above (and requires a blank line above that comment)
- choice `require_comment` requires a comment in the line above (and requires a blank line above that comment)

Now an additional option could be `require_block_comment` where you would, as the name suggests, require a block comment above the keyword (which itself requires a blank line, a few commented lines, a blank line again and finally the line with the actual keyword). If that can be done is up to you to judge. If not, than that's fine for me (as it would be compatible with the choice `require_blank_line`.

Why I do propose to move away from the 4 individual true/false configuration parameters is that you open up 2**4 options that you'd need to document - and not all of these 16 options are actually making sense (in fact only a very small subset of them).

Please, if you come up with any other option that you see would be useful, we can add it to the list, but let's avoid having 4 (or even more) settings per rule (as this configuration would apply to quite some rules...).

jeremiah-c-leary, 2021-01-19:
> For the currently documented 4 configuration parameters I'd say that this is fully feasible:
>
> * choice require_blank_line requires a blank line. No comment, no block comment.
> * choice no_code [...]

Okay...I think I get it now.  Each option defines a very specific format, which can cover multiple aspects.  If another request came in for a different format, then the new format would be given a name and it can be as complex as you wanted.

Let me update the documentation and see how that looks.

staerz, 2021-01-19:
Yes, exactly. The only thing is that we should come up with self-explanatory names for the actual setting and all it's options.

I made a few suggestions already, so we can take this as a start but I'm certainly not saying that these would be the optimal and only choices. If you have any suggestions, let me know.

jeremiah-c-leary, 2021-01-20:
> Yes, exactly. The only thing is that we should come up with self-explanatory names for the actual setting and all it's options.

I updated the documentation again and went with a single option that can be configured.  We can work on the names, but what do you think of current documentation?

jeremiah-c-leary, 2021-01-23:
Hey @staerz ,

I believe this is now completed.  When you get a chance, could you validate on your end.

Thanks,

--Jeremy

staerz, 2021-01-25:
Hi @jeremiah-c-leary,

thanks for implementing this!

Looking into the documentation again, I realise that you should add again the list of rules to which [the configuration](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_previous_line_rules.html) applies, just like you started introducing for the other configurations.
That makes it easier to configure on the user side.

I have tested with my required setting `allow_comments` and it works like a charm.

One note though that I only realise now: `instantiation_034` also has a configuration option called `method`.
I would (strongly) suggest that we find a solution for this ambiguity: Either we now simply rename this config, or the one for the above-line setting.
And let me suggest to keep the `instantiation_034` (as I'd say that's really a "method") and maybe use "style" for the above-line rules!?

And another piece on documentation: In the rules you link to `https://vhdl-style-guide.readthedocs.io/en/latest/configuring_previous_line_rules.html` which exists but doesn't figure in the TOC. When I try to find the page from the TOC, then it leads to `https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#configuring-previous-line-rules`, so a sub-anchor of another page. That seems to be a feature of Sphinx, but I'd suggest to link to the latter (for context) in the rule descriptions.

jeremiah-c-leary, 2021-01-25:
> And let me suggest to keep the instantiation_034 (as I'd say that's really a "method") and maybe use "style" for the above-line rules!?

I like `style`.  It more accurate.

> And another piece on documentation: In the rules you link to https://vhdl-style-guide.readthedocs.io/en/latest/configuring_previous_line_rules.html which exists but doesn't figure in the TOC. When I [...]

I will track down the issue and fix it.

> Lastly, I think you missed to indicate the default setting for this configuration.

I will indicate the default setting, although at the moment it differs by the rule.

jeremiah-c-leary, 2021-01-26:
Hey @staerz ,

I have updated the documentation and changed the configurable item to `style`.

As for the linking, it must be a Sphinx thing.  I updated the links to the appropriate page and heading.

I think all the updates were done.  When you get a chance could you validate it.

Thanks,

--Jeremy

staerz, 2021-01-26:
Hi @jeremiah-c-leary,

yes, I confirm that you renamed the option to "style".
Unfortunately it seems that you accidentally also renamed it for `instantiation_034` (while the documentation still calls it `method` as it should be).

I checked the documentation again and found the section [Configuring Blank Lines](https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#configuring-blank-lines) - isn't this supposed to be replaced by "Configuring Previous Line Rules"?

I did not check all links to that section, but a few random rules and the links are now correct.

So apart for those 2 small remainders, I'd say we're good.

staerz, 2021-01-26:
Hi @jeremiah-c-leary,

I'm looking over this again and I'm wondering if we now have introduced an inconsistency or not: I realise that e.g. `architecture_015` still is on `allow_comments`.

This rule checks for "blank line **below** the architecture". The option for this rule is `allow_comments` again - but this option makes no sense here if you ask me, as the comment would refer to the next element, i.e. a signal declaration or similar.

What I think this rule should check is "the next line": It could either be "require a blank line" or "require no blank line" - so the rule should check if it's a blank (true, default) or not a blank (false). `allow_comment` doesn't apply here in my eyes.

If you disable the rule, then it's don't care, but updating the rule to this would allow to enforce to have no blank line there. Of course, the configuration for signal (and similar) should then be made compatible but that's up to the user.

staerz, 2021-01-27:
... similar `architecture_016`: "blank lines or comments **above** the begin keyword."

Usually you don't put a comment there, but what can be checked (again, similar to `architecture_015`) is that you have or don't have a blank line, so I'd opt for the setting "blank_line" to be add (default) or remove (= enforced to not have one). And if you don't care, you disable the rule.

... same story for `architecture_017`: "blank line **below** the begin keyword.": enforce blank line to be there, to not be there or don't care (= disable rule).

... similar `architecture_018`: I think this should be blank line add/remove, but not previous line setting.

... similar `block_201`: I think this should be blank line add/remove, but not previous line setting. (If you want constants and others commented in that section, user make the setting compliant with that rule).

... similar `block_202`: I think this should be blank line add/remove, but not previous line setting.

... similar `block_203`: I think this should be blank line add/remove, but not previous line setting. (it checks **below**, so checking "previous" makes no sense to me)

... similar `block_204`: I think this should be blank line add/remove, but not previous line setting.

... similar `block_205`: I think this should be blank line add/remove, but not previous line setting.

On a similar basis:

... `process_011`: I think this should be blank line add/remove

... `process_021`: should be upgraded to also have this add/remove option

... `process_022`: I think this should be blank line add/remove, not allow_comment

... `process_023`: I think this should be blank line add/remove, not allow_comment

... `process_026`: should be upgraded to also have this add/remove option

... `process_027`: should be upgraded to also have this add/remove option

jeremiah-c-leary, 2021-01-28:
> Unfortunately it seems that you accidentally also renamed it for instantiation_034 (while the documentation still calls it method as it should be).

I think I got a little overzealous with my grep command and forgot to not change that rule.

jeremiah-c-leary, 2021-01-28:
> I'm looking over this again and I'm wondering if we now have introduced an inconsistency or not: I realise that e.g. architecture_015 still is on allow_comments.
>
> This rule checks for "blank line [...]

That is a good point on comments.  You do not comment about code above it.

>
> What I think this rule should check is "the next line": It could either be "require a blank line" or "require no blank line" - so the rule should check if it's a blank (true, default) or not a blank (false). allow_comment doesn't apply here in my eyes.

I think I see your point here, and I believe it also follows a discussion I was having in the  discussions #527.  @BasF0 was wondering why the converse part of rules are not always enforced.  Such as this case, where it should be either a blank line or no blank line.  There should be three options:

 1)  `require_blank` If you want a blank line then it should be added
 2)  `no_blank` If you do not want a blank line then it should be removed
 3)  `no_code` if you want to add a blank line unless there is a comment
 4)  If you don't care, then you disable the rule

The rule would then have a `style` option that could be configured like the previous line rule.  Then they would be consistent with each other.

jeremiah-c-leary, 2021-01-28:
> ... similar architecture_016: "blank lines or comments above the begin keyword."
>
> Usually you don't put a comment there, but what can be checked (again, similar to architecture_015) is that you have [...]

I struggled with a lot of these rules as I was updating them.  I do not put a comment above the begin statement, but then you never know, someone might.  The previous line rules do not have an option to remove a blank line.   Maybe I should add the option:  `no_blank_line`.  Then the previous line rule would cover your case.  It would also give the user the ability to enforce the other options.

staerz, 2021-01-28:
Hi @jeremiah-c-leary,

do not make your live more complicated than it already is: For all rules checking **below** it should either be: require bland line or require no blank line. The latter case already covers the option of having a comment there. And it makes the configuration and rule description straight forward: "Require a blank line **below** the xxx keyword" - if option <invert> is set to true, then the opposite is checked, i.e. that **below** the xxx keyword there is no blank line(s).

Now what concerns comments in front of begin keywords, I can only judge from my experience and what Doxygen does: It requires the comment for an entity, block, architecture, process (or anything else) **directly above** the element keyword and not anywhere else. One would have to check other documenting tools (like sphynx), but I would expect them to not be different (what sense does it make to put a comment there?)

So for all these "below begin" keyword (and non-element-identifying keywords) it should be a simple require blank or require no blank. End of story. Similar also for the end keywords: you set if you require a blank line after it or require not blank there (or you don't care) - but the comment (relative to the next element) would always be considered already by this next element's rule.

jeremiah-c-leary, 2021-01-30:
Hey @staerz ,

I reviewed all the rules checking previous and next lines and organized them into the following:

Blank line after rules with add/remove blank only options:
```mono
architecture_015
architecture_017
architecture_200
block_201
block_203
block_205
case_010
case_008
component_018
context_023
context_025
function_007
generate_003
if_030
instantiation_019
package_011
package_body_201
package_body_203
process_011
process_022
type_011
```
Previous line rules with current previous line method:
```mono
architecture_003
block_200
case_007
component_003
context_003
entity_003
function_006
generate_004
if_031
instantiation_004
library_003
package_003
package_body_200
process_015
type_010
```
Previous line rules with add/remove blank options only:
```mono
architecture_016
architecture_018
block_202
block_204
case_009
context_024
package_012
package_body_202
process_023
```
...and to be consistent, the add/remove blank option will be added to:
```mono
process_021
process_026
process_027
```

I think that covers everything.

jeremiah-c-leary, 2021-01-31:
Hey @staerz ,

I have just pushed updates for this issue.  I believe I have everything working the way it should.  When you get a chance, could you  check it out and let me know?

Thanks,

--Jeremy

staerz, 2021-02-01:
Hi @jeremiah-c-leary,

I did a `grep -ir above -B 10` in `VSG/docs` and compared with your lists above:
- architecture_003: OK
- architecture_016: OK
- architecture_018: OK
- block_200: OK
- block_202: OK
- block_204: OK
- case_009: OK
- component_003: OK 
- component_016: should also be previous=add/remove
- context_003: OK
- context_024: OK
- entity_003: OK
- entity_016: should also be previous=add/remove
- function_006: OK
- generic_001: should also be previous=add/remove
- instantiation_004: OK
- library_003: OK
- library_007: should also be previous=method (note that doxygen likes the use clause documented and hence that suggestion)
- package_003: OK
- package_012: OK
- package_body_200: OK
- package_body_202: OK
- port_001: should also be previous=add/remove
- process_015: OK
- process_023: OK
- type_010: OK

I did a `grep -ir below -B 10` in `VSG/docs` and compared with your lists above:
- architecture_015: OK
- architecture_017: OK
- architecture_200: OK
- block_201: OK
- block_203: OK
- block_205: OK
- case_008: OK
- case_010: OK
- component_018: OK 
- context_023: OK
- context_025: OK
- function_007: OK
- instantiation_019: OK
- package_011: OK
- package_body_201: OK
- package_body_203: OK
- process_022: OK
- type_011: OK

Then comparing your lists with my lists I found a few minor wording inconsistencies:

"after" &rarr; "below":
- generate_003: "checks for a blank line after" &rarr; "checks for a blank line below"
- if_030: "checks for at least a single blank line after" &rarr; "checks for a single blank line below"
- process_011: "checks for a blank line after" &rarr; "checks for a blank line below"

"before" &rarr; "above":
- case_007: "checks for blank lines or comments before" &rarr; "checks for blank lines or comments above"
- generate_004: "checks for blank lines or comments before" &rarr; "checks for blank lines or comments above"
- if_031: "checks for blank lines or comments before" &rarr; "checks for blank lines or comments above"
- process_021: "checks for blank lines before" &rarr; "checks for blank lines above"

For process_026 and process_027, I would suggest, in order to make it consistent with the wording of other rules, the following:
- process_026: "This rule checks for blank lines or comments above the declarative line (if existing)."
- process_027: "This rule checks for blank lines above the begin keyword."

jeremiah-c-leary, 2021-02-03:
Hey @staerz ,

> component_016: should also be previous=add/remove
> entity_016: should also be previous=add/remove
> generic_001: should also be previous=add/remove
> port_001: should also be previous=add/remove
> library_007: should also be previous=method (note that doxygen likes the use clause documented and hence that suggestion)

Those were originally an `excessive_blank_line` or `remove_blank_lines_above_line_starting_with_token` rules, which can now probably be changed to the `blank_line` rule now that it can remove and add blanks.

> Then comparing your lists with my lists I found a few minor wording inconsistencies:

I will fix the wording.

jeremiah-c-leary, 2021-02-03:
Hey @staerz ,

This was easier than I thought it would be.  All rules are and documentation have been updated.

On `library_007`....I added a `no_blank` option to the `previous_line` rule.  I defaulted `library_007` to `no_blank` to match the current implemention.

staerz, 2021-02-03:
Hi @jeremiah-c-leary,

that looks very good.

I now checked the updated documentation again and still found an inconsistency in naming which I suggest you decide for one or the other option:
- [Configuring Blank Lines](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_blank_lines.html) provides the `style` options `require_blank` and `no_blank` while in the example they are called `require_blank_line` and `no_blank_line`. So I'd update the options to `require_blank_line` and `no_blank_line`. I'm wondering if that's just the documentation or also in the implementation ...
- [Configuring Blank Lines](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_blank_lines.html): The link (I came from `library_007`) is again to the html page, not the integrated documentation, maybe have a look and fix that in all cases referencing to it.
- [Configuring Previous Line Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#configuring-previous-line-rules): I know we initially called them that, but now with the similar configuration for "Configuring Blank Lines", also here I'd suggest a renaming of the style option "no_blank" to "no_blank_line" to make it consistent.

Once you'd implement that, I'd have another look and then configure VSG on my side and have a test run to see if I'd find any last things.

We're getting closer to an end, I'd say, and it's getting really good :+1:

jeremiah-c-leary, 2021-02-04:
Hey @staerz ,

I scrubbed the documentation for links and fixed the references so they would link to the correct location.

I updated the `blank_line` and `previous_line` rules to accept `require_blank_line` and `no_blank_line`.

When you get a chance, if you could double check I would appreciate it.

Thanks,

--Jeremy

staerz, 2021-02-04:
Hi @jeremiah-c-leary,

I checked the documentation again and it's really getting there.

One last thing I read and think should be corrected: In [Configuring Blank Lines](https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#configuring-blank-lines), in the table explaining the 2 style options, it's written "Requires a blank line on the line above." and "Removes blank lines above the line." but just a line above you have "blank lines either above or below a line".

Hence 2 suggestions on that:
1) Change the order, to make it like the previous line table.
2) Change the text to "Removes blank lines on the line above/below." and "Requires a blank line on the line above/below.", to make it identical (modulo "/below") to the previous line setting.

I'll give it a shot tomorrow configuring on my side, I'll let you know if anything unexpected will show up.

jeremiah-c-leary, 2021-02-04:
Hey @staerz ,

I copied the two rows from the previous line documentation and added `or below` to the end.

staerz, 2021-02-05:
Hi @jeremiah-c-leary,

so as promised (a bit late) today I went over our configuration with regard to these rules.

I stumbled over the following rule which (you may consider it to be another issue, but in the spirit of consistency, I think it would be fine to treat it here as well) is `generate_011`: this should also be an `action: add/remove` rule I'd say.

Actually, having the list of rules for which a configuration does apply is very helpful in finding these ones!

In the list for [Rules Enforcing Previous Lines](https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#rules-enforcing-previous-lines), the link to `package_body_200` is broken (dash vs. underscore :wink: )
Is it possible that this rule is not properly integrated? I just installed the very latest master and got `ERROR: Rule package_body_200 referenced in configuration could not be found`.

For some reason I'm getting complaints on the following rules:
-  architecture_003          
-  block_200                 
-  block_202                 
-  entity_003                
-  function_006              
-  generate_004              
-  library_003               
-  type_010                  

For `block_202` I set `"style":"no_blank_line"` and for the others I set `"style":"allow_comment"`.
I carefully checked the complaints and I'd say the comments are there and the space above as well, so I'm wondering what's going wrong now.

If you'd say that this shouldn't happen, then there's something fishy with me installing the latest version of VSG (I'm experiencing some python issues here).

Unfortunately, running "vsg --version" doesn't yet give me a SHA of what is actually installed, so I cannot tell for certain what's going on here.

So could you please double check on your side and let me know? In case I could ask a third person to try on his side...

jeremiah-c-leary, 2021-02-05:
Hey @staerz,

> I stumbled over the following rule which (you may consider it to be another issue, but in the spirit of consistency, I think it would be fine to treat it here as well) is generate_011: this should also be an action: add/remove rule I'd say.

That rule looks like it would fall into the configuring optional items.  I should scrub the existing rules to find other candidates.

> In the list for Rules Enforcing Previous Lines, the link to package_body_200 is broken (dash vs. underscore 😉 )

Good catch.

> Is it possible that this rule is not properly integrated? I just installed the very latest master and got ERROR: Rule package_body_200 referenced in configuration could not be found.

Thanks for finding that.  I forgot to include the rules an `__init__.py` file.  I scrubbed the file against the rules and all the other rules were included.

jeremiah-c-leary, 2021-02-05:
> For block_202 I set "style":"no_blank_line" and for the others I set "style":"allow_comment".

So I created this example file:
```vhdl
  1
  2 -- Comment
  3 architecture rtl of fifo is
  4
  5 begin
  6
  7   -- Comment
  8   block1 : block is
  9
 10     constant width : integer := 32;
 11   begin
 12
 13   end block block1;
 14
 15 end architecture rtl;
```
...and this configuration:
```yaml
   rule:
     block_202:
       style: 'no_blank_line'
     architecture_003:
       style: 'allow_comment'
     block_200:
       style: 'allow_comment'
```
and ran with this command:
```mono
vsg -f block_example.vhd -c config.yaml -of syntastic
```
and no errors were reported.

I changed the configuration to this:
```yaml
   rule:
     block_202:
       style: 'require_blank_line'
     architecture_003:
       style: 'require_blank_line'
     block_200:
       style: 'no_code'
```
and run with the same command:
```mono
vsg -f block_example.vhd -c config.yaml -of syntastic
```
and I get the following errors:
```mono
ERROR: block_example.vhd(3)architecture_003 -- Insert blank line above
ERROR: block_example.vhd(11)block_202 -- Insert blank line above
```

Could you copy the above example code and configuration and recreate my steps and compare your results against mine?

jeremiah-c-leary, 2021-02-05:
Hey @staerz ,

I just pushed an update to the link issue and the package_body rules issue.

staerz, 2021-02-05:
:facepalm: I had a particular commit of VSG checked out and hence a pull and reinstall didn't change anyhting.

Back to master and now VSG runs.

When running, I found a few particularities:
architecture_015 and 17 demand me to "Insert blank line below" while they are configured to `no_blank_line` - could you check if possibly you're dumping the wrong error message?

Also, I now figured that some configurations between rules may run into dependencies: I.e. architecture_017 set to `no_blank_line` is incompatible with instantiation_004 set to `allow_comment`.

Now I think this is nothing that you have to find a solution for as that is up to the user to come up with a meaningful and consistent configuration, but I would suggest you simply add a phrase possibly on the page on the configuration `style` that sort of "this may happen and it's the responsibility of the user to make a consistent configuration".

And one very peculiar bug (which I think we should take to a dedicated issue): Have you ever tried to pipe your vsg output into `head`?
If I do that, I get
```bash
[staerz@steffenswkdell LASP (279-implement-vsg-3-0-0-coding-rules)]$ vsg $(cat .vsg/params.txt) | head -n 20
================================================================================
File:  modules/fpga/context/registers.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 79
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  package_body_002          | Error      |        170 | Add *package body* keywords
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
================================================================================
File:  modules/fpga/context/mapping.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 510
Total Violations:    0
Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg===master', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/__main__.py", line 400, in main
    oRules.report_violations(commandLineArguments.output_format)
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/rule_list.py", line 259, in report_violations
    report.vsg_stdout.print_output(dRunInfo)
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/report/vsg_stdout.py", line 14, in print_output
    print_violations(dRunInfo)
  File "/usr/local/lib/python3.8/site-packages/vsg-master-py3.8.egg/vsg/report/vsg_stdout.py", line 101, in print_violations
    print(sOutputString)
BrokenPipeError: [Errno 32] Broken pipe
```

It doesn't happen for `tail`, nor for `grep` or `sed` and I have no clue what that might be caused by.
If you can reproduce that bug, I'd open an issue.

I'll continue my testing (as I have not a big bunch of files to adjust again, and keep posting if I find anything...

staerz, 2021-02-05:
Other candidates where the error messages seems inverted:
- process_011

Then I'm wondering on block_201: is it possible that you're checking for the "is" and then apply the check for the next line?
We're not using the optional is in blocks and I have the impression that then this rule is not checked at all (as it is not found).

staerz, 2021-02-05:
OK, that's all I found on my side when trying to get my code compliant again.

One thought went through my brain when doing so concerning the possible incompatibilities: Maybe one should abolish the rules that look for below. You might always find an example where the "require no blank below" would contradict with a "require blank above".
That would, of course, only be meaningful if you had a "check for above" for all possible VHDL language construction (which I think you almost have).
Now you may not directly see the use case, as this usually is not the case if you pragmatically require a blank everywhere, but that also puffs up your code at some places where it might not be necessary.
Just an idea, maybe give it a thought. (I.e. I would have to re-adjust architecture_015 to 18)

This could also be something for the next version of VSG.

jeremiah-c-leary, 2021-02-06:
> And one very peculiar bug (which I think we should take to a dedicated issue): Have you ever tried to pipe your vsg output into head?

I tried this with a file and I did not get an error.  I am running with Ubuntu using Windows Subsystem for Linux.

Maybe there is something in your params.txt file that I am not using?

> When running, I found a few particularities:
>architecture_015 and 17 demand me to "Insert blank line below" while they are configured to no_blank_line - could you check if >possibly you're dumping the wrong error message?
> * process_011

Yes...looks like I used the same solution for either case.

> Also, I now figured that some configurations between rules may run into dependencies: I.e. architecture_017 set to no_blank_line is incompatible with instantiation_004 set to allow_comment.

Agree.  I added a warning in the configuring section for both previous and blank line documentation.

> Then I'm wondering on block_201: is it possible that you're checking for the "is" and then apply the check for the next line?
> We're not using the optional is in blocks and I have the impression that then this rule is not checked at all (as it is not found).

You are correct.  The `is` keyword triggers the rule.  I will update this rule to handle optional keywords.  For the block_statement, there is also an optional guard_condition I will need to check for.

> One thought went through my brain when doing so concerning the possible incompatibilities: Maybe one should abolish the rules that look for below. You might always find an example where the "require no blank below" would contradict with a "require blank above".
That would, of course, only be meaningful if you had a "check for above" for all possible VHDL language construction (which I think you almost have).
Now you may not directly see the use case, as this usually is not the case if you pragmatically require a blank everywhere, but that also puffs up your code at some places where it might not be necessary.

There seem to be blank line cases I do not cover yet.  For example:
```vhdl
  architecture rtl of fifo is
    constant c_width : integer := 32;
    signal w_data : std_logic_vector(31 downto 0);
  begin
    data_out <= w_data;
  end architecture rtl;
```
Without architecture_015 through architecture_018, I would need rules for above a constant declaration and signal declaration and signal assignment.  I am trying to think of a config option that would enforce a blank line above the constant declaration but not the signal declaration.

jeremiah-c-leary, 2021-02-06:
Hey @staerz ,

I just pushed a fix for block_201 and the incorrect solution for process_011, architecture_015 and architecture_017.  When you get a chance, could you double check.

Thanks,

--Jeremy

staerz, 2021-02-08:
Hi @jeremiah-c-leary,

I think your new commit introduced an error:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/vsg-0.0.0_master_7c25a3df6b9a249d27af214bba82ec7cfb05a871-py3.8.egg/vsg/vhdlFile/extract/utils.py", line 90, in get_indexes_of_tokens_between
    iNextStart = lStartIndexes[iStartIndex + 1]
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/vsg", line 11, in <module>
    load_entry_point('vsg===0.0.0-master-7c25a3df6b9a249d27af214bba82ec7cfb05a871', 'console_scripts', 'vsg')()
  File "/usr/local/lib/python3.8/site-packages/vsg-0.0.0_master_7c25a3df6b9a249d27af214bba82ec7cfb05a871-py3.8.egg/vsg/__main__.py", line 399, in main
    oRules.check_rules(bAllPhases=commandLineArguments.all_phases, lSkipPhase=commandLineArguments.skip_phase)
  File "/usr/local/lib/python3.8/site-packages/vsg-0.0.0_master_7c25a3df6b9a249d27af214bba82ec7cfb05a871-py3.8.egg/vsg/rule_list.py", line 217, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "/usr/local/lib/python3.8/site-packages/vsg-0.0.0_master_7c25a3df6b9a249d27af214bba82ec7cfb05a871-py3.8.egg/vsg/rule.py", line 131, in analyze
    lToi = self._get_tokens_of_interest(oFile)
  File "/usr/local/lib/python3.8/site-packages/vsg-0.0.0_master_7c25a3df6b9a249d27af214bba82ec7cfb05a871-py3.8.egg/vsg/rules/blank_line_below_line_ending_with_several_possible_tokens.py", line 36, in _get_tokens_of_interest
    return oFile.get_blank_lines_below_line_ending_with_several_possible_tokens(self.lTokens)
  File "/usr/local/lib/python3.8/site-packages/vsg-0.0.0_master_7c25a3df6b9a249d27af214bba82ec7cfb05a871-py3.8.egg/vsg/vhdlFile/vhdlFile.py", line 249, in get_blank_lines_below_line_ending_with_several_possible_tokens
    return extract.get_blank_lines_below_line_ending_with_several_possible_tokens(lTokens, self.lAllObjects, self.oTokenMap)
  File "/usr/local/lib/python3.8/site-packages/vsg-0.0.0_master_7c25a3df6b9a249d27af214bba82ec7cfb05a871-py3.8.egg/vsg/vhdlFile/extract/get_blank_lines_below_line_ending_with_several_possible_tokens.py", line 12, in get_blank_lines_below_line_ending_with_several_possible_tokens
    lIndexes = utils.get_indexes_of_tokens_between(lTokens[0], lTokens[1:], oTokenMap)
  File "/usr/local/lib/python3.8/site-packages/vsg-0.0.0_master_7c25a3df6b9a249d27af214bba82ec7cfb05a871-py3.8.egg/vsg/vhdlFile/extract/utils.py", line 92, in get_indexes_of_tokens_between
    iNextStart = lEndIndexes[-1] + 1
IndexError: list index out of range
```

I'm also looking at the updated documentation for `block_201` and it still states "This rule checks for a blank line below the **is** keyword." - shouldn't this now be "This rule checks for a blank line below the **block** keyword.".

Looking into the related commit, I see that you also changed `package_body_201`'s description from **package** to **is**.
I think, although internally it is likely that you run the check after the **is** keyword, for the sake of documenting on *package*, I would still keep it **package** here - as that's what "requires" the new line (frow the view of the item type).
Similar for `package_011`.

jeremiah-c-leary, 2021-02-10:
Hey @staerz ,

Could you send me a snippet that causes the error?

I also pushed an update for the documentation.

staerz, 2021-02-10:
Hi @jeremiah-c-leary,

I'm not 100% certain, but I removed commit f80 from the history and then ran through again, trying to identify the file in question.

I think it is due to this (experimental) code:

```vhdl
-- EMACS settings: -*- tab-width: 2; indent-tabs-mode: nil -*-
-- vim: tabstop=2:shiftwidth=2:expandtab
-- kate: tab-width 2; replace-tabs on; indent-width 2;

------------------------------<-    80 chars    ->------------------------------
--! @file
--! @brief LASP fpga top module registers definition
--! @author Steffen Stärz <steffen.staerz@cern.ch>
--! @todo Find out how to properly group interface sections with doxygen
--! @details This file contains the registers for the LASP fpga top module.
--!
--! This is a very first implementation and adjustments will be needed,
--! most likely including adjustments to the port definitions to carry out or
--! port in external registers.
--------------------------------------------------------------------------------

------------------------------<-    80 chars    ->------------------------------
--! LASP fpga top module registers definition
--------------------------------------------------------------------------------

--! @cond
library fpga;
  context fpga.interfaces;

--! @endcond
entity fpga_registers is
  port (
    --! Clock from `avmm` interface.
    clk_avmm_i    : in    std_logic;
    --! Reset from `avmm` interface (sync with clk_avmm_i).
    rst_avmm_i    : in    std_logic;
    --! Master-to-slave bus (request)
    avmm_master_i : in    t_avmm_master;
    --! Slave-to-master bus (reply)
    avmm_slave_o  : out   t_avmm_slave
  );
end entity fpga_registers;

--! @cond
library fpga;
  use fpga.fpga_cst.all;

library misc;

--! @endcond
--! Implementation of registers of the fpga component
architecture behavioral of fpga_registers is

begin

  blk_fpga_avmm_registers : block

    --! Uptime as 64 bit unsigned
    signal fpga_uptime   : std_logic_vector(G_FPGA_UPTIME_W - 1 downto 0);
    --! Downtime as 32 bit unsigned
    signal fpga_downtime : std_logic_vector(31 downto 0);

    --! @brief Read-only registers (for trial).
    --!
    --! Read-only registers (for trial).
    --!
    --! Since it's a mixture of signals and constants, the reading part is after the 'begin' keyword.
    --! If it were only constants, one could declare it as a constant here.
[... 75 lines cut ...]
      MREGS_RO_SIZE => mregs_ro_fpga_firmware'high,
      MREGS_RW_SIZE => MREGS_RW_INIT_FPGA_FIRMWARE'length,
      MREGS_RW_INIT => MREGS_RW_INIT_FPGA_FIRMWARE
    )
    port map (
      -- Clock from `avmm` interface.
      clk_avmm_i    => clk_avmm_i,
      -- Reset from `avmm` interface (sync with clk_avmm_i).
      rst_avmm_i    => rst_avmm_i,
      -- Master-to-slave bus (request)
      avmm_master_i => avmm_master_i,
      -- Slave-to-master bus (reply)
      avmm_slave_o  => avmm_slave_o,
      -- read only (status) multi-registers
      mregs_ro_i    => mregs_ro_fpga_firmware,
      -- soft-reset writeable registers individually (sync to clk_avmm_i)
      -- rst_mregs_rw_i   => ...
      -- read-write (config) multi-registers
      mregs_rw_o    => mregs_rw_fpga_firmware
    );

  end block blk_fpga_avmm_registers;

end architecture behavioral;
```

jeremiah-c-leary, 2021-02-11:
Hey @staerz ,

I copied your file down and ran it on master at HEAD with a minimal configuration:
```json
 {
   "rule": {
     "instantiation_034": {
       "method": "entity"
     },
     "length_001": {
       "disable" : True
     },
     "block_201": {
       "style" : "no_blank_line"
     }
   }
 }
```
and VSG did not error out.  Could you try the configuration above and see if you get the same error?

staerz, 2021-02-12:
Hi @jeremiah-c-leary,

what I did is to completely drop my local configuration for rules and just ran again. Then the error doesn't show up.
Of course, I get plenty of complaints of rules being violated, but at least no python exception.

Now I'm trying to work myself to the offending rule.

staerz, 2021-02-12:
OK, so that's a tricky one!

What I did now was to have all my settings for rules disabled and reenable them one by one.
Rules that then made VSG crash are these:

```json
    "instantiation_033":{ ###### Problem
      "action":"remove"
    },
    "instantiation_034":{ ###### Problem
      "method":"entity",
    },
```

Now the problem is if I do the reverse check, i.e. remove all settings except these two, then I don't get the error.

So it must be some cross relation between these rules and some other(s).

From the original error report, I would now have to see where `get_blank_lines_below_line_ending_with_several_possible_tokens` is being used and then try to see where `get_indexes_of_tokens_between` points to.

Then, one could try to find a link to `instantiation_033` and `instantiation_034`.

jeremiah-c-leary, 2021-02-12:
I got the same error when using your configuration.  When I disable `block_201` it makes it through the file.  Which makes sense as that is the only rule that uses `get_blank_lines_below_line_ending_with_several_possible_tokens`.

The question is why.

jeremiah-c-leary, 2021-02-12:
I think I got it.

In all of my test examples I have more than one block defined.  In your example there is a single block.

I just pushed a fix.

staerz, 2021-02-12:
Hi @jeremiah-c-leary,

that works like a charm now!

jeremiah-c-leary, 2021-02-13:
> I think that finishes that issue, doesn't it?

Sounds like you are happy, so I will close it.

One more down, just a few more to go to 3.0.0!!!

closed 2021-02-13 by jeremiah-c-leary