jeremiah-c-leary/vhdl-style-guide #505: conflicts between whitespace_010, assert_001 and  concurrent_003

Emilv2, 2021-01-01:
**Environment**
installed latest master (`0d0e35c9`) with `python setup.py install`

**Describe the bug**
I have 2 issues, but I think they are related enough to be in the same report so I put them in the same bug report, I can make separate bug reports if you want to.

* vsg still complains about fixed code:
  the output looks correct, to me but vsg still complains about a `whitespace_010` error on line 14. `whitespace_010` isn't applied here and that seems correct to me so I'd say this should not be counted as an error.
* `assert_001` isn't always applied:
 on line 18 `whitespace_010` is applied, but I don't think it should precede over `assert_001` here.

**To Reproduce**
`vsg -f test.vhd --fix` file output:
```vhdl
-- test.vhd

entity test is
end entity test;

architecture rtl of test is

  signal test_string : string(127 downto 0);
  signal test_int    : integer;

begin

  test_string <= "Lorem ipsum dolor sit amet, consectetuer adipiscing elit. Aenean "
                 & "commodo ligula eget dolor. Aenean massa. Cum sociis natoque pen";
  test_int    <= 100;

  assert (test_slv = 101) report "test_int should be 100 but is "
 & integer'image(test_int)
    severity FAILURE;

end architecture rtl;
```
terminal output:
```
================================================================================
File:  test.vhd
================================================================================
Phase 2 of 7... Reporting
Total Rules Checked: 189
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  whitespace_010            | Error      |         14 | Remove all but one space before &. Remove all but one space after &
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
**Expected behavior**

I think the correct output should be:

```vhdl
-- test.vhd

entity test is
end entity test;

architecture rtl of test is

  signal test_string : string(127 downto 0);
  signal test_int    : integer;

begin

  test_string <= "Lorem ipsum dolor sit amet, consectetuer adipiscing elit. Aenean "
                 & "commodo ligula eget dolor. Aenean massa. Cum sociis natoque pen";
  test_int    <= 100;

  assert (test_slv = 101) report "test_int should be 100 but is "
    & integer'image(test_int)
    severity FAILURE;

end architecture rtl;
```
Alignment on line 18 could also be with the rest of the string I suppose, but it looks a bit off to me:
```vhdl
  assert (test_slv = 101) report "test_int should be 100 but is "
                                 & integer'image(test_int)
    severity FAILURE;

```

jeremiah-c-leary, 2021-01-02:
Hey @Emilv2 ,

I just pushed a fix for this issue.  It turns out I was not taking into account if the concat operator was at the start of the line.

> Alignment on line 18 could also be with the rest of the string I suppose, but it looks a bit off to me:

Currently only the keywords in the `assert` will be indented.

So in your example you could do something like this:

```vhdl
  assert (test_slv = 101)
    report "test_int should be 100 but is " & integer'image(test_int)
    severity FAILURE;
```
although that does not really address what should happen if the report line spans multiple lines.

When you get time, could you check if the invalid reporting is fixed?

Thanks,

--Jeremy

Emilv2, 2021-01-02:
```vhdl
  assert (test_slv = 101)
    report "test_int should be 100 but is " & integer'image(test_int)
    severity FAILURE;
```
This works, the problem it that in my case the string is a bit long and I'd like to split it over multiple lines.

The invalid report is fixed.

jeremiah-c-leary, 2021-01-02:
> This works, the problem it that in my case the string is a bit long and I'd like to split it over multiple lines.

I just created issue #506 to cover the formatting of the expression after the report keyword.

> The invalid report is fixed.

Would you be good with closing this issue?

Emilv2, 2021-01-02:
Sure

closed 2021-01-02 by Emilv2