jeremiah-c-leary/vhdl-style-guide #506: Add rule for indenting the expression of a report

jeremiah-c-leary, 2021-01-02:
The following code example:
```vhdl
  assert condition report "This is output to the user" &
       "That continues on a new line."
    severity warning;
```
Could be formatted as the following:
```vhdl
  assert condition report "This is output to the user" &
                          "That continues on a new line."
    severity warning;
```
or
```vhdl
  assert condition report "This is output to the user" &
    "That continues on a new line."
    severity warning;
```
or maybe:
```vhdl
  assert condition report "This is output to the user" &
      "That continues on a new line."
    severity warning;
```

jeremiah-c-leary, 2021-01-12:
Hey @joshrsmith,

I just pushed a new rule to handle multiline report statements.  I believe you had raised the issue in a different ticket.  When you get a chance could you check if it fits your needs.

Thanks,

--Jeremy

joshrsmith, 2021-01-14:
Hmmm I don't remember having a particular issue with this.

jeremiah-c-leary, 2021-01-14:
My apologies, it was @Emilv2  who brought it up in issue #505.  I created this issue and forgot to link it.

Emilv2, 2021-01-14:
Seems good to me!

jeremiah-c-leary, 2021-01-14:
awesome!!

closed 2021-01-14 by jeremiah-c-leary