jeremiah-c-leary/vhdl-style-guide #508: Procedure without input arguments within architecture declaration causes parsing error

joshrsmith, 2021-01-04:
**Environment**

* Master circa https://github.com/jeremiah-c-leary/vhdl-style-guide/commit/65c61ce01a585b04bd4041f0012722f83718c14f
* Running on Linux

**Describe the bug**

Error: Unexpected token detected while parsing architecture_body @ Line XXX, Column 4
       Expecting : begin
       Found     : procedure

**To Reproduce**
Have entity / architecture pair.
Within the architecture (before the `begin`)

OK:
```
   procedure ok_proc(my_input : in boolean) is
      variable foo : boolean;
   begin
      null;
   end ok_proc;
```

Causes error:
```
   procedure not_ok_proc is
      variable foo : boolean;
   begin
      null;
   end not_ok_proc;
```

May be related: There are multiple procedure and function definitions within the architecture.

**Expected behavior**
Parsing error would not occur.

jeremiah-c-leary, 2021-01-04:
@joshrsmith ,

I will check this out.

jeremiah-c-leary, 2021-01-05:
Hey @joshrsmith ,

I just pushed an update for this issue.  Could you let me know if it is resolved on your end.

Thanks,

--Jeremy

joshrsmith, 2021-01-05:
Yes, it looks like the issue is resolved.

closed 2021-01-05 by joshrsmith