jeremiah-c-leary/vhdl-style-guide #509: Rule to prevent multiple ports on the same line

staerz, 2021-01-04:
**Is your feature request related to a problem? Please describe.**
It is a feature request for a new rule or set of rules.

**Describe the solution you'd like**
VHDL allows the definition of multiple ports per line, separated by comma.

This may be acceptable for small projects, but the far majority of code (I've seen) uses the convention to have 1 port per line.

If I haven't overlooked anything, VSG currently doesn't implement rules to mark the following as a violation:
```vhdl
  port (
    --! Input clock and reset
    clk, reset : in    std_logic;
    ...
  );
```

There should be a rule, possibly disabled by default, that prevents to declare multiple ports per line as given in the example above.

To be checked: Do the rules related to alignment of port definitions possibly depend on having a single port defined per line or would they also be compatible with the above example which could be considered valid (depending on the project's coding standard)? 

Note: `port_013` prevents only the following:
```vhdl
  port (
    --! Input clock and reset
    clk : in    std_logic; reset : in    std_logic;
    ...
  );
```
i.e. it complain on non-comment following the semi-colon.

**Describe alternatives you've considered**
There is no alternative :wink:

**Additional context**
A related rule for port maps already exists (`port_map_009`).

Possibly one could think of similar rules when declaring signals, constants, types, variables, files (etc. - did I forget sth.?).
Maybe some already exists, I didn't check.

jeremiah-c-leary, 2021-01-09:
Hey @staerz ,

I just pushed an update for this feature.  When you get a chance, could you validate it?

Thanks,

--Jeremy

staerz, 2021-01-11:
Hi @jeremiah-c-leary,

it seems you have changed something with the parser for the configuration files. If I install the latest version of master and run VSG as I always did, I get this:

```
VHDL Style Guide (VSG): error: unrecognized arguments: { "file_list":[ "modules/fpga/context/*.vhd", "modules/fpga/src/external/*.vhd", "modules/fpga/src/private/components.vhd", "modules/fpga/src/private/fpga_registers.vhd", "modules/fpga/src/private/fpga_top.vhd":{ "rule":{ "block_200":{ "disable": true }, }, }, "modules/fpga/src/private/s10_devkit_top_arch.vhd":{ "rule":{ "architecture_025":{ "names":[ "fpga_lasp_top", ], }, "length_001":{ "disable":true, }, "signal_016":{ "disable":true, }, }, }, "modules/fpga/src/private/sctrl_*.vhd", "modules/fpga/src/public/*.vhd", "modules/fpga/simulation/private/fpga_top/fpga_top_tb.vhd":{ "rule":{ "signal_016":{ "disable": true }, }, }, "modules/fpga/simulation/private/s10_devkit_top/s10_devkit_top_tb.vhd", "modules/fpga/simulation/public/*/*.vhd", "modules/fpga/target_hw/devkit/s10_devkit_top_ent.vhd":{ "rule":{ "port_025":{ "disable": true }, } }, ], }
```
The syntax follows the documentation, would you be able to tell me what has changed?

jeremiah-c-leary, 2021-01-11:
Hey @staerz ,

That is strange.  I would expect those errors to be induced by file changes at the top level, but when I checked when they last changed...

[image]

...and it has been awhile.

What command did you issue?

staerz, 2021-01-11:
Hi @jeremiah-c-leary,

false alert, I was using a wrong command :facepalm: ... I realise I was in weekend and my mind not yet reset to work :rofl:

I checked your implementation and it now complains as it should when trying to define 2 ports on the same line.

Please feel free to close the issue.

jeremiah-c-leary, 2021-01-11:
> I realise I was in weekend and my mind not yet reset to work

Fully understand.  It was hard for me get going this morning.

closed 2021-01-11 by jeremiah-c-leary