jeremiah-c-leary/vhdl-style-guide #510: Procedure declaration with array argument within process causes parser crash

joshrsmith, 2021-01-05:
**Environment**
Linux
Circa https://github.com/jeremiah-c-leary/vhdl-style-guide/commit/b9ff42e5b1f395646f1fdfe24c115ca2322409aa

**Describe the bug**

Error: Unexpected token detected while parsing process_statement @ Line XXX, Column YY
       Expecting : begin
       Found     : )

**To Reproduce**

I believe this will reproduce the error:
```
Setup_Baz : process
   procedure init_baz(constant mode : in std_logic_vector(31 downto 0)) is
   begin
      -- foo
   end procedure;
begin
   -- bar
end process Setup_Baz;
```

This will not have the error (note the argument type for the procedure is different):
```
Setup_Baz : process
   procedure init_baz(constant mode : in boolean) is
   begin
      -- foo
   end procedure;
begin
   -- bar
end process Setup_Baz;
```

**Expected behavior**
Parser does not crash

jeremiah-c-leary, 2021-01-06:
Hey @joshrsmith ,

I just pushed a fix for this.  The classifier was not handling the closing parenthesis correctly.  When you get a chance, could you validate it is fixed on your end.

Thanks,

--Jeremy

joshrsmith, 2021-01-06:
Yes it looks like this was resolved.

closed 2021-01-06 by joshrsmith