jeremiah-c-leary/vhdl-style-guide #511: Parser crashes on case statement when indexing an array of case value

joshrsmith, 2021-01-06:
**Environment**
Linux
Circa https://github.com/jeremiah-c-leary/vhdl-style-guide/commit/9b7ba24e0b5a360bf9ae1c9b42dfb290d091c193

**Describe the bug**
Error: Unexpected token detected while parsing case_statement @ Line XXX, Column YY
       Expecting : is
       Found     : i

**To Reproduce**

I suspect the problematic code is:

```
for i in 0 to 3 loop
   case mode(i) is
      when "00" =>
           ...
```

Note that `mode` is an array of `std_logic_vector` which is being indexed with `i`.

**Expected behavior**
Parser would not crash.

jeremiah-c-leary, 2021-01-06:
Hey @joshrsmith ,

This was a subtle bug.  I almost have a fix completed.  I should hopefully get it done tonight.

jeremiah-c-leary, 2021-01-08:
Hey @joshrsmith ,

I think I finally have this resolved.  When you get a chance, could you validate on your end.

Thanks,

--Jeremy

joshrsmith, 2021-01-08:
Resolved.

closed 2021-01-08 by joshrsmith