jeremiah-c-leary/vhdl-style-guide #512: Print current file being parsed when parser crash/error occurs

joshrsmith, 2021-01-06:
**Is your feature request related to a problem? Please describe.**
I have been encountering quite a few parser problems (which I have been reporting). When a parser problem occurs, it gets reported like this: 

```
Error: Unexpected token detected while parsing case_statement @ Line 423, Column 37
       Expecting : is
       Found     : i
```

But it does not indicate what file Line/Column belongs to. This requires me to look at files it recently parsed and guess where it currently is in my directory structure.

**Describe the solution you'd like**
It would be a lot better if the output was this:

```
Error: Unexpected token detected while parsing case_statement @ Line 423, Column 37 in file XXX
       Expecting : is
       Found     : i
```

**Describe alternatives you've considered**
Maybe there is a debugging option that I could turn on so that it prints out the filename of every file when it stats to parse it?

jeremiah-c-leary, 2021-01-14:
Hey @joshrsmith ,

This one took me awhile to figure out how to implement, but I finally got it done.  Could you check it out when you get a chance?

Thanks,

--Jeremy

joshrsmith, 2021-01-14:
Yes, this is perfect and much easier to track down exactly where the parser is getting hung up!

closed 2021-01-14 by joshrsmith