jeremiah-c-leary/vhdl-style-guide #513: Support for VHDL Pragmas?

joshrsmith, 2021-01-08:
**Is your feature request related to a problem? Please describe.**

I have some code that utilizes VHDL pragmas like this:

```
      --vhdl_comp_off
      Foo : FOO_COMPONENT
      --vhdl_comp_on
      --synthesis translate_off
      Foo : entity work.FOO_ENTITY
      --synthesis translate_on
         generic map (
               ....
```

This allows the code to be used in a few different use cases across simulation tools and vendor tool chains (e.g. Xilinx and Intel). I agree it is a bit ugly and there may be better ways to handle this, but that is how it is right now.

Since the HDL parser does not recognize `--vhdl_comp_off`, the parser crashes when it encounters the line between `--synthesis translate_off/on`.

```
Error: Unexpected token detected while parsing component_instantiation_statement @ Line 262, Column 7
       Expecting : ;
       Found     : Foo
```

**Describe the solution you'd like**
Other than finding another work-around for this in my code which does not use the pragmas, I am not sure what the best way to gracefully handle this in VSG would be. VHDL Style guide could make its own pragmas to ignore lines?  This is tricky because VHDL does not have a standardized set of pragmas and how they should behave. I feel like the only reliable option would be to define VSG specific pragmas to ignore lines of code.

jeremiah-c-leary, 2021-01-08:
I could add a pragma option that would let the user define the pragmas.  Then the parser would just use them to ignore code.

Is `--vhdl_comp_off` a custom pragma, or is it defined somewhere?

joshrsmith, 2021-01-08:
Here is a list of some

https://insights.sigasi.com/tech/list-known-vhdl-metacomment-pragmas/

Honestly it might be best to add your own: —vsg_off/on instead of trying to make reasonable decisions about all of the other ones.

I can imagine users might also want to ignore single rules on one line of code. Might want to take a look at how other tools handle this (pylint and flake8 for python come to mind. This might be a related but separate issue since the problem here isn’t ignoring failing rules, we need to prevent some code from being parsed in the parsing engine.

Josh

jeremiah-c-leary, 2021-01-08:
> https://insights.sigasi.com/tech/list-known-vhdl-metacomment-pragmas/

Most of those would not matter for the parser.  I think the only ones that do are `--vhdl_comp_[on|off]`.  Even `--synthesis translate_[on|off]` would still need valid syntax.

It looks like in your example when you compile it you want to entity instantiation, but when you synthesize it you want the component instantiation.

I think this is do-able.  I just need to figure out how to get the information to the parser.  I need to do that anyway to print out the filename if an error occurs.

> I can imagine users might also want to ignore single rules on one line of code.

There is support for that with the code tags.

jeremiah-c-leary, 2021-01-09:
Hey @joshrsmith ,

I just pushed an update for this issue.  Could you validate on your side?

Thanks,

--Jeremy

joshrsmith, 2021-01-11:
This works for me.
Resolved.

closed 2021-01-11 by joshrsmith