jeremiah-c-leary/vhdl-style-guide #514: Parser does not support infinite/unconditional loops

joshrsmith, 2021-01-08:
**Environment**
Linux, circa https://github.com/jeremiah-c-leary/vhdl-style-guide/commit/fa5acc0794990a94dce40b2f649b864b0a39eca8

**Describe the bug**

Error: Unexpected token detected while parsing process_statement @ Line XXX, Column YY
       Expecting : process
       Found     : loop

**To Reproduce**

This causes error:
```
   Foo_Proc : process
   begin
      L1 : loop -- error here
         wait until rising_edge(clock);
      end loop;
   end process Foo_Proc ;
```

This works around the error:
```
   Foo_Proc : process
   begin
      L1 : while true loop -- no error
         wait until rising_edge(clock);
      end loop;
   end process Foo_Proc ;
```

**Expected behavior**
Unconditional loops would be supported.

jeremiah-c-leary, 2021-01-08:
Hey @joshrsmith ,

I found the issue with the infinite loops.  When you get a chance could you validate it is working on your end.

Thanks,

--Jeremy

joshrsmith, 2021-01-11:
This appears to be resolved.

closed 2021-01-11 by joshrsmith