jeremiah-c-leary/vhdl-style-guide #515: Parser does not recognize calling procedure without arguments from another procedure

joshrsmith, 2021-01-08:
**Environment**
Linux, circa https://github.com/jeremiah-c-leary/vhdl-style-guide/commit/fa5acc0794990a94dce40b2f649b864b0a39eca8

**Describe the bug**

Error: Unexpected token detected while parsing loop_statement @ Line XXX, Column YY
       Expecting : loop
       Found     : `<name of procedure>`

**To Reproduce**

```
My_Proc : process
  ....
   procedure foo is
   begin
      ...
   end procedure foo;

   procedure bar is
   begin
      foo; --error here
   end procedure bar;
   ...
```

**Expected behavior**
This would be supported by the parser.

jeremiah-c-leary, 2021-01-08:
Hey @joshrsmith ,

I created this file:
```vhdl
architecture rtl of fifo is

 begin

   proc_label : process

     procedure foo is
     begin
     end procedure foo;

     procedure bar is
     begin
       foo;
     end procedure bar;

   begin

   end process;

 end architecture;
```

and it was parsed okay.  I noticed the error was related to a `loop_statement` and was wondering if this is fixed with the update to for issue #514 .  Could you check if this issue is resolved when you check the other issue?

Thanks,

--Jeremy

joshrsmith, 2021-01-11:
There is still an issue here. 
I think I narrowed it down...

This triggers the problem:
```
architecture rtl of fifo is

 begin

   proc_label : process

     procedure foo is
     begin
     end procedure foo;

     procedure bar is
     begin
       foo;
       for i in 0 to G_GENERIC-1 loop
               -- stuff
       end loop;
     end procedure bar;

   begin

   end process;

 end architecture;
```

This works around the problem (note the label on the loop):
```
architecture rtl of fifo is

 begin

   proc_label : process

     procedure foo is
     begin
     end procedure foo;

     procedure bar is
     begin
       foo;
       my_loop : for i in 0 to G_GENERIC-1 loop
               -- stuff
       end loop my_loop;
     end procedure bar;

   begin

   end process;

 end architecture;
```

jeremiah-c-leary, 2021-01-11:
Hey @joshrsmith ,

That example helped.  I replicated the error and just pushed a fix.

Thanks,

--Jeremy

joshrsmith, 2021-01-11:
Resolved.

closed 2021-01-11 by joshrsmith