jeremiah-c-leary/vhdl-style-guide #516: Parser error with VHDL'93 file opening mechanism with function call

joshrsmith, 2021-01-08:
**Environment**
Linux, circa https://github.com/jeremiah-c-leary/vhdl-style-guide/commit/fa5acc0794990a94dce40b2f649b864b0a39eca8

**Describe the bug**
Error: Unexpected token detected while parsing file_declaration @ Line XXX, Column YY
       Expecting : ;
       Found     : output_path

**To Reproduce**

The error is triggered by this:
```
My_Proc : process(clk)
   file bin_out            : bin_file open write_mode is (output_path(argument) & C_CONSTANT_FILE_EXT);
```
Possibly it is triggered because a function (`output_path`) is being used to determine the filename, and it is combined with a constant that specifies the extention.

**Expected behavior**
This syntax would be supported.

jeremiah-c-leary, 2021-01-09:
Hey @joshrsmith ,

I believe I have this issue addressed.  When you get a chance, could you validate on your end.

Thanks,

--Jeremy

joshrsmith, 2021-01-11:
Resolved.

closed 2021-01-11 by joshrsmith