jeremiah-c-leary/vhdl-style-guide #528: Unit test failures when run inside a Docker container

imd1, 2021-02-09:
Hi

I'm installing VSG inside a Docker image and as part of the image build, I run the unit tests for assurance. Recently (master in 2021) some of the tests fail and I assume it is related to the environment inside the Docker container, and I'm looking for some insight to determine what the problem may be.

Here is the unit test output:

```..............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................F.....................................................................................................................................................................................................................................Traceback (most recent call last):
  File "bin/vsg", line 14, in <module>
    main()
  File "/usr/local/lib/python3.8/site-packages/vsg-0.0.0.dev0-py3.8.egg/vsg/__main__.py", line 344, in main
    configuration = read_configuration_files(dStyle, commandLineArguments)
  File "/usr/local/lib/python3.8/site-packages/vsg-0.0.0.dev0-py3.8.egg/vsg/__main__.py", line 122, in read_configuration_files
    for sKey in tempConfiguration.keys():
AttributeError: 'NoneType' object has no attribute 'keys'
F...........ss...............................
======================================================================
FAIL: test_file_no_permission (vsg.tests.source_file.test_source_file.testOSError)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/vhdl-style-guide-master/vsg/tests/source_file/test_source_file.py", line 51, in test_file_no_permission
    self.assertEqual(utils.replace_total_count(lActual), lExpected)
AssertionError: Lists differ: ['===[631 chars] 0 | File empty.', '--------------------------[193 chars], ''] != ['===[631 chars] 0 | Permission denied', '--------------------[199 chars], '']

First differing element 11:
'  source_file_001           | Error      |          0 | File empty.'
'  source_file_001           | Error      |          0 | Permission denied'

Diff is 1169 characters long. Set self.maxDiff to None to see it.

======================================================================
FAIL: test_no_permission_configuration_file (vsg.tests.vsg.test_vsg.testVsg)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/vhdl-style-guide-master/vsg/tests/vsg/test_vsg.py", line 317, in test_no_permission_configuration_file
    subprocess.check_output(['bin/vsg', '-c', sNoPermissionFile])
subprocess.CalledProcessError: Command '['bin/vsg', '-c', 'no_permission.yml']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/vhdl-style-guide-master/vsg/tests/vsg/test_vsg.py", line 324, in test_no_permission_configuration_file
    self.assertEqual(sActual, sExpected)
AssertionError: '' != 'ERROR: encountered PermissionError, Permi[62 chars]ml\n'
+ ERROR: encountered PermissionError, Permission denied while opening configuration file: no_permission.yml

----------------------------------------------------------------------
Ran 1633 tests in 53.831s

FAILED (failures=2, skipped=2)

```

jeremiah-c-leary, 2021-02-10:
Hey @imd1 ,

Those tests are checking file permission errors are reported correctly.

On line 40 of `vsg/tests/source_file/test_source_file.py` a touch is performed on a file that does not exist.  Permissions are set so that when the file is opened a permissions error will be reported.
```python
 38
 39     def test_file_no_permission(self):
 40         pathlib.Path(sNoPermissionFile).touch(mode=0o222, exist_ok=True)
```
You can tell in the error being reported that `File empty` was reported as the error instead of `Permission denied`.

The same occurs with the second test.

It is as if the touch command does not work.

imd1, 2021-02-10:
* Ran the unit tests inside my Docker container as root and I get the same error
* navigated to the python test folder and looked around...

```
[root@d2dee70be146 source_file]# ll
total 16
-rw-r--r-- 1 root root    0 Feb 10 02:53 __init__.py
drwxr-xr-x 2 root root   76 Feb 10 12:13 __pycache__
-rw-r--r-- 1 root root    0 Feb 10 02:53 empty_file.vhd
-rw-r--r-- 1 root root  853 Feb 10 02:53 output_empty_file.txt
-rw-r--r-- 1 root root  842 Feb 10 02:53 output_no_file.txt
-rw-r--r-- 1 root root  840 Feb 10 02:53 output_no_permission.txt
-rw-r--r-- 1 root root 2194 Feb 10 02:53 test_source_file.py
[root@d2dee70be146 source_file]# cat output_empty_file.txt 
================================================================================
File:  vsg/tests/source_file/empty_file.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 200
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  source_file_001           | Error      |          0 | File empty.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
[root@d2dee70be146 source_file]# cat output_no_file.txt    
================================================================================
File:  no_file.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 200
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  source_file_001           | Error      |          0 | No such file or directory
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
[root@d2dee70be146 source_file]# cat output_no_permission.txt 
================================================================================
File:  no_permission.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 200
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  source_file_001           | Error      |          0 | Permission denied
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

* so the no permission test IS getting a permission denied error somewhere?
```

imd1, 2021-02-10:
* oh yes, these files are the expected output....sorry if I've misunderstood

imd1, 2021-02-10:
```
[root@d2dee70be146 source_file]# touch --version
touch (GNU coreutils) 8.22
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Paul Rubin, Arnold Robbins, Jim Kingdon,
David MacKenzie, and Randy Smith.
[root@d2dee70be146 source_file]# touch --help
Usage: touch [OPTION]... FILE...
Update the access and modification times of each FILE to the current time.

A FILE argument that does not exist is created empty, unless -c or -h
is supplied.

A FILE argument string of - is handled specially and causes touch to
change the times of the file associated with standard output.

Mandatory arguments to long options are mandatory for short options too.
  -a                     change only the access time
  -c, --no-create        do not create any files
  -d, --date=STRING      parse STRING and use it instead of current time
  -f                     (ignored)
  -h, --no-dereference   affect each symbolic link instead of any referenced
                         file (useful only on systems that can change the
                         timestamps of a symlink)
  -m                     change only the modification time
  -r, --reference=FILE   use this file's times instead of current time
  -t STAMP               use [[CC]YY]MMDDhhmm[.ss] instead of current time
      --time=WORD        change the specified time:
                           WORD is access, atime, or use: equivalent to -a
                           WORD is modify or mtime: equivalent to -m
      --help     display this help and exit
      --version  output version information and exit

Note that the -d and -t options accept different time-date formats.

GNU coreutils online help: <http://www.gnu.org/software/coreutils/>
Report touch translation bugs to <http://translationproject.org/team/>
For complete documentation, run: info coreutils 'touch invocation'
```

imd1, 2021-02-10:
Found this definition of pathlib.touch()

```
Path.touch(mode=0o666, exist_ok=True)
Create a file at this given path. If mode is given, it is combined with the process’ umask value to determine the file mode and access flags. If the file already exists, the function succeeds if exist_ok is true (and its modification time is updated to the current time), otherwise FileExistsError is raised.
```

imd1, 2021-02-10:
I created an empty file with the correct permissions (?) and ran vsg...gives file empty error confirming the unit test behavior

```
[root@d2dee70be146 source_file]# ll
total 16
-rw-r--r-- 1 root root    0 Feb 10 02:53 __init__.py
drwxr-xr-x 2 root root   76 Feb 10 12:13 __pycache__
-rw-r--r-- 1 root root    0 Feb 10 02:53 empty_file.vhd
--w--w--w- 1 root root    0 Feb 10 13:40 fred.vhd
-rw-r--r-- 1 root root  853 Feb 10 02:53 output_empty_file.txt
-rw-r--r-- 1 root root  842 Feb 10 02:53 output_no_file.txt
-rw-r--r-- 1 root root  840 Feb 10 02:53 output_no_permission.txt
-rw-r--r-- 1 root root 2194 Feb 10 02:53 test_source_file.py
[root@d2dee70be146 source_file]# vsg -f fred.vhd 
================================================================================
File:  fred.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 80
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  source_file_001           | Error      |          0 | File empty.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.

```

imd1, 2021-02-10:
Another go...

```
[root@d2dee70be146 source_file]# ll
total 16
-rw-r--r-- 1 root root    0 Feb 10 02:53 __init__.py
drwxr-xr-x 2 root root   76 Feb 10 12:13 __pycache__
-rw-r--r-- 1 root root    0 Feb 10 02:53 empty_file.vhd
---------- 1 root root    0 Feb 10 13:40 fred.vhd
-rw-r--r-- 1 root root  853 Feb 10 02:53 output_empty_file.txt
-rw-r--r-- 1 root root  842 Feb 10 02:53 output_no_file.txt
-rw-r--r-- 1 root root  840 Feb 10 02:53 output_no_permission.txt
-rw-r--r-- 1 root root 2194 Feb 10 02:53 test_source_file.py
[root@d2dee70be146 source_file]# vsg -f fred.vhd 
================================================================================
File:  fred.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 80
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  source_file_001           | Error      |          0 | File empty.
----------------------------+------------+------------+--------------------------------------

```

jeremiah-c-leary, 2021-02-10:
That is strange.  I just ran this experiment outside a container:
```mono
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide/issue_validation/issue_528$ ls -lart
total 8
drwxr-xr-x 31 jcleary jcleary 4096 Feb 10 07:45 ..
-rw-r--r--  1 jcleary jcleary    0 Feb 10 07:46 test.vhd
drwxr-xr-x  2 jcleary jcleary 4096 Feb 10 07:46 .
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide/issue_validation/issue_528$ chmod 222 test.vhd
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide/issue_validation/issue_528$ ls -lart
total 8
drwxr-xr-x 31 jcleary jcleary 4096 Feb 10 07:45 ..
--w--w--w-  1 jcleary jcleary    0 Feb 10 07:46 test.vhd
drwxr-xr-x  2 jcleary jcleary 4096 Feb 10 07:46 .
jcleary@DESKTOP-HV9NHA3:~/projects/vhdl-style-guide/issue_validation/issue_528$ vsg -f test.vhd
================================================================================
File:  test.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 80
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  source_file_001           | Error      |          0 | Permission denied
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
Is the container not honoring permissions?

imd1, 2021-02-10:
Following your suggestions, I created an empty file and removed all permissions and was able to cat the file without error inside the container, but when repeated outside a permissions error was generated. Both environments are Centos 7. I'll keep investigating....

imd1, 2021-02-10:
Ahh...inside the container, I am root, and do these permissions not apply to root?

imd1, 2021-02-10:
so the theory is that the unit test will fail if you installed as root and are running as root

jeremiah-c-leary, 2021-02-10:
Those tests could be skipped.

I wonder if I can detect root and just ignore the test.

imd1, 2021-02-10:
In the Dockerfile, I could create a user and switch before running the unit tests?

imd1, 2021-02-10:
I'm going to try and prototype a new user and check it fixes the problem....

imd1, 2021-02-10:
Your suggestio also makes sense...

```
import os
 
if os.geteuid()==0:
  print "Running as root."
else:
  print "User is not root."

```

imd1, 2021-02-10:
I'm struggling slightly with my idea, so would be happy to test your suggestion

imd1, 2021-02-10:
I have successfully installed VSG inside my Docker image as a non root user and run the unit tests with no errors as the non root user.

imd1, 2021-02-10:
Many thanks for your help and insight - obviously it's up to you if you want to add a root user fix to your code

imd1, 2021-02-10:
Just noticed an interesting difference....
```
[imd@fpga-dev-proj-vsg-3-0 ~]$ vsg --version
VHDL Style Guide (VSG) version 2.0.0
Git commit SHA: 0ba68a4
```
* no mention of the fact that it was downloaded as a zip file
* using same version as #486

imd1, 2021-02-10:
Here is my Dockerfile snippet

```
# Create a new user "vsg" with a home directory and shell
RUN groupadd --gid 5000 vsg && \
    useradd --home-dir /home/vsg --create-home --uid 5000 \
        --gid 5000 --shell /bin/sh --skel /dev/null vsg && \
    chmod 755 /home/vsg

# Switch to non root user "vsg" to install VSG and run the unit tests
# which require file permissions to work which will not if run as root
USER vsg

# Download new version of VSG to folder accessible by non root user
ARG VSG_ZIP_FILE2=${VSG_FOLDER}-5d43a912d59e82ded163c69bab2d25ddf56e3f5b
RUN echo "Downloading and extracting ${VSG_ZIP_FILE} from ${ARTIFACT_HOST}" && \
  cd /home/vsg && \
  curl ${ARTIFACT_HOST}/vsg/${VSG_ZIP_FILE2}.zip -o ${VSG_ZIP_FILE2}.zip --cacert /etc/pki/ca-trust/source/anchors/ca.crt && \
  unzip ${VSG_ZIP_FILE2}.zip && \
  rm ${VSG_ZIP_FILE2}.zip

# Run VSG setup as non root user
RUN echo "Setting up ${VSG_ZIP_FILE2}" && \
  cd /home/vsg/${VSG_FOLDER} && \
  python3 setup.py install --user

# Run VSG unit tests as non root user
RUN echo "Running ${VSG_ZIP_FILE2} unit tests" && \
  cd /home/vsg/${VSG_FOLDER} && \
  python -m unittest

```

imd1, 2021-02-10:
Within the container,  if I navigate to a folder that is outside my git repo, I get this...

```
$ vsg --version
VHDL Style Guide (VSG) version 2.0.0

```

jeremiah-c-leary, 2021-02-10:
> Many thanks for your help and insight - obviously it's up to you if you want to add a root user fix to your code

I think I will implement it.  If I do not, someone else will run into the same issue and I will have forgotten what the solution is.

> no mention of the fact that it was downloaded as a zip file
> using same version as #486
> I'm running vsg from the root folder of a git repository and it's picking up the git sha from there somehow!?!
> I installed vsg with the --user option as my non root user did not have permissions

Those are some odd behaviors.

Well...I did not check for the `.git` directory and relied on the git command failing.  Let me update it to check for the `.git` directory explicitely.

jeremiah-c-leary, 2021-02-11:
Hey @imd1 ,

I updated the permission tests to not run if the user is root or has sudo priviliges.

Emilv2, 2021-02-11:
Sorry for that, I didn't think of running the tests with root. For the fix, I think using the `unittest.skipIf` would be a slightly better way to deal with it.

imd1, 2021-02-11:
I can confirm that the two permission unit tests are skipped when running the tests as root

imd1, 2021-02-11:
I am happy for this issue to be closed

jeremiah-c-leary, 2021-02-11:
Woot!!

closed 2021-02-11 by jeremiah-c-leary