jeremiah-c-leary/vhdl-style-guide #532: process after a block is not parsert correctly when block does not have closing label

alancanniff, 2021-02-13:
**Environment**
python 3.9.1
Windows 10

VHDL Style Guide (VSG) version: 2.0.0.dev580
Git commit SHA: ga7e47883

**Describe the bug**
vsg errors out on the process after a block statement, when the closing block label is missing. Removing the process or the block and the file is parsed without error. I've disabled all rules in the provided `json.`

**To Reproduce**
Steps to reproduce the behavior:
1. using the vhdl code and json provided below
2. `vsg -c block_test.json -f block.vhd`
4. See error
Error: Unexpected token detected while parsing process_statement @ Line 17, Column 5 in file None
       Expecting : process
       Found     : end

  ```
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity block_test is
end entity block_test;

architecture rtl of block_test is
begin

    b1  : block
    begin
        process
        begin
            wait;
        end process;
    end block;

    process
    begin
        wait;
    end process;

end architecture rtl;

```

```
{
    "rule":{
        "global":{
            "disable": true,
            "indentSize": 4
        }
    }
}
```

**Expected behavior**
if you add the the closing label 
i.e. 'end block b1;'
the file is parsed without error

================================================================================
File:  block.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 0
Total Violations:    0
  Error   :     0
  Warning :     0

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
I've done my best to provide a minimal reproducible test - I believe this is valid VHDL...

jeremiah-c-leary, 2021-02-13:
Hey @alancanniff ,

It looks like my process classifier was being too greedy.  After the first process it was searching for another one.  It found the process keyword after the `end block;` and decided it found another process.

A label for the process would not have caused the error either.

I pushed a fix for this.  When you get a chance, could you double check it is working on your end.

Thanks,

--Jeremy

alancanniff, 2021-02-13:
yep, looks fixed. Thanks

closed 2021-02-13 by alancanniff