jeremiah-c-leary/vhdl-style-guide #534: `case?` keyword causes hang

pkorpine, 2021-03-09:
**Environment**
```
pip install git+https://github.com/jeremiah-c-leary/vhdl-style-guide.git
```

**Describe the bug**
`case?` keyword (VHDL-2008) is not recognized and causes a hang (breaks with ctrl-c).

**To Reproduce**

```bash
cat > sample.vhd <<EOF
library ieee;
  use ieee.std_logic_1164.all;

entity sample is
  port (
    i_d : in    std_logic_vector(1 downto 0);
    o_q : out   std_logic
  );
end entity sample;

architecture rtl of sample is
begin
  proc_1 : process is
  begin
    case? i_d is
      when "1-" =>  o_q <= '0';
      when "00" =>  o_q <= '1';
    end case?;
  end process proc_1;
end architecture rtl;
EOF

vsg -f sample.vhd
```

**Expected behavior**
Treat `case` and `case?` similarly.

**Screenshots**
```shell
$ vsg -f sample.vhd
Error: Unexpected token detected while parsing process_statement @ Line 19, Column 9 in file None
       Expecting : process
       Found     : case?
^CProcess SpawnPoolWorker-11:
Process SpawnPoolWorker-10:
Process SpawnPoolWorker-12:
Process SpawnPoolWorker-8:
Process SpawnPoolWorker-7:
Process SpawnPoolWorker-5:
Process SpawnPoolWorker-9:
Process SpawnPoolWorker-6:
Process SpawnPoolWorker-13:
Process SpawnPoolWorker-4:
Process SpawnPoolWorker-3:
Process SpawnPoolWorker-1:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.9/3.9.2_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/multiprocessing/pool.py", line 853, in next
    item = self._items.popleft()
IndexError: pop from an empty deque

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/pekka/Library/Caches/pypoetry/virtualenvs/fpga-PAFWn192-py3.9/bin/vsg", line 8, in <module>
    sys.exit(main())
  File "/Users/pekka/Library/Caches/pypoetry/virtualenvs/fpga-PAFWn192-py3.9/lib/python3.9/site-packages/vsg/__main__.py", line 412, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "/usr/local/Cellar/python@3.9/3.9.2_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/multiprocessing/pool.py", line 858, in next
    self._cond.wait(timeout)
  File "/usr/local/Cellar/python@3.9/3.9.2_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
KeyboardInterrupt```

**Additional context**
Add any other context about the problem here.

jeremiah-c-leary, 2021-03-09:
Hey @pkorpine ,

I looked in my classifier and it says it handles the ?.  I suspect my tokenizer is combining the `case` and `?` into a single token instead of splitting them into two.

You could validate it by changing `case?` to `case ?`.

I will see if I can get a fix in tonight.

Thanks,

--Jeremy

pkorpine, 2021-03-09:
Thanks Jeremy. I don't have access to the spec, but at least Ashenden's book "VHDL-2008: Just the New Stuff" presents it concatenated:

[image]

jeremiah-c-leary, 2021-03-09:
Hey @pkorpine ,

I just pushed an update to this issue.  When you get a chance, could you run it on your end and let me know if it fixes the issue for you.

Thanks,

--Jeremy

jeremiah-c-leary, 2021-03-10:
Awesome...I will close the issue.

--Jeremy

closed 2021-03-10 by jeremiah-c-leary