jeremiah-c-leary/vhdl-style-guide #536: How to align => in assignments?

pkorpine, 2021-03-11:
Is it possible to align `=>` in signal/variable assignments similar to port/generic maps? I could not find a suitable rule for this. With default configuration, `=>` are not aligned at all, and I could not find a rule or configuration to change this.

My target:
```vhdl
    s_foo <= (
        item         => 12,
        another_item => 34
    );
```

jeremiah-c-leary, 2021-03-11:
Hey @pkorpine ,

You are correct, I do not have a rule to align `=>` assignments.   I am currently updating rule `concurrent_003` on the issue-499 branch.

I will schedule this for release 3.1.0, but it may make it into release 3.0.0.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-07-04:
Hey @pkorpine,

Well, it has taken me far too long to get to this issue.

The good news is I have pushed an update to the issue-536 branch.  When you get a chance could you check it out and let me know if it addresses your issue?

Thanks,

--Jeremy

pkorpine, 2022-07-07:
Sure, of course.

I'll get this error though (pip install from commit d28a5f91808478c7c551173f8ec0dc094d0a2bc2).

```
❯ vsg -c vsg.yaml
multiprocessing.pool.RemoteTraceback:
"""
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.9/3.9.10/Frameworks/Python.framework/Versions/3.9/lib/python3.9/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/foo/.venv/lib/python3.9/site-packages/vsg/apply_rules.py", line 64, in apply_rules
    oVhdlFile = vhdlFile.vhdlFile(lFileContent, sFileName, eError)
  File "/foo/.venv/lib/python3.9/site-packages/vsg/vhdlFile/vhdlFile.py", line 64, in __init__
    self._processFile()
  File "/foo/.venv/lib/python3.9/site-packages/vsg/vhdlFile/vhdlFile.py", line 95, in _processFile
    set_aggregate_tokens(self.lAllObjects)
  File "/foo/.venv/lib/python3.9/site-packages/vsg/vhdlFile/vhdlFile.py", line 546, in set_aggregate_tokens
    if isinstance(lTokens[iIndex], token.aggregate.open_parenthesis):
AttributeError: module 'vsg.token' has no attribute 'aggregate'
"""

```

jeremiah-c-leary, 2022-07-08:
Hey @pkorpine,

I was able to replicate the error you encountered and pushed an update.

If you could try again, I would appreciate it.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-07-15:
Hey @pkorpine,

Just wanted to ping you on this issue and whether the latest update is working for you.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-07-23:
Hey @pkorpine,

Just wanted to ping you to see if you had a chance to checkout my latest update.

Thanks,

--Jeremy

pkorpine, 2022-08-27:
Sorry for the late reply, I was AFK.

Tested with a2747ab74ae276f6101ecadf4e170ed3bf0da8bd. All good!

jeremiah-c-leary, 2022-09-24:
Hey @pkorpine,

No problem.  I will get this merged to master.

--Jeremy

PR #793 (2022-09-25): Issue 536
Resolves #536
closed 2022-09-25 by jeremiah-c-leary