jeremiah-c-leary/vhdl-style-guide #537: Crash on comment alignment

pkorpine, 2021-03-11:
**Environment**

```sh
pip install git+https://github.com/jeremiah-c-leary/vhdl-style-guide.git
```

**Describe the bug**
VSG crashes on the following structure. Note that this is just a minimal version that can reproduce the bug.

Note that the crash does not happen if you:
1. remove any of the lines in the body
2. align manually the comment `-- B` 
3. add a line before `--A`
4. add a line before `s <= cc...`

```vhdl
architecture a of e is
begin
  s <= a;
  -- A
  s <= b; -- B
  s <= cc; -- C
end architecture a;
```

**To Reproduce**
```sh
cat > sample.vhd <<EOF
architecture a of e is
begin
  s <= a;
  -- A
  s <= b; -- B
  s <= cc; -- C
end architecture a;
EOF

vsg -f sample.vhd
```

**Expected behavior**
No crash.

**Screenshots**
```
multiprocessing.pool.RemoteTraceback:
"""
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.9/3.9.2_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/Users/pekka/Library/Caches/pypoetry/virtualenvs/fpga-PAFWn192-py3.9/lib/python3.9/site-packages/vsg/__main__.py", line 465, in apply_rules
    oRules.fix(commandLineArguments.fix_phase, commandLineArguments.skip_phase, fix_only)
  File "/Users/pekka/Library/Caches/pypoetry/virtualenvs/fpga-PAFWn192-py3.9/lib/python3.9/site-packages/vsg/rule_list.py", line 149, in fix
    oRule.fix(self.oVhdlFile, dFixOnly)
  File "/Users/pekka/Library/Caches/pypoetry/virtualenvs/fpga-PAFWn192-py3.9/lib/python3.9/site-packages/vsg/rule.py", line 98, in fix
    self._fix_violation(oViolation)
  File "/Users/pekka/Library/Caches/pypoetry/virtualenvs/fpga-PAFWn192-py3.9/lib/python3.9/site-packages/vsg/rules/concurrent/rule_008.py", line 172, in _fix_violation
    if isinstance(lTokens[iTokenIndex - 1], parser.whitespace):
IndexError: list index out of range
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/pekka/Library/Caches/pypoetry/virtualenvs/fpga-PAFWn192-py3.9/bin/vsg", line 8, in <module>
    sys.exit(main())
  File "/Users/pekka/Library/Caches/pypoetry/virtualenvs/fpga-PAFWn192-py3.9/lib/python3.9/site-packages/vsg/__main__.py", line 412, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "/usr/local/Cellar/python@3.9/3.9.2_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/multiprocessing/pool.py", line 870, in next
    raise value
IndexError: list index out of range
```

jeremiah-c-leary, 2021-03-11:
Hey @pkorpine ,

I tried your example file:
```vhdl
  1 architecture a of e is
  2 begin
  3   s <= a;
  4   -- A
  5   s <= b; -- B
  6   s <= cc; -- C
  7 end architecture a;
```
and ran it with the latest on master:
```mono
================================================================================
File:  example.orig.vhd
================================================================================
Phase 3 of 7... Reporting
Total Rules Checked: 259
Total Violations:    4
  Error   :     4
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  architecture_015          | Error      |          1 | Insert blank line below
  architecture_016          | Error      |          2 | Insert blank line above
  architecture_017          | Error      |          2 | Insert blank line below
  architecture_018          | Error      |          7 | Insert blank line above
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
I changed the file to this:
```vhdl
  1
  2 architecture a of e is
  3
  4 begin
  5
  6   s <= a;
  7   -- A
  8   s <= b;  -- B
  9   s <= cc; -- C
 10
 11 end architecture a;
```
and it went through fine:
```mono
================================================================================
File:  example.mod.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 520
Total Violations:    0
  Error   :     0
  Warning :     0
```
Could you give me the version you are running:
```mono
$ vsg --version
VHDL Style Guide (VSG) version: 2.0.0.dev636
Git commit SHA: ga97d9f09
```
Are you using a configuration?

pkorpine, 2021-03-11:
I'm running my own fork as I needed to have recursive glob, i.e. '**/*.vhd'. It's based on a97d9f092f4f175c46eff27e0b8f8cbd71de512c. I'll try again against your master.

pkorpine, 2021-03-11:
Ah.. You need to specify the `--fix`. I'll correct the report.

jeremiah-c-leary, 2021-03-11:
I was able to reproduce the crash by adding the `--fix`

jeremiah-c-leary, 2021-03-11:
Hey @pkorpine ,

I believe it found the issue.  The rule was losing track of which line it was on.

I pushed an update.  When you get a chance, could you validate it?

Thanks,

--Jeremy

pkorpine, 2021-03-12:
Not crashing anymore - tested using 6ad423b7a56fb9b4c6b01703a36a7ab5f845d4e9.
Excellent!

jeremiah-c-leary, 2021-03-12:
Awesome...I will close this issue.

closed 2021-03-12 by jeremiah-c-leary