jeremiah-c-leary/vhdl-style-guide #540: Broken links in documentation

staerz, 2021-03-23:
**Describe the bug**
The link to "Configuring Prefix and Suffix Rules" in [process_036](https://vhdl-style-guide.readthedocs.io/en/latest/process_rules.html#process-036) is broken (missing "-and-suffix-rules" in link).

(VSG 3.0.0)

jeremiah-c-leary, 2021-03-23:
Hey @staerz ,

Documentation, my Achilles heel.
 
I will get this in the 3.1.0 release.

jeremiah-c-leary, 2021-03-24:
Hey @staerz,

I pushed the documentation updates.

I ended up splitting the prefix and suffix configuration into individual prefix and suffix configurations.

Let me know what you think.

--Jeremy

staerz, 2021-03-24:
Hi @jeremiah-c-leary,

yes, splitting prefixes and suffixes is an option that I also considered but didn't propose as I valued more the idea to have a direct comparison of the 2 options right next to each other.
If you see your current options (rules), you'll see that for now they are asymmetric what concerns prefix vs. suffix, i.e. some rules only exist in the prefix version.

I'm not sure how many users would use e.g. suffixes for their constants, signals or variables (e.g. we don't at all), but maybe then it could be an idea to simply create those rules as well and then having the 2 sections separated (with the similar list of rules) would be fine.

Please also note that I'll use this issue to add further small observations that I might find as we're now finally migrating to VSG 3.0.0 (I have done it for a central part of our code base while we were iterating over all these issues towards 3.0.0, but now also my colleagues apply those rules to their respective code).

jeremiah-c-leary, 2021-03-24:
> If you see your current options (rules), you'll see that for now they are asymmetric what concerns prefix vs. suffix, i.e. some rules only exist in the prefix version.

I could recombine them and put in a table like this:

```mono
element                  prefix_rule    suffix_rule
-------------------------------------------------------
package identifier        package_003     package_004
signal identifier         signal_005          NA
process label             process_023         NA
```
That would indicate the element that has a prefix and/or suffix rule and then list the prefix and suffix rule for that element.
The rules would also be links.

staerz, 2021-03-24:
Yes, this is exactly what I had in mind. As usual, have the rules be a link to them and that's it.

jeremiah-c-leary, 2021-03-24:
Hey @staerz ,

I just pushed an update for the prefix and suffix rules.

I do agree it looks better in a table.

staerz, 2021-03-25:
Hi @jeremiah-c-leary,

I had a look and I also think now in a table it looks much more organised.

A few details that I still would change:
- The headline: Configuring Prefix Suffix Rules &rarr; Configuring Prefix and Suffix Rules (watch out to update the link to this section everywhere correctly ... I know it's trivial to miss)
- You always use the slash ("prefixes/suffixes") which I think makes hick-up reading a bit. I suggest you use "or" or "and" in the following:
   - "There are several rules that enforce specific prefixes **or** suffixes"
   - "what the default prefixes **and** suffixes"
   - "All prefix **and** suffix rules are disabled by default."
- You usually capitalise your headlines, but not these one: "Rules enforcing prefixes and suffixes"
- Also there, the first row in the table should be a headline (bold)
- In that table:
   - "Generic Name" &rarr; "Generic Identifier"
   - "Port Name" &rarr; "Port Identifier"

That's all I have to say on the documentation of the prefixes and suffixes :wink:

jeremiah-c-leary, 2021-03-26:
Hey @staerz ,

I pushed an update for the documentation.

staerz, 2021-03-26:
Hi @jeremiah-c-leary,

thanks for the update. It reads much smoother now with the slash removed :+1:

staerz, 2021-03-26:
Hi @jeremiah-c-leary,

I just went over the issue description and marked those sections treated which are by now.
That is in fact all of them.

If you agree, I would though still keep the issue open let's say for another week (or two) as some other minor issues might be reported to me by the developers of my group.

As you have seen, I have opened dedicated issues if it's not documentation related.

jeremiah-c-leary, 2021-03-26:
Hey @staerz,

I want to release a 3.1.0 for some of the bugs you and others have found in my parser, but we can leave this issue open since it just covers documentation.  I will just move it to the 3.2.0 project after I release 3.1.0.

staerz, 2021-03-31:
Hi @jeremiah-c-leary,

I just found another thing that intrigued me: Violation of `length_001` results in a warning (I think we agreed that this could be a good idea) but that seems not to be documented...

jeremiah-c-leary, 2021-03-31:
Hey @staerz,

> I just found another thing that intrigued me: Violation of length_001 results in a warning (I think we agreed that this could be a good idea) but that seems not to be documented...

I was also starting to think about how to document rules that are off by default.  Should there be a separate section in the documentation that lists the rules off by default?  We could also add a list of rules that are warnings by default.

>And now as I think of it with a little distance, it would possibly be a good idea if such rules (that only produce a warning) would have an option to make it an error. What do you think?

You can change the severity level of any rule, and create your own severity levels:  https://vhdl-style-guide.readthedocs.io/en/latest/rule_severity.html

staerz, 2021-03-31:
Hi @jeremiah-c-leary,

thanks for the pointer, I think I saw that but entirely forgot about it :rofl:

So yes, that answers my second question fully.

Found a typo in the severity level doc: "it is based on.. It will" ... two dots.

To your question on the list of by default disabled rules or those that are warnings, yes, I think that would make it easier to find them.
Now I'm just wondering where this would be put best.

Maybe the following could be a good idea: In the configuration section, there you add a new subsection giving those 2 lists.
In the Rules main section you just add one short sentence that not all rules are enabled by default and also some rules are just throwing warnings - and you link that to the configuration section where one would find the details.
And the link to the severity level (configuration) should then also figure there.

jeremiah-c-leary, 2021-04-01:
Hey @staerz,

Could you look at the documentation on read-the-docs.  I implemented icons with links.

I also fixed the double period issue.

staerz, 2021-04-01:
Hi @jeremiah-c-leary,

I see, you have 1 icon for the phase, 1 for the warning and 1 for default disabled rules.

I like that.

There's a big **BUT**: I think although it may appear nice to directly see it in the headline, it is not a very good idea to have it there due to the fact how readthedocs works: This has direct impact on the link to that rule.
I.e. the link to the rule changes when ever you decide to move a rule to a different phase, make it default, or change the severity level.
That's really not a good idea not only because you'd have to change all your links within your documentation, but also all users who might point to certain rules (I do that for example) also have to update their documentation.

So I would highly recommend to have those icons on the first row, but not on the headline.

jeremiah-c-leary, 2021-04-01:
Hey @staerz,

> There's a big BUT: I think although it may appear nice to directly see it in the headline, it is not a very good idea to have it there due to the fact how readthedocs works: This has direct impact on the link to that rule.

I had to add in anchors to make the links work.

I pushed an update where I put the icons on the next line.  Let me know what you think versus having them on the same line as the rule header.

staerz, 2021-04-01:
Hi @jeremiah-c-leary,

yeah, I think that's better.

Of course, it would be ideal if they were on the same line as the header, but maybe right-aligned (real headline on the left - space - icoons on the right), but if that influences how readthedoc creates the anchors for the headlines I'd not go for it and keep things as you have them now.

Of course, you're free to play around as I think you understood my main concern - and you're way more expert for readthedocs than I am (I wouldn't know if that's possible with (GitLab flavoured) Markdown). HTML-wise (as this is what it always is ultimately) one could of course always inject a table with 2 columns where the left is the headline and the right is the labels, but that would utterly overcomplicate the solution I think - so in the end I'd rather go with the simpler syntax even if the price is that you don't get 100% of what you'd ideally wish for.

jeremiah-c-leary, 2021-04-01:
> right-aligned (real headline on the left - space - icoons on the right)

That would like nice.  Although I do not believe restructured text would support it.

I will work on updating the documentation with the icons.  This should also address issue #488 .

jeremiah-c-leary, 2021-04-02:
Hey @staerz,

I added icons to all the rules.  Let me know what you think.

--Jeremy

staerz, 2021-05-31:
> Hey @staerz,
> 
> I added icons to all the rules. Let me know what you think.
> 
> --Jeremy

Hi @jeremiah-c-leary ,

sorry, I lost a little track of VSG during recent weeks.

I'm just coming back to it due to a fancy GitLab feature that I just came across (and created #596 ).

This issue here is fully addressed and we can close it. As always I leave the pleasure to you.

Thanks!

jeremiah-c-leary, 2021-06-01:
Awesome, closing the issue.

closed 2021-06-01 by jeremiah-c-leary