jeremiah-c-leary/vhdl-style-guide #542: Fix comment_010 to not complain on block comments

staerz, 2021-03-23:
**Environment**
VSG 3.0.0

**Describe the bug**
The rule [`comment_010`](https://vhdl-style-guide.readthedocs.io/en/latest/comment_rules.html#comment-010) mistakenly kicks in for block comments (covered by [Block Comment Rules](https://vhdl-style-guide.readthedocs.io/en/latest/block_comment_rules.html#block-comment-rules).

**To Reproduce**
Have an architecture with a block comment in it:
```
architecture something of something_else is

begin

--+-------------[ Header ]==============
--|  Comment 1
--|  Comment 2
----------------------------------------
```

Steps to reproduce the behavior:
1. Run VSG (with default settings on block comments)
2. Observe that `comment_010` kicks in for the lines containing "Comment 1"  and "Comment 2"

**Expected behavior**
I expect that `comment_010` would never complain about a line covered by a block comment.

jeremiah-c-leary, 2021-03-24:
Hey @staerz,

We must have missed something with the block comments.  The block comment rules do not have the ability to force left alignment.  I could possibly add an option to the existing block comment rules or add another rule to allow forcing the block comments to the left.

staerz, 2021-03-24:
Hi @jeremiah-c-leary,

yes, that's what I figured. Or let's say, I was really not sure why sometimes we get the complaint and I just realised yesterday that I had `comment_010` still disabled.

But concerning the options, wasn't `allow_indenting` meant to cover this? So when I initially proposed this block comment, my intention was that a block comment would always start at column 0 (left alignment). If you ask me, that's only what makes sense as the header has a very well defined length. If you indent this, the line gets longer and longer ... a bit inconsistent, isn't it?

jeremiah-c-leary, 2021-03-24:
Hey @staerz ,

> But concerning the options, wasn't allow_indenting meant to cover this?

That does sound familiar.  I need to think a little bit on how to handle the indenting.

jeremiah-c-leary, 2021-03-24:
Hey @staerz ,

I think I have a fix for this issue.  Could you try it on your end.

Thanks,

--Jeremy

staerz, 2021-03-25:
Hi @jeremiah-c-leary,

I tried your update and there's still sth. wrong. It looks like your indentation for it is not correct.

In a port definition I make use of preprocessor commands and there VSG is picking up the indent:
```vhdl
#endif
#ifdef G_EMIF_DDR3_ENABLE
    -- vsg_off length_001
```
so it want to have
```vhdl
#endif
#ifdef G_EMIF_DDR3_ENABLE
-- vsg_off length_001
```

Having had a look at other complaints I think that the problem is that you use a look-behind for guessing the indentation level which is not correct as comments are aligned with the code ahead of them, so you need to look-ahead.
Additionally, as given in the example above, you need to ignore preprocessor commands.

jeremiah-c-leary, 2021-03-26:
Hey @staerz ,

Preprocessor commands are completely ignored.  From your description I imagine a code snippet would look like this:
```vhdl
  1
  2 entity fifo is
  3   port (
  4     I_CLK : in std_logic;
  5 #ifdef G_EMIF_DDR3_ENABLE
  6 -- vsg_off length_001
  7
  8 #endif
  9     I_RST_F : in std_logic
 10   );
 11 end entity fifo;
```
To VSG this looks like:
```vhdl
1
  2 entity fifo is
  3   port (
  4     I_CLK : in std_logic;
  5
  6 -- vsg_off length_001
  7
  8
  9     I_RST_F : in std_logic
 10   );
 11 end entity fifo;
```
So I could see why the comment was pushed to the right.

In the documentation there is a section on [pragmas](https://vhdl-style-guide.readthedocs.io/en/latest/pragmas.html).  Would you want the same behavior applied to preprocessor commands?

staerz, 2021-03-26:
Hi @jeremiah-c-leary,

it is the opposite way: the comment was pushed to the left. The original comment is indented with the next port being defined (I did not add it in the example)  and I would expect VSG not to touch it.

But I think the conclusion might still be the same: Yes, VSG should ignore any preprocessor command (anything starting with hash-tag). In particular, it should not see that line at all rather seeing it as a blank line (as also blank lines affect VSG as they may or may not be desired here or there as defined by certain rules).

Note that doxygen actually supports these preprocessor commands (when configured such) and hence I think it would only  be fair if VSG did also :)

jeremiah-c-leary, 2021-03-26:
I used this snippet:
```vhd
  1
  2 entity fifo is
  3   port (
  4     i_clk : in    std_logic;
  5 #ifdef G_EMIF_DDR3_ENABLE
  6     -- vsg_off length_001
  7
  8 #endif
  9     i_rst_f : in    std_logic
 10   );
 11 end entity fifo;
```
and ran with the `--fix` option and it did not move the comment to the left.

There must be something missing in my example.

staerz, 2021-04-07:
OK, so as the problem persists, maybe a little more context is needed:

The one file I have is just a long port list, intersected by these preprocessor commands. The parts where VSG `comment_010` complains are:

```vhdl
#ifdef G_ENET_ENABLE
    --! @name Ethernet (via SGMII to external Marvell 88E1111 PHY), controlled by G_ENET_ENABLE

    --! Ethernet Management bus data clock
    enet_mdc            : out   std_logic;
    --! Ethernet Management bus data
    enet_mdio           : inout std_logic;
    --! @brief Ethernet Management bus interrupt
    --! @todo golden top has out, but example has input
    enet_intn           : in    std_logic;
    --! Ethernet Device reset
    enet_resetn         : out   std_logic;
    --! Ethernet SGMII TX Data
    enet_tx_p           : out   std_logic;
    --! Ethernet SGMII RX Data
    enet_rx_p           : in    std_logic;
#endif
#if G_EMIF_DDR3_ENABLE || G_EMIF_DDR4_ENABLE
    -- vsg_off length_001
    --! @name EMIF (HiLo External Memory Interface: DDR3/DDR4 x72), controlled by G_EMIF_DDR3_ENABLE or G_EMIF_DDR4_ENABLE
    --! @todo may need changing of vector size to accommodate I/O bank standards
    --! @details Port definitions follow the EMIF user guide naming convention,
    --! except for the clock port(s) which are 'mem_clk' instead of 'mem_ck'.
    --! Port references can be found in Sections 4.1.1.9 (p. 54) and 4.1.1.10 (p. 55) of
    --! <a href="https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/hb/stratix-10/ug-s10-emi.pdf">the EMIF User Guide</a>.
    -- vsg_on length_001

    --! DDR interface Calibrated On-Chip Termination (OCT) RZQ input
    oct_rzqin           : in    std_logic;
    --! DDR interface Address and Command Input clock
    mem_clk_p           : out   std_logic;
```
complaining about `    -- vsg_off length_001`;

```vhdl
#ifdef G_FMCA_IM_V2_0_ENABLE
    --! INTERMEZZO DAC6311 converter SPI signals, data
    fmca_dac_spi_data   : out   std_logic;
    --! INTERMEZZO DAC6311 converter SPI signals, syncb
    fmca_dac_spi_syncb  : out   std_logic;
    --! INTERMEZZO DAC6311 converter SPI signals, clk
    fmca_dac_spi_clk    : out   std_logic;
#endif
#ifdef G_PCIE_ENABLE
    --! @name PCIe (x16 PCI Express, 3-slot long form factor), controlled by G_PCIE_ENABLE

    --! PCIe Reset
    pcie_perstn         : in    std_logic;
    --! PCIe receive bus
    pcie_rx_p           : in    std_logic_vector(15 downto 0);
    --! PCIe transmit bus
    pcie_tx_p           : out   std_logic_vector(15 downto 0);
    --! @brief PCIE SMB clock:
    --! wired but not required for normal operation
    --! @todo golden top has inout, example has input
    pcie_smbclk         : in    std_logic;
    --! @brief PCIE SMB data:
    --! wired but not required for normal operation
    pcie_smbdat         : inout std_logic;
    --! @brief PCIE Wake Signal
    --! @todo golden top has output, example has input
    pcie_waken          : in    std_logic;
    --! PCIe status LED (x1)
    pcie_led_x1         : out   std_logic;
```
complaining about `    --! @brief PCIE SMB clock:`;

and finally
```vhdl
#ifdef G_DP_ENABLE
    --! @name Display Port, controlled by G_DP_ENABLE
    --! @todo Get the missing port descriptions

    --! Display Port transceiver data out
    dp_ml_lane_p        : out   std_logic_vector(3 downto 0);
    --! DP: todo
    dp_hot_plug         : in    std_logic;
    --! DP: todo
    dp_aux_tx_drv_out   : out   std_logic;
    --! DP: todo
    dp_aux_tx_drv_in    : in    std_logic;
    --! DP: todo
    dp_aux_tx_drv_oe    : out   std_logic;
    --! DP: todo
    dp_config1          : inout std_logic;
    --! DP: todo
    dp_config2          : inout std_logic;
    --! DP: todo
    dp_aux_ch_p         : in    std_logic;
    --! DP: todo
    dp_aux_ch_n         : in    std_logic;
#endif
    --! @name User buttons
    --! When (pressing/releasing) the button, the device pin is set to logic (0/1).
    --! When the switch is in the (ON/OFF) position, logic (0/1) is selected.
    --! @{

    --! User-defined Push Buttons.
    user_pb             : in    std_logic_vector(2 downto 0);
    --! User-Defined DIP Switches.
    user_dipsw          : in    std_logic_vector(3 downto 0);
    --! @}

    --! @name User-Defined LEDs.
    --! The LEDs (illuminate/are off) when a logic (0/1) is driven.
    --! @{

    --! Red LEDs
    user_led_r          : out   std_logic_vector(3 downto 0);
    --! Green LEDs
    user_led_g          : out   std_logic_vector(3 downto 0);
    --! @}

    --! Virtual pins
    virtual             : out   std_logic_vector(G_VIRTUAL_W - 1 downto 0)
  );
```
complaining about `    --! @name User buttons` **and** `    --! @name User-Defined LEDs.`.

From a different file, I get complaints on the first line of comment on that construct:
```vhdl
  signal cpu_resetn         : std_logic;
  --! @name Transceiver Reference Clocks
  --
  --! FPGA clock for Bank 3L, 100MHz (Programmable via Si5338)
  signal clk_fpga_b3l_p     : std_logic;
```
Here I agree, it's actually the line `  --` that shouldn't be there but I put them in order to keep the alignment of the colon together with the rest of the following signal declarations (I know, it's a configuration of `architecture_026` in fact, but still).

There are more examples in my code that make `comment_010` pop up, but I'd say let's first see if you can make sense of that already.

jeremiah-c-leary, 2021-04-07:
Hey @staerz ,

I think I figured this one out.  I was always setting the indent of the first line in a comment block to 0 even if it was not a block comment.

The fix was pushed to the issue-542 branch.

When you get a chance, could you let me know if it solves the issue on your end.

Thanks,

--Jeremy

staerz, 2021-04-07:
I just checked it.

Now I get a big bunch of complaints from `block_comment_001`.

In particular it's miscalculating now the size (length) of the header, one char is missing each left and right, so instead of
```vhdl
------------------------------<-    80 chars    ->------------------------------
```
VSG want's me to put
```vhdl
-----------------------------<-    80 chars    ->-----------------------------
```
while this is 78 chars, not 80 :wink:

Could you double-check your counting/init value/math here?

jeremiah-c-leary, 2021-04-07:
yeah, I will check on it.

jeremiah-c-leary, 2021-04-08:
Hey @staerz,

I have this snippet:
```vhdl
  1
  2 ------------------------------<-    80 chars    ->------------------------------
  3 --! Comment
  4 --! Comment
  5 --------------------------------------------------------------------------------
  6
  7 architecture rtl of fifo is
  8
  9 ------------------------------<-    80 chars    ->------------------------------
 10 --! Comment
 11 --! Comment
 12 --------------------------------------------------------------------------------
 13
 14 begin end architecture rtl;
```
where the block comment header and footer are 80 characters long.

I have this configuration:
```yaml
  1 rule:
  2     block_comment_001:
  3         disable : False
  4         allow_indenting : False
  5         header_left : '-'
  6         header_left_repeat : '-'
  7         header_string : '<-    80 chars    ->'
  8         header_right_repeat : '-'
  9         max_header_column : 80
 10     block_comment_002:
 11         disable : False
 12         allow_indenting : False
 13         comment_left : '!'
 14     block_comment_003:
 15         disable : False
 16         allow_indenting : False
 17         footer_left : '-'
 18         footer_left_repeat : '-'
 19         max_footer_column : 80
```
I run with this command: `vsg -c config.yaml -f snippet.vhd` and I get:
```mono
================================================================================
File:  snippet.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 524
Total Violations:    0
  Error   :     0
  Warning :     0
```
If I change the `allow_indenting` to `True` and re-run I get:
```mono
================================================================================
File:  snippet.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 85
Total Violations:    2
  Error   :     2
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  block_comment_001         | Error      |          9 | Change block comment header to : -----------------------------<-    80 chars    ->-----------------------------
  block_comment_003         | Error      |         12 | Change block comment footer to : ------------------------------------------------------------------------------
----------------------------+------------+------------+--------------------------------------
```

It does not flag either block comment when the `allow_indenting` option is `False`.   It does flag the second block comment when the `allow_indenting` option is `True`.  This is behavior I would expect.

Could you give me more context around the lines being flagged?

I am running with vsg version:
```mono
VHDL Style Guide (VSG) version: 3.0.0.dev31
Git commit SHA: g9d4d2969
```

--Jeremy

It appears to be detecting

jeremiah-c-leary, 2021-09-18:
Hey @staerz ,

let me know when you want to pick this issue back up.

staerz, 2022-02-09:
Hi @jeremiah-c-leary,

I lost track of this one! But yes, in my config, `comment_010` is disabled for the reason it still being buggy.

I just ran with the latest master (3.7.0-5-g1d75e565) re-enabling it.
My understanding from a quick look into the history is that the patches that you applied are already in master. Correct me if I miss anything!

With that I still ran into a few instances, where a block comment was taken for a comment:

```vhdl
< block comment on the file >

< another block comment on the file >

< library call >
< context call>

< entity definition >

--! Implementation of the avst_packetiser
architecture behavioral of avst_packetiser is

< commented signal declarations here >

begin

------------------------------<-    80 chars    ->------------------------------
--! FPGA RX to Avalon-ST TX interface
--! to make it really stable, two cycles are required to produce the eof signals
--------------------------------------------------------------------------------

  avst_stream_rdy_o <= avst_packet_rdy_i;

< more code (in particular a commented process starts here) >
```

Here, the first 3 of the 4 lines of the block comment are listed as violation to `comment_010` ("Indent level 1").

The other instance is this:
```vhdl
< library call and context call >

< architecture declaration with all signals etc. >

begin

< a bunch of blocks and instances >

  end block blk_map_leds;

------------------------------<-    80 chars    ->------------------------------
--! Driving unused ports of the design
--------------------------------------------------------------------------------

  clk_si5395_out_p  <= '0';
  ffly_levelshifter <= (others => '0');
  ffly_g1_sda       <= '0';
  ffly_g1_scl       <= '0';
  ffly_g2_sda       <= '0';
  ffly_g2_scl       <= '0';

end architecture fpga_lasp_top;
```
In this instance I get a complaint on all 3 lines (Indent level 1).

The 3rd instance is similarly in an architecture, so like the 2nd case, but again it doesn't complain about the last line (all dashes).

Another instance has a block comment in the architecture section and there only the lines with the actual comments are complained about.

Other instances hit when the block comment is in the port list of an entity definition, but then not the block comment header is mocked on.

While going over all these remaining errors, I think that the indent for where to expect the comment (following regular comments) still needs refinement:

```vhdl
  gen_delay : case D_DEPTH generate
    -- No delay of #sig_i
    when 0 =>
      delay_reg(0) <= sig_i;
    -- Delay #sig_i by one clock cycle
    when 1 =>
```
I get a complaint about both comments here although I'd say they are where they should be but VSGs requests "Indent level 3".

This is my feedback. Is there any cross-relation with the settings for the block comments that `comment_010` would take into account?

staerz, 2022-02-09:
Ah, and to add: I have observed similar behaviour when I comment in the library section. VSG always wants me to put the comment 1 indent further than I'd put it. This is the look-ahead thing that we discussed already at a different point (but I forgot where).

jeremiah-c-leary, 2022-02-10:
hmm...So this is a little confusing.

I have this test file:
```vhdl
  1
  2 architecture rtl of fifo is
  3
  4 begin
  5
  6 ------------------------------<-    80 chars    ->------------------------------
  7 --! FPGA RX to Avalon-ST TX interface
  8 --! to make it really stable, two cycles are required to produce the eof signals
  9 --------------------------------------------------------------------------------
 10
 11 end architecture rtl;
```
with this configuration:
```yaml
  1 rule:
  2     block_comment_001:
  3         disable : False
  4         allow_indenting : False
  5         header_left : '-'
  6         header_left_repeat : '-'
  7         header_string : '<-    80 chars    ->'
  8         header_right_repeat : '-'
  9         max_header_column : 80
 10     block_comment_002:
 11         disable : False
 12         allow_indenting : False
 13         comment_left : '!'
 14     block_comment_003:
 15         disable : False
 16         allow_indenting : False
 17         footer_left : '-'
 18         footer_left_repeat : '-'
 19         max_footer_column : 80
```
...and I get this result:
```mono
$ vsg -c config.yaml -f example.vhd
================================================================================
File:  example.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 599
Total Violations:    0
  Error   :     0
  Warning :     0
```
If add a single space to line 6 and re-run I get the following:
```mono
$ vsg -c config.yaml -f example.vhd
================================================================================
File:  example.vhd
================================================================================
Phase 4 of 7... Reporting
Total Rules Checked: 390
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  comment_010               | Error      |          6 | Indent level 0
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
I do not understand why we get different results.

It looks like I am ahead of you on master by 6 commits, but none of those commits should have made a difference.

Could you try my example with my configuration and see if you get the same results I do?

--Jeremy

staerz, 2022-02-10:
Hi @jeremiah-c-leary,

I have some good and some bad news!

Thanks for posting your configuration again. By comparing with my configuration I saw that I hadn't set `allow_indenting` to `false` in `block_comment_002` and **that makes all the difference**!

If it is disabled, then the line is mistakenly taken as a comment and complained on by `comment_010`. Is that something that you would expect (say it's intentional) then?

This was basically my question if comment and block comment rules have some hidden internal dependency. I think the answer now is: Yes, they have, and more precisely: It's very well hidden :rofl:

What I would suggest is that in the documentation for comments and block comments, we add a note that if the configuration is not properly set up, block comments could be taken as comments. (Or you check in the code if there's sth. you could do)

So that's the good news which made 90% of all the false alerts go away.

I'm remaining with 1 single case which is the one that I noted under "The other instance is this:".

The difference to your test file is that although it is a block comment, the comment itself only comprises 3 lines, i.e. only one line of actual comment. I noted that as soon as I add another line, even as simple as `--`, the complaint of `comment_010` disappears and VSG is happy.

In all the documentation on block comments I see that there are always 2 lines of actual comments. This is sth. to look at.

In my code, I also have the following:

```vhdl
------------------------------<-    80 chars    ->------------------------------
-- Map internal signals to ports
--------------------------------------------------------------------------------
  -- Data:
  avst_stream_o.data  <= dout;
  -- Valid:
  avst_stream_o.valid <= vout;

  -- End of File:
  eof <= s_eof;

end architecture behavioral;
```
And guess what, VSG doesn't complain.

But here it comes:
```vhdl
------------------------------<-    80 chars    ->------------------------------
-- Map internal signals to ports
--------------------------------------------------------------------------------

  -- Data:
  avst_stream_o.data  <= dout;
  -- Valid:
  avst_stream_o.valid <= vout;

  -- End of File:
  eof <= s_eof;

end architecture behavioral;
```
by the single (blank) line added after the block comment, I get a complaint on all 3 lines of that block comment which is the exact same behaviour as for the other instance reported above.

So that makes me think that you have a minimal requirement of 4 lines to make a block comment, could that be!?

staerz, 2022-02-14:
Hi @jeremiah-c-leary,

yes, I confirm that now block comments are taken into account already as of being 3 lines long.

I though discovered now a new thing: the `-- vsg_on <rule>` is now also checked, so it seems you may have fiddled with the number of lines to skip upon encountering this directive and are now off by 1, i.e. the line (before that new commit) wasn't reported on, but now is.

In the particular instance we talk about a `-- vsg_on comment_010` of course which does not align with the contextual alignment (so yes, in general the error is correct, but the alignment shouldn't be checked just yet in that line).

I didn't discover any other side effects, but I also didn't play with any configuration.

On a side note, for which I think it would be most appropriate to create a new issue: Did you notice that `comment_010` is a bit quirky in the library and case section? For your illustration, here's a short diff:

```diff
+--! @cond
 library altera;
   use altera.altera_syn_attributes.all;
+  --! @endcond
 
 --! Implementation of the #delay_chain
 architecture behavioral of delay_chain is
@@ -109,10 +111,11 @@ begin
   -- now set the last item in the chain
   --! @cond
   gen_delay : case D_DEPTH generate
-    -- No delay of #sig_i
+      -- No delay of #sig_i
     when 0 =>
       delay_reg(0) <= sig_i;
-    -- Delay #sig_i by one clock cycle
+      -- Delay #sig_i by one clock cycle
     when 1 =>
```

In the first section (libraries), I just added some Doxygen directive, but I had to indent the `endcond` one more level than I'd usually do, and in the second instance you directly see that I also needed to change the indent to 1 more than expected (the added blank line is due to you new case rules...).

Would you want to treat that issue directly here (as you're fiddling with `comment_010`) or should I open a new one?

jeremiah-c-leary, 2022-02-14:
Hey @staerz,

I will address the `-- vsg ` issues here.

> Would you want to treat that issue directly here (as you're fiddling with comment_010) or should I open a new one?

`comment_010` is affected by some other rules which are trying to align comments based on specific patterns.  If you could open a new ticket we could use it as a vehicle to take another look at how the rules interact.

staerz, 2022-02-14:
OK, there we have it and coincident has it that it's exactly 200 issues apart :rofl:

jeremiah-c-leary, 2022-02-16:
Hey @staerz,

I pushed another update for this to handle code tags.  We did not talk specifics, so try this one out and let me know if I got it correct.

Thanks,

--Jeremy

staerz, 2022-02-16:
Hi @jeremiah-c-leary,

I just tried and it doesn't make a difference.

I looked at what you changed and would make the following comment, although I'm not fully sure if I read your code correctly: I do not think that all `-- vsg_<something>` should be considered as an exception, it should in fact only be the directive for `comment_010` itself. I.e. `comment_010` checks if comments are with their correct indentation and I think also the `-- vsg_on`, `--vsg_off` and `-- vsg_disable_next_line <rule>` should in general obey `comment_010`.

**The only exception** is `-- vsg_on [<other rules>] comment_010` itself: The simple reason is that the user explicitly has the ability to switch off the `comment_010` rule for a certain code section and will (very most likely) re-enable that rule again later in the source code by `-- vsg_on comment_010` [note that the exact same is true for the general `-- vsg_off` (without explicit rule(s)].

So the line containing `-- vsg_on [<other rules>] comment_010` or `-- vsg_on` (which re-enables `comment-010`) itself should not be checked against `comment_010`.

I hope that what I describe makes sense to you and am fully confident that you'd find the correct implementation.

jeremiah-c-leary, 2022-02-17:
Hey @staerz,

I think I understand now.  Code tags apply after the tag itself because almost all of the rules do not deal with comments.  However, `comment_10` does and so the code tag: `-- vsg_off comment_010` is being applied after the code tag itself.  Yet in this case, because it is a comment, it should also apply.

It seems the generate rules for code tags should be:

1)  `vsg_on` code tags apply on the token that defined it
2)  `vsg_off` code tags apply after the token that defined it

--Jeremy

jeremiah-c-leary, 2022-02-17:
hey @staerz,

I think I got the issue this time.  My example is passes now:
```vhdl
  1
  2 architecture rtl of fifo is
  3
  4 begin
  5
  6     -- vsg_off comment_010
  7
  8 -- Comment 1
  9                  -- Comment 2
 10 -- vsg_on comment_010
 11
 12 end architecture rtl;
```
When previously line 10 would have been flagged with `comment_010`.

--Jeremy

staerz, 2022-02-17:
Hi @jeremiah-c-leary,

yes, you got it! And yes, your understanding about the code tags is absolutely matching mine.

Congrats, you can close this issue now (once merged to master).

PR #734 (2022-02-20): Updates for issue #542:
Rule comment_010 was firing after block_comment rules were ran:

  1)  Updated indent logic to take block comments into consideration
  2)  Added attributes to comment class to indicate if it is a block comment
  3)  Updated tests
  4)  Simplified extractor
  5)  Added method to token_map

fixes #542
closed 2022-02-20 by jeremiah-c-leary