jeremiah-c-leary/vhdl-style-guide #543: Error parsing port map assignment with value of std_logic'('1'),

ericjmcd, 2021-03-24:
**Environment**
VSG 3.0.0 Ubuntu 18.04

**Describe the bug**
"Error: Unexpected token detected while parsing component_instantiation_statement" is thrown when parsing port assignments using type specifications (or whatever you call this syntax).  The usage is a bit odd but is valid and useful with Vivado due to some component matching logic they have.  If you care:
https://forums.xilinx.com/t5/Synthesis/Has-Vivado-unlearned-to-do-type-inference/td-p/553544

**To Reproduce**
Passes:
```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

entity junk is
end entity junk;

architecture behav of junk is

begin

  uut : component test_module
    port map (
      in1 => '1',
      --  in2 => std_logic'('1'),
      in3 => '0'
    );

end architecture behav;
```
Fails (and hangs) with
```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

entity junk is
end entity junk;

architecture behav of junk is

begin

  uut : component test_module
    port map (
      in1 => '1',
      in2 => std_logic'('1'),  -- explicit type constructor
      in3 => '0'
    );

end architecture behav;
```
Error text when running `vsg -f junk.vhd -of syntastic`
```
Error: Unexpected token detected while parsing component_instantiation_statement @ Line 15, Column 29 in file None
       Expecting : ;
       Found     : ,
None
```

jeremiah-c-leary, 2021-03-24:
Hey @ericjmcd ,

I was able to recreate the issue.  I will work on it tonight.

--Jeremy

jeremiah-c-leary, 2021-03-24:
Hey @ericjmcd ,

Quick question about this:
```vhdl
std_logic'('1');
```
Should the tick be present after the `std_logic`?

jeremiah-c-leary, 2021-03-24:
I found it in the LRM.

jeremiah-c-leary, 2021-03-24:
Hey @ericjmcd ,

I found the issue in the tokenizer.  It was considering `'('` as a character literal.

I pushed an update to master and will get a 3.1.0 build out by tomorrow.

--Jeremy

ericjmcd, 2021-03-31:
It looks like this fixes the problem - thanks!

jeremiah-c-leary, 2021-03-31:
Awesome!!

closed 2021-03-31 by jeremiah-c-leary