jeremiah-c-leary/vhdl-style-guide #544: Fix whitespace_011

staerz, 2021-03-25:
**Environment**
VSG 3.0.0

**Describe the bug**
whitespace_011 checks for spaces around signs like plus, minus, etc.

It does not make a distinction between a regular subtraction or the prefix of a negative number.

The following example illustrates the problem:
```vhdl
100   constant C_WIENER     : integer_vector(0 to 5) :=                             
101   (                                                                             
102     -9379,  -- c(0)                                                             
103     17867,  -- c(1)                                                             
104     44585,  -- c(2)                                                             
105     -68495, -- c(3)                                                             
106     36820,  -- c(4)                                                             
107     -8483   -- c(5)                                                             
108   );
```
cause the following errors
```
----------------------------+------------+------------+--------------------------------------
  whitespace_011            | Error      |        106 | Add 1 space(s) after -
  whitespace_011            | Error      |        108 | Add 1 space(s) after -
----------------------------+------------+------------+--------------------------------------
```
Note also that the reported lines are misplaced by one line.

A possible improvement for this rule could be to actually check for dual math operators, i.e. if there is a valid expression in front of the checked character.
Only whitespace or a comma (`,`) or semicolon (`;`) as VHDL syntactic indicators in front of the checked character should  result in whitespece_011 not to kick in.

**To Reproduce**
Steps to reproduce the behavior:
1. Run VSG on the code example above
2. Observe the error

**Expected behavior**
Whitespace_011 not kicking in in these cases.

jeremiah-c-leary, 2021-03-25:
Hey @staerz ,

I wonder if the comment before the negative sign is confusing the parser.  I remember putting in checks to determine if the `-` is a subtraction or a sign.

jeremiah-c-leary, 2021-03-25:
Hey @staerz ,

I just pushed an update for this.  If you could check it out I would appreciate it.

--Jeremy

staerz, 2021-03-25:
Hi @jeremiah-c-leary,

thanks for the ultimately quick fix for that. Yes, it works now. You can close the issue.

jeremiah-c-leary, 2021-03-25:
Woot!!!

closed 2021-03-25 by jeremiah-c-leary