jeremiah-c-leary/vhdl-style-guide #545: Support new VHDL-2008 unary operators

staerz, 2021-03-25:
**Environment**
VSG 3.0.0

**Describe the bug**
VHDL-2008 creates unary versions of AND, OR, NOR, NAND, XOR, and XNOR for logic array types, see [this brief summary](https://s3.amazonaws.com/verificationhorizons.verificationacademy.com/volume-8_issue-3/articles/stream/vhdl-2008-why-it-matters_vh-v8-i3.pdf).

VSG 3.0.0 does not yet properly react on those unary versions and requires a space instead, e.g. in the following simple code example:
```vhdl
user_led_g <= (others => not (and prbs_locked));                              
user_led_r <= (others => not (or error_flag));  
```

VSG 3.0.0 output: 
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  whitespace_013            | Error      |        247 | Add 1 space(s) before and. 
  whitespace_013            | Error      |        248 | Add 1 space(s) before or. 
----------------------------+------------+------------+--------------------------------------
```
but if one applies the recommended solutions the following errors are thrown:
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  whitespace_005            | Error      |        247 | Remove spaces after open (.
  whitespace_005            | Error      |        248 | Remove spaces after open (.
----------------------------+------------+------------+--------------------------------------
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run VSG 3.0.0 on the above code example
2. Observe the error report

**Expected behavior**
The solution should be to accept the initial code as no space is needed.
That would be by checking if the operator is unary or not by analysing the element to the left: if it's not an identifier, no space should be needed.

jeremiah-c-leary, 2021-03-25:
hmm...that was not obvious to me in the LRM, but I see it now.

staerz, 2021-03-25:
Don't worry, it wasn't obvious to me either. I just learned it today :wink:

jeremiah-c-leary, 2021-03-25:
Hey @staerz ,

I just pushed an update for this.  Could you validate on your side?

Thanks,

--Jeremy

staerz, 2021-03-25:
Yup, that works :+1: 

Feel free to close that one as well.

jeremiah-c-leary, 2021-03-25:
These are easy fixes.  It's indenting that is hard. :smile:

closed 2021-03-25 by jeremiah-c-leary