jeremiah-c-leary/vhdl-style-guide #546: Prevent option --fix to fix any passed file

staerz, 2021-03-26:
**Environment**
VSG 3.0.0

**Describe the bug**
Running VSG on all files to be checked may result in no errors be found. Taking a particular file, `bl_corr`, as an example:
```
================================================================================
File:  src/BCORR/bl_corr.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 520
Total Violations:    0
  Error   :     0
  Warning :     0
```
Now running VSG with the auto-fixing option `--fix` over the same files results in unwanted changes to the files without violations.
In particular the lines for the integration of images with doxygen are partly removed.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Have a VSG 3.0.0 compliant file with a doxygen comment including an image (spread over 2 lines where the quotes for the subtitle are spread over both lines)
2. Run VSG
3. Run VSG with `--fix`
4. See error

**Expected behavior**
If a file is passes without errors, running VSG with the `--fix` option should not result in any changes to the file.

jeremiah-c-leary, 2021-03-26:
Hey @staerz,

I just pushed a fix for this issue.  My tokenizer was not handling comments correctly.  Could you validate it on your end?

Thanks,

--Jeremy

staerz, 2021-03-26:
Hi @jeremiah-c-leary,

wow, that was quick!

Yes, I confirm the issue is fixed now. Thanks, you may close the issue.

jeremiah-c-leary, 2021-03-26:
> wow, that was quick!

I am for 5 star service  :smile:

closed 2021-03-26 by jeremiah-c-leary