jeremiah-c-leary/vhdl-style-guide #547: VSG crashes upon encountering empty comment in entity declaration

staerz, 2021-03-29:
**Environment**
VSG 3.0.0.

**Describe the bug**
VSG crashes upon encountering empty comment in entity declaration.

In particular it will crash with the following error:
```
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/pool.py", line 119, in worker
    result = (True, func(*args, **kwds))
  File "/home/sam/.local/lib/python3.6/site-packages/vsg/__main__.py", line 469, in apply_rules
    oRules.check_rules(bAllPhases=commandLineArguments.all_phases, lSkipPhase=commandLineArguments.skip_phase)
  File "/home/sam/.local/lib/python3.6/site-packages/vsg/rule_list.py", line 217, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "/home/sam/.local/lib/python3.6/site-packages/vsg/rules/block_comment/rule_002.py", line 50, in analyze
    if not is_header(oToken.get_value()):
  File "/home/sam/.local/lib/python3.6/site-packages/vsg/rules/block_comment/rule_002.py", line 80, in is_header
    if sComment[2] not in string.punctuation:
IndexError: string index out of range
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/sam/.local/bin/vsg", line 11, in <module>
    sys.exit(main())
  File "/home/sam/.local/lib/python3.6/site-packages/vsg/__main__.py", line 412, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "/usr/lib/python3.6/multiprocessing/pool.py", line 735, in next
    raise value
IndexError: string index out of range
```
when running on the following code snippet:
```vhdl
    inst_dummy : entity lib.module
    port map (
      -- Clocks
      clk   => clk
      --
      -- Dummy comment
      --
      data_i => data_i
    );
```

It doesn't seem to be every empty comment, just ones that follow this pattern (and only in entity declarations):
The comment with an empty comment above and an empty comment below.

**To Reproduce**
Steps to reproduce the behavior:
1. Run VSG on the above code snippet
1. See error

**Expected behavior**
VSG should not crash.

jeremiah-c-leary, 2021-03-29:
Hey @stears,

I looks like my block comment detector is having an issue.

--jeremy

jeremiah-c-leary, 2021-03-29:
Hey @staerz,

I used this snippet:
```vhdl
  1
  2 architecture rtl of fifo is
  3
  4 begin
  5
  6   inst_dummy : entity lib.module
  7     port map (
  8       -- Clocks
  9       clk => clk
 10       --
 11       -- Dummy comment
 12       --
 13       data_i => data_i
 14     );
 15
 16 end architecture rtl;
```
with this configuration:
```yaml
  1 rule:
  2    instantiation_034:
  3       disable : True
  4    block_comment_001:
  5       disable : False
  6       allow_indenting : False
  7       header_left : '+'
  8       header_right_repeat : '='
  9       header_string : '[ Header ]'
 10       max_header_column : 40
 11    block_comment_002:
 12       disable : False
 13       allow_indenting : False
 14       comment_left : '|'
 15    block_comment_003:
 16       disable : False
 17       allow_indenting : False
 18       max_footer_column : 40
```
and it does not crash.

I also tried adding this configuration for indenting:
```json
  1 {
  2   "indent": {
  3     "tokens": {
  4       "port_map_aspect": {
  5         "port_keyword": {
  6           "after": "current",
  7           "token": "-1"
  8         }
  9       }
 10     }
 11   }
 12 }
```
and it did not crash.

Could it possibly be something in your configuration?

I also noticed the comma was missing after `clk => clk`.

staerz, 2021-03-29:
Hi @jeremiah-c-leary,

OK, so here is our configuration (I couldn't find a better way than simply c&p here):

```json
{
  "indent":{
    "tokens":{
      "component_instantiation_statement":{
        "instantiation_label":{
          "token" : "current",
          "after" : "current",
        },
        "semicolon":{
          "token" : "current",
          "after" : "current",
        },
      },
    },
  },
  "rule":{
    "architecture_003": {
      "style":"require_comment"
    },
    "architecture_010":{
      "action":"add",
    },
    "architecture_015":{
      "style":"require_blank_line"
    },
    "architecture_016":{
      "style":"require_blank_line"
    },
    "architecture_017":{
      "style":"require_blank_line"
    },
    "architecture_018":{
      "style":"require_blank_line"
    },
    "architecture_025":{
      "disable":false,
      "names":[
        "behavioral",
        "emulational",
        "empty",
        "tb",
      ],
    },
    "assert_400":{
      "alignment":"left",
    },
    "block_002":{
      "action":"remove"
    },
    "block_007":{
      "action":"add"
    },
    "block_200":{
      "style":"allow_comment"
    },
    "block_201":{
      "style":"no_blank_line"
    },
    "block_202":{
      "style":"no_blank_line"
    },
    "block_203":{
      "style":"require_blank_line"
    },
    "block_204":{
[... 265 lines cut ...]
    },
    # This rule checks for comments within multiline sequential statements.
    "sequential_006":{
      "disable":true
    },
    # This rule checks for a blank line below the type declaration.
    # Checking below sth. makes few sense...
    "type_011":{
      "disable":true
    },
    # This rule checks for comments in multiline variable assignments.
    "variable_assignment_006":{
      "disable":true
    },
    # how does one dare to tag this as an error ...
    "with_001":{
      "disable":true
    },
  }
}
```

Maybe this would provoke the error.

If not, I would try to get you more context on the source file it crashes on.

jeremiah-c-leary, 2021-03-31:
Hey @staerz ,

I tried your configurations and I was not able to get my snippet to fail.

--Jeremy

samdejong86, 2021-04-01:
Hi @jeremiah-c-leary 

I'm working with @staerz and first found this issue. Here's an example that crashes for me. I've hacked it down to the minimum required to make it fail

```vhdl
entity foo is
  generic (
    --! enable downlink path
    blah                 : boolean := true
  );
  port (
    -- Clocks
    --! Downlink datapath clock (MGT tx clock)
    clk     : in    std_logic                        := '0';
    -- Control
    ctr  : out   std_logic
  );
end entity foo;

--! Implementation of lpGBT_top
architecture behavioral of foo is

begin

    inst_dummy : entity lib.module
    port map (
      -- Clocks
      clk   => clk
      --
      -- Dummy comment
      --
      data_i => data_i
    );

  gen_bar : if FOOBAR generate

    inst_something : entity work.something
    generic map (
      C_CLOCKRATIO  => 8,
      C_OUTPUTWIDTH => 32
    )
    port map (
      -- Clocks
      clk_i               => clk,
      rdy_o               => Ready_o

    );

  end generate gen_bar;

end architecture behavioral;

```

jeremiah-c-leary, 2021-04-01:
Hey @samdejong86,

Thanks for the code.  I was able to recreate the error on my side and reduce it further:
```vhdl
  1
  2 architecture behavioral of foo is
  3
  4 begin
  5
  6     inst_dummy : entity lib.module
  7     port map (
  8       --
  9       -- Dummy comment
 10       --
 11       data_i => data_i
 12     );
 13
 14     -- Comment
 15
 16 end architecture behavioral;
```
It is the comment on line 14 that is causing the issue.  I will investigate.  It will be later tonight though.

--Jeremy

jeremiah-c-leary, 2021-04-01:
Hey @samdejong86,

I just pushed an update for this issue.  When you get a chance, could you validate it on your end.

Thanks,

--Jeremy

samdejong86, 2021-04-01:
Thanks @jeremiah-c-leary , I can confirm that it works.

jeremiah-c-leary, 2021-04-01:
Awesome!!! I will close this ticket.

closed 2021-04-01 by jeremiah-c-leary